thesofproject/sof #1526: [BUG] start/stop of SSP/nodec stream interferes with another stream

kv2019i, 2019-06-06:
**Describe the bug**
This is a clone of Linux bug https://github.com/thesofproject/linux/issues/854 
The history is a bit complex to follow, but upon study of the bug today, the case is as follows:

Two PCM playback clients connected to SSP in nocodec mode. One client is kept stream, while other continuously starts and stops. 

**To Reproduce**
Aplay instructions in 
https://github.com/thesofproject/linux/issues/854#issuecomment-497228999

**Reproduction Rate**
1/10

**Expected behavior**
Starting and stopping one stream does not interfere with another stream.

**Impact**
End-product cases with multiple pipelines will show errors (e.g. a UI sound feedback may interfere with HDMI output).

**Environment**
APL UP2 pcm512x & APL UP2 nocodec 
sof (master): 92fc7067
kernel (topic/sof-dev): 80a2cb8
topology: sof-apl-pcm512x.tplg & sof-apl-nocodec.tplg

**Screenshots or console output**
See https://github.com/thesofproject/linux/issues/854#issuecomment-497228999

kv2019i, 2019-06-06:
[soflog.txt](https://github.com/thesofproject/sof/files/3262189/soflog.txt)

Annotated bits from above log.
client 1 (all the time running) is started
```
    0      2         PIPE 14.62   2799191127.291667        22.031250 src/audio/pipeline.c:607 	pipeline_trigger()
    0      2         HOST         2799191131.406250         4.114583     src/audio/host.c:262 	host_trigger()
    0      2          DMA         2799191135.572917         4.166667 intel/cavs/hda-dma.c:531 	hda-dmac: 5 channel 1 -> start
    0      2          DMA         2799191139.218750         3.645833 intel/cavs/hda-dma.c:367 	hda-dmac: 5 channel 1 -> enable
    0      2       VOLUME         2799191176.510417        37.291668 udio/volume/volume.c:464 	volume_trigger()
    0      2          SRC         2799191180.468750         3.958333  src/audio/src/src.c:682 	src_trigger()
    0      2        MIXER         2799191184.479167         4.010417    src/audio/mixer.c:211 	mixer_trigger()
    0      2         COMP         2799191187.916667         3.437500 rc/audio/component.c:105 	comp_set_state(), state already set to 5
```

Then client 2 (start/stop in loop) is stopped and started and start fails
```
    0      2         PIPE 1.9     2817044901.093750       166.718750 src/audio/pipeline.c:378 	pipeline_prepare()
    0      2         HOST         2817044905.052083         3.958333     src/audio/host.c:614 	host_prepare()
    0      2       VOLUME         2817044918.489583        13.437500 udio/volume/volume.c:522 	volume_prepare()
    0      2        MIXER         2817044931.770833        13.281250    src/audio/mixer.c:358 	mixer_prepare()
    0      1         COMP         2817045834.635417       902.864563 rc/audio/component.c:237 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      2         COMP         2817045840.156250         5.520833 of/audio/component.h:726 	comp_overrun(), ((dev->comp.id << 16) | sink->free) = 131072, 
```

Xrun recovery fails for client 2, but that also brings down client 1:

```
    0      2         PIPE 14.62   2817439722.968750        21.041666 src/audio/pipeline.c:607 	pipeline_trigger()
    0      2         HOST         2817439726.770833         3.802083     src/audio/host.c:262 	host_trigger()
    0      2          DMA         2817439730.833333         4.062500 intel/cavs/hda-dma.c:626 	hda-dmac: 5 channel 1 -> stop
    0      2       VOLUME         2817439735.156250         4.322917 udio/volume/volume.c:464 	volume_trigger()
    0      2          SRC         2817439739.062500         3.906250  src/audio/src/src.c:682 	src_trigger()
    0      2        MIXER         2817439742.812500         3.750000    src/audio/mixer.c:211 	mixer_trigger()
    0      2          IPC         2817439894.218750       151.406250    src/ipc/handler.c:328 	ipc: comp 56 -> free
    0      2         PIPE 14.62   2817439915.260417        21.041666 src/audio/pipeline.c:686 	pipeline_reset()
```

lgirdwood, 2019-06-06:
@kv2019i are these two pipelines connected via the mixer ? or are they not connected. Just confused seeing HDMI and SSP then seeing MIXER in the logs.

kv2019i, 2019-06-07:
@lgirdwood wrote:
> @kv2019i are these two pipelines connected via the mixer ? or are they not connected. Just confused seeing HDMI and SSP then seeing MIXER in the logs.

Good point, yes they are. This is the standard pipe of:
https://github.com/thesofproject/sof/blob/master/tools/topology/sof-apl-nocodec.m4

[image]

Updated the title to reflect this related to two streams connected via mixer.

tlauda, 2019-06-07:
@kv2019i Fail happens when you are stopping client 0 and client 1 is running. As long as you are stopping client 1 everything is working. It doesn't really matter, because it's still a bug, but just wanted to make sure you are aware of that.

PR #1589 (2019-06-29): pipeline: change preload conditions
Changes pipeline's preload conditions to take into
consideration the situation, where we don't always
want to do playback preload, because the sink part
of pipeline is already running e.g. mixer topology.
However the host component still needs to perform
preload on its own, so for now we make the decision
based on stream's direction.

Fixes #1526.
closed 2019-06-29 by lgirdwood