thesofproject/sof #1527: [BUG] KWD: not able to trigger detection with glk-012-stable-branch based FW

yongzhi1, 2019-06-06:
**Describe the bug**

Checkout glk-012-stable-branch and build FW, not able to trigger key phase detection using default clap hands sound.

mkdir build_apl && cd build_apl
cmake -DTOOLCHAIN=xtensa-apl-elf -DROOT_DIR=`pwd`/../../xtensa-root/xtensa-apl-elf ..
make apollolake_defconfig
make bin -j4

**To Reproduce**
arecord -Dhw:0,8 -M -N -c2 -f S16_LE -r 16000 /tmp/test.wav -vvv

**Reproduction Rate**
100%

**Expected behavior**
Keyword should be detected.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1. kernel: cros-sof-v4.14-rebase
2. firmware: glk-012-stable-branch
3. topology: sof-glk-da7219-kwd.m4

**Screenshots or console output**

localhost ~ # arecord -Dhw:0,8 -M -N -c2 -f S16_LE -r 16000 /tmp/test.wav -vvv
Recording WAVE '/home/tmp.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Stereo
Hardware PCM card 0 'sof-glkda7219max' device 8 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : MMAP_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 16000
  exact rate   : 16000 (16000/1)
  msbits       : 16
  buffer_size  : 8016
  period_size  : 2016
  period_time  : 126000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 2016
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 8016
  silence_threshold: 0
  silence_size : 0
  boundary     : 9025213653250473984
  appl_ptr     : 0
  hw_ptr       : 0
mmap_area[0] = 0x7e234b7e9000,0,32 (16)
mmap_area[1] = 0x7e234b7e9000,16,32 (16)

lgirdwood, 2019-06-07:
@yongzhi1 can you include the firmware log too. @mrajwa could we be missing a patch ?

mrajwa, 2019-06-07:
@lgirdwood, he has also notified me about the problem via email (providing logs). I checked that and it looks like his gcc build is treating config structs differently than xt-xcc does. Therefore he has incorrect config. For example `cd->config.activation_threshold` has some trash value, therefore detection will never happen. Once I am done with my current work I will provide a gcc fix for that.

mrajwa, 2019-10-25:
@tlauda, no we can close that.