thesofproject/sof #1543: [BUG] Cppcheck warnings

plbossart, 2019-06-10:
**Describe the bug**

Cppcheck throws quite a few warnings on the SOF code.

**To Reproduce**
Install cppcheck and run this command
cppcheck --platform=unix32 --force --max-configs=1024 --inconclusive --enable=all src

**Reproduction Rate**
N/A

**Expected behavior**
Zero unexplained warnings

**Impact**
Code quality

**Environment**
SOF master

**Screenshots or console output**
N/A. Results are attached

[log_filtered.txt](https://github.com/thesofproject/sof/files/3273356/log_filtered.txt)

lgirdwood, 2019-06-10:
@xiulipan fyi, cppcheck could be added to CI

xiulipan, 2019-06-11:
@lgirdwood 
Got it.

@plbossart 
Do we also need cppcheck for linux kernel?

lgirdwood, 2019-06-11:
@mengdonglin fyi, will need someone to fix the issues.

plbossart, 2019-06-11:
On 6/10/19 10:39 PM, xiulipan wrote:
> @lgirdwood <https://github.com/lgirdwood>
> Got it.
> 
> @plbossart <https://github.com/plbossart>
> Do we also need cppcheck for linux kernel?

Cppcheck generates a lot of false alarms and is quite hard to configure, 
so I am not sure if it makes sense to add it as a required gate keeping 
tool.
I know the Canonical folks use it but to check for new errors, not to go 
down to zero warnings.
If we can add it as an informative tool it could be useful though.

I run cppcheck on the kernel manually, along with coccinelle. you're 
certainly welcome to automate both, but they should be more informative 
than pre-requisites to merge.

xiulipan, 2019-06-13:
@plbossart 
I am going to add following cmd to run in CI with cppcheck, do you think it is ok?
`cppcheck --platform=unix32 --force --max-configs=1024 --inconclusive --enable=all sound/soc/sof`

plbossart, 2019-06-13:
On 6/13/19 10:18 AM, Pan Xiuli wrote:
> @plbossart <https://github.com/plbossart>
> I am going to add following cmd to run in CI with cppcheck, do you think 
> it is ok?
> |cppcheck --platform=unix32 --force --max-configs=1024 --inconclusive 
> --enable=all sound/soc/sof

As I mentioned you are going to see a ton of warnings. cppcheck does not 
deal well with callbacks - reports them as being unused - and provides 
too many reports of variable scope
I think cppcheck should be used to check *new* warnings only - and only 
on an informative basis. It shouldn't block CI.

lgirdwood, 2019-06-14:
@xiulipan @plbossart so this is maybe something we need to run weekly instead of CI so that a human can fix without blocking CI. Fwiw, we could store the false positives in git so that any developer can diff the false positives against the latest generated report file.

lgirdwood, 2019-06-14:
@mengdonglin most of these issues are valid and are relatively simple fixes. One days work to fix most of them.

xiulipan, 2019-06-14:
@lgirdwood 
Like @plbossart said, there will be tons of false alarm and I do not think human can identify what is **new**  warning when we see them.

Maybe store every scan log in daily and try to diff the logs may be a good solution here. I will try to enable cppcheck for daily build and store the report.

plbossart, 2019-06-15:
> @mengdonglin most of these issues are valid and are relatively simple fixes. One days work to fix most of them.

The more complicated parts are the void pointer arithmetic, which is easy to fix in a wrong way or to make the code really unreadable.

PR #2550 (2020-03-16): sai: remove duplicated values
Removes duplicated values from expression.

Fixes #1543.
closed 2020-03-16 by dbaluta