thesofproject/sof #1556: [FEATURE] topology IPC builder

lgirdwood, 2019-06-12:
**Is your feature request related to a problem? Please describe.**
Static topology is currently hard coded into FW binaries as C code. This is inflexible and does not support reuse of existing tools. We need to be able to stitch in topology data as an array of IPC messages into the FW binary (as an ELF section).

**Describe the solution you'd like**
1) Small cmd line tool that converts native kernel topology binaries into IPC topology.
2) Kconfig option to pass in file name of IPC topology in 1) and stitch into FW image.

**Additional context**
@lyakh @ranj063 and @singalsu have all coded parts of this for test bench, fuzzer etc. Neds to be consolidated and integrated as library code that can be used by fuzzer, test bench, and 1).

ranj063, 2019-06-12:
@lgirdwood @kv2019i the testbench only does the tplg binary parsing but skips the IPC. But the fuzzer branch has the complete code for this.

lgirdwood, 2019-06-13:
@ranj063 great thanks, we can separate this code out and build as a library that can bus used by fuzzer, test bench and FW.

kv2019i, 2019-09-20:
After discussion with @lgirdwood tagging with P3 and removing Notable for now.

lgirdwood, 2021-01-12:
@ranj063 @singalsu I guess we can load topoies in testbench now ? Can we close ?

lgirdwood, 2021-05-10:
Static C topology removed from FW. Testbench uses topology library.
Moved to v1.9 for completion of Testbench library in FW.

kv2019i, 2023-02-15:
This is still a valid item, but currently no development is ongoing, so moving this to TBD milestone.

lgirdwood, 2023-02-20:
WIP: See lrg/topic/plugin branch