thesofproject/sof #1574: [BUG] Build fails with TRACE disabled

lgirdwood, 2019-06-18:
**Describe the bug**
Disable all trace in Kconfig.
```
[ 83%] Building C object CMakeFiles/sof.dir/src/drivers/dw/dma.c.o
/home/lrg/work/sof/xtensa-cnl-elf/bin/xtensa-cnl-elf-gcc  -I/home/lrg/work/sof/sof/src/platform/intel/cavs/include -I/home/lrg/work/sof/sof/src/arch/xtensa/include -I/home/lrg/work/sof/sof/src/arch/xtensa/smp/include -I/home/lrg/work/sof/sof/src/arch/xtensa/smp/xtos -I/home/lrg/work/sof/sof/src/platform/icelake/include -I/home/lrg/work/sof/sof/src/platform/icelake/include/arch -I/home/lrg/work/sof/sof/../xtensa-root/xtensa-cnl-elf/include -I/home/lrg/work/sof/sof/src/include -I/home/lrg/work/sof/sof/build_icl_gcc/generated/include  -nostdlib -fno-inline-functions -mlongcalls -O2 -g -Wall -Werror -Wl,-EL -Wmissing-prototypes -mtext-section-literals -o CMakeFiles/sof.dir/src/drivers/dw/dma.c.o   -c /home/lrg/work/sof/sof/src/drivers/dw/dma.c
/home/lrg/work/sof/sof/src/drivers/intel/cavs/hda-dma.c: In function 'hda_dma_wait_for_buffer_full':
/home/lrg/work/sof/sof/src/drivers/intel/cavs/hda-dma.c:263:11: error: variable 'wp' set but not used [-Werror=unused-but-set-variable]
  uint32_t wp;
           ^~
/home/lrg/work/sof/sof/src/drivers/intel/cavs/hda-dma.c:262:11: error: variable 'rp' set but not used [-Werror=unused-but-set-variable]
  uint32_t rp;
           ^~
/home/lrg/work/sof/sof/src/drivers/intel/cavs/hda-dma.c: In function 'hda_dma_wait_for_buffer_empty':
/home/lrg/work/sof/sof/src/drivers/intel/cavs/hda-dma.c:291:11: error: variable 'wp' set but not used [-Werror=unused-but-set-variable]
  uint32_t wp;
           ^~
/home/lrg/work/sof/sof/src/drivers/intel/cavs/hda-dma.c:290:11: error: variable 'rp' set but not used [-Werror=unused-but-set-variable]
  uint32_t rp;
           ^~
```
Error trace should should write only to persistent/shared buffer when TRACE is disabled.

lgirdwood, 2019-06-18:
rimage also broken too when TRACE disabled..
```

Module Reading bootloader-icl
info: ignore .bss section for bootloader module
error: can't find section .static_log_entries in module sof-icl
info: ignore .static_log_entries section for bootloader module
error: failed to read base sections -22
info: ignore .fw_ready section for bootloader module
make[2]: *** [src/arch/xtensa/CMakeFiles/run_rimage.dir/build.make:60: src/arch/xtensa/CMakeFiles/run_rimage] Error 234
  Found 22 sections, listing valid sections......
make[1]: *** [CMakeFiles/Makefile2:1357: src/arch/xtensa/CMakeFiles/run_rimage.dir/all] Error 2
	No	Start		End		Size	Type	Name
make: *** [Makefile:133: all] Error 2
	0	0x00000000	0x00000000	0x0
	1	0xb0038000	0xb00380fb	0xfb	TEXT	.boot_entry.text
	2	0xb0038120	0xb0038140	0x20	TEXT	.boot_entry.literal
	3	0xb0038150	0xb0038809	0x6b9	TEXT	.text
	4	0xb0039000	0xb0039008	0x8	DATA	.rodata

 module: input size 2012 (0x7dc) bytes 5 sections
 module: text 2004 (0x7d4) bytes
    data 8 (0x8) bytes
    bss  0 (0x0) bytes
```

abonislawski, 2019-06-18:
rimage problem looks like rimage is always looking for "static_log_entries" section and needs it to move on, but in this case it was not created by linker (TRACE disabled).
I think we can fix this by removing this requirement, it will still be looking for "static_log_entries" section but its still ok if not found.
Also looks like rimage is already ready for that scenario because later will check if that section was found before using it.
@jajanusz please comment if it looks ok

jajanusz, 2019-06-18:
@abonislawski @lgirdwood Yea, It was optional before, just changed it to mandatory in non-bootloader modules while it should be optional at all.

jajanusz, 2019-06-25:
Fixed in #1579 + #1580

closed 2019-06-25 by jajanusz