thesofproject/sof #1575: [BUG] 2 Sue Creek regressions

lyakh, 2019-06-18:
SOF support for Sue Creek doesn't have a high priority, but so far it is the only platform, that can be used for testing of the Cascading Interrupt Controller patch series #1053 .

Two regressions have been identified in the current SOF master branch (as of https://github.com/thesofproject/sof/commit/0dd6c238146d2465d39d5d26eb6ebf191a20f251) on Sue Creek:

1. PR #1483 broke bootloader startup on Sue Creek. The easiest fix might be reverting those changes for Sue Creek only, if possible.

2. With reverted #1483 booting still fails sporadically. In these cases the SPI communication with the host stops after the initial 4 IPC messages. This results in the following output on the host:
```
[  538.231905] sof-audio sof-audio: ipc rx: 0x70000000: FW_READY
[  538.231930] sof-audio sof-audio: Firmware info: version 0:0-0 build 291 on Jun 18 2019:15:05:39
[  538.231940] sof-audio sof-audio: booting DSP firmware completed
[  538.231958] sof-audio sof-audio: ipc rx done: 0x70000000: FW_READY
[  538.231975] sof-audio sof-audio: firmware boot complete
[  538.232405] sof-audio sof-audio: generating page table for 50e73d5d size 0x10000 pages 16
[  538.232421] sof-audio sof-audio: dma_trace_pages: 16
[  538.232474] sof-audio sof-audio: stream_tag: 0
[  538.232516] sof-audio sof-audio: ipc tx: 0x90010000: GLB_TRACE_MSG
[  538.540225] sof-audio sof-audio: error: ipc timed out for 0x90010000 size 0x28
[  538.540253] sof-audio sof-audio: error: can't set params for DMA for trace -110
```
whereas the normal case looks as follows:
```
[   61.684563] sof-audio sof-audio: ipc rx: 0x70000000: FW_READY
[   61.684584] sof-audio sof-audio: Firmware info: version 0:0-0 build 291 on Jun 18 2019:15:05:39
[   61.684594] sof-audio sof-audio: booting DSP firmware completed
[   61.684615] sof-audio sof-audio: ipc rx done: 0x70000000: FW_READY
[   61.684637] sof-audio sof-audio: firmware boot complete
[   61.685315] sof-audio sof-audio: generating page table for (ptrval) size 0x10000 pages 16
[   61.685331] sof-audio sof-audio: dma_trace_pages: 16
[   61.685383] sof-audio sof-audio: stream_tag: 0
[   61.685427] sof-audio sof-audio: ipc tx: 0x90010000: GLB_TRACE_MSG
[   61.685690] sof-audio sof-audio: ipc rx: 0x10000000: GLB_REPLY
[   61.685701] sof-audio sof-audio: error: ipc reply unknown
[   61.685712] sof-audio sof-audio: ipc rx done: 0x10000000: GLB_REPLY
[   61.685735] sof-audio sof-audio: ipc tx succeeded: 0x90010000: GLB_TRACE_MSG
[   61.686046] sof-audio sof-audio: created machine sof-nocodec
```

lgirdwood, 2019-06-18:
@lyakh this should be 2 issues with descriptive titles. Have you tried booting on qemu ?

abonislawski, 2019-06-18:
> 1. PR #1483 broke bootloader startup on Sue Creek. The easiest fix might be reverting those changes for Sue Creek only, if possible.

As I already commented in this PR: there is obvious bug for suecreek and it is already fixed in PR #1567 
We cannot test suecreek platform, that's why we didnt spotted it before

lgirdwood, 2019-06-20:
@xiulipan I guess qemu CI did not detect this issue ? I would have expected boot failure.

xiulipan, 2019-06-20:
@lgirdwood 
We did not enable QEMU with Suecreek yet, is the QEMU support it now?

abonislawski, 2019-06-20:
@xiulipan @lgirdwood it would be great to enable boot test for sue creek if possible

lyakh, 2019-06-21:
@abonislawski no, #1567 doesn't fix it. I'm surprised we don't have Sue Creek in our CI, @xiulipan hasn't it been once included there?

jajanusz, 2019-06-21:
@lyakh There is building in CI, but no boot test. I guess that's because sue boot process is different than on other platforms.

abonislawski, 2019-06-21:
@lyakh yes, #1567 fix boot problem because boot stack was placed at address "0". If there is another bug then we need a platform to test it.

lgirdwood, 2019-06-23:
@xiulipan yes we have sue creek qemu support, it should boot.

xiulipan, 2019-06-24:
@lgirdwood 
Let me try and add support into CI. And I think I need to test with latest QEMU also.

lyakh, 2019-06-24:
@lgirdwood would you have a pointer to up-to-date qemu instructions?

lgirdwood, 2019-06-24:
@lyakh I guess qemu is not building for you ? 

https://alsa-project.org/wiki/Firmware#Using_the_Qemu_DSP_emulator

There are differences with dependencies compared with the old instruction on alsa wiki on newer distros, can you try configuring with 
```
'./configure' '--prefix=.' '--target-list=xtensa-softmmu,x86_64-softmmu' '--enable-gtk' '--enable-sdl' '--enable-spice' '--audio-drv-list=alsa' '--enable-libusb' '--enable-usb-redir' '--enable-coroutine-pool' '--disable-opengl' '--enable-fdt'
```
Please also add the latest build instruction to sof-docs.

lyakh, 2019-06-25:
@lgirdwood sorry, I just never tried using qemu for a Sue Creek emulation. Would that only run the boot simulation, since I don't see support for an SPI connection between the host and the xtensa?

lgirdwood, 2019-06-25:
@lyakh yes, that's correct there is no SPI simulation yet (although I have not checked if DW SPI is used by other qemu targets). Boot test is all we are doing, so FW will write to SPI regs (and wont crash).

lyakh, 2019-06-25:
> @lyakh yes, #1567 fix boot problem because boot stack was placed at address "0". If there is another bug then we need a platform to test it.

@abonislawski I just retested - the current top of "master" branch with this your PR applied still doesn't boot. It might well fix one issue with the memory-layout commit on Sue Creek, but there are definitely more problems. I would prefer a complete fix over a patch that improves the situation theoretically, but still cannot be successfully tested.

tlauda, 2020-03-12:
@xiulipan Is there any progress with adding sue to qemu boot test?

slawblauciak, 2020-09-15:
@lgirdwood Sue Creek support is not complete in SOF as a whole and there are no use cases/commitment. Do you think we could close this for the time being?

lgirdwood, 2020-09-15:
@slawblauciak @lyakh Luckily enough Sue creek is supported by Zephyr so we can remove the IRQ code. 

Lets keep the SUE target and platform stuff atm though as this may be needed to complete the Zephyr integration. We can remove this when we know it's no longer needed (probably after v1.6 and after Zephyr 2.5 releases)..

mengdonglin, 2020-09-15:
@xiulipan Can you help disable Sue Creek building in Travis CI?

marc-hb, 2021-06-15:
@lyakh do you still care about this?

lyakh, 2021-06-16:
> @lyakh do you still care about this?

I still have two Sue Creek boards! :-D But frankly - their support has never been completed, it's just sitting there bit-rotting by the minute. So I'd be grateful to anyone submitting a clean PR to completely but **cleanly** wipe out Sue Creek support from SOF.

closed 2025-07-04 by github-actions[bot]