thesofproject/sof #1576: [FEATURE] Linker Section alignment for BYT, HSW, IMX8

lgirdwood, 2019-06-18:
**Is your feature request related to a problem? Please describe.**
Linker section alignment can be optimised like https://github.com/thesofproject/sof/pull/1568 for CAVS targets.

**Describe the solution you'd like**
Same optimisation for non CAVS targets.

dbaluta, 2019-06-20:
@paulstelian97 can you please have a look at this?

dbaluta, 2019-06-24:
@paulstelian97 Thanks! So, this is resolved for IMX8. Intel people need to have a look for byt and hsw.

lgirdwood, 2019-06-24:
@dbaluta great, @mengdonglin this one is trivial, needed for non CAVS versions.

lgirdwood, 2019-08-30:
@mengdonglin anyone with a free hour to do this for v1.4rc2 ?

lgirdwood, 2020-01-27:
@mengdonglin ping

lgirdwood, 2020-08-18:
Seems not needed on older platforms and done on the newer ones. Closing.

closed 2020-08-18 by lgirdwood