thesofproject/sof #1577: [FEATURE] Allocator custom block mappings via kconfig

lgirdwood, 2019-06-18:
**Is your feature request related to a problem? Please describe.**
Allow allocator HEAP pool and block mappings to be customised via Kconfig.

**Describe the solution you'd like**
Kconfig menu would allow for sizing of HEAP pools like IMR, LPSRAM, uncache, etc (size == 0 means pools is disabled) and also allow the block mappings for each size to be specified. Kconfig not expected to validate the selection (unless it can) as compiler should throw an error if pre-processing fails.

lgirdwood, 2021-01-12:
Supported in Zephyr. Moving to v1.9 as part f Zephyr integration.

kv2019i, 2023-02-15:
Moving to v2.6 for evalution. This is partially addressed with current SOF Zephyr support, but the heap sizes are still not configurable via Kconfig.

lgirdwood, 2023-05-10:
Fixed by Zephyr.

closed 2023-05-10 by lgirdwood