thesofproject/sof #1578: [BUG] Overrun during pause-release

ranj063, 2019-06-18:
**Describe the bug**
System shows I/O error when pause arecord then resume. Similar behaviour with all arecord devices. Looks like a pipeline overrun when the user releases pause.

**To Reproduce**
1. Arecord via "arecord -Dhw:0,1 -r48000 -c2 -fs16_le 1.wav -vv -i"
2.Pause arecord
3.Resume asrecord

**Reproduction Rate**
100% on GLK Ubuntu
On BYT it happens after a few iterations of pause/release

**Expected behavior**
Resume playback after pause-release

**Impact**
showstopper

**Environment**
Firmware: c0c858c https://github.com/thesofproject/sof/commits/master
Kernel: a50a20a https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-glk-da7219.tplg same as firmware
platform： GLK

**Screenshots or console output**
please refer to https://github.com/thesofproject/linux/issues/1022

ranj063, 2019-06-20:
@tlauda FYI. what I notice is that on GLK, the first dai_copy() after release ends up in a overrun because dd->dma_buffer->free is always 0. Could you please take a look?

ranj063, 2019-06-20:
@michalgrodzicki could you please assign someone to take a into this?

mrajwa, 2019-06-20:
@ranj063 looks like we are facing the same issue on in KWD although conditions are a bit different. Have you spotted commit which causes it?

ranj063, 2019-06-20:
@mrajwa no I am not sure when this feature ever worked on GLK. I'll try to find out today.

keyonjie, 2019-06-21:
@mrajwa @ranj063 I think you hit 2 kinds of xrun here, they are both because of copy_bytes == 0:
copy_bytes = MIN(avail_bytes, dd->dma_buffer->free),
but @ranj063's case is dd->dma_buffer->free == 0 after pause/release, and @mrajwa's case is avail_bytes from dw-dma driver is 0.

For the former case, we can check how it handled the pausing(e.g. does the last pipeline_copy() is finished before paused? if the dai_copy() is done but the downstream copying(e.g. volume_copy()) is not run, then the dd->dma_buffer may be full).

For the latter case, looks like the logic for chan->ptr_data.current_ptr for capture(without preload) stream is problematic, but let's wait for the author @tlauda's confirm.

ranj063, 2019-06-21:
> @mrajwa no I am not sure when this feature ever worked on GLK. I'll try to find out today.

@mrajwa I've gone back as far as a month and I still can reproduce this issue.

YvonneYang2, 2019-06-24:
Issue reproduced when stop via "ctrl + z" , then resume via command "fg".

**Test recipe:**
Firmware: 767a81bd-2 https://github.com/thesofproject/sof/tree/master
Kernel: 0087d4af-2 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: sof-glk-da7219.tplg same as firmware
platform ：GLK

**console**
```
Max peak (4080 samples): 0x01da6ba9 #                    1%
Max peak (4080 samples): 0x011ca6ff #                    0%
Max peak (4080 samples): 0x017b8954 #                    1%
^Z
[1]+  Stopped                 arecord -Dhw:0,1 -f S32_LE -r 48000 -c 2 -i test.wav -vvv -i
test@test-Bobba:~$ fg
arecord -Dhw:0,1 -f S32_LE -r 48000 -c 2 -i test.wav -vvv -i
Max peak (816 samples): 0x011ca6ff #                    0%
overrun!!! (at least 986.694 ms long)
Status:
  state       : XRUN
  trigger_time: 4317.159813239
  tstamp      : 0.000000
  delay       : 0
  avail       : 10224
  avail_max   : 10224
arecord: pcm_read:2103: read error: Input/output error
```
[xrun_error_dmesg.log](https://github.com/thesofproject/sof/files/3319057/xrun_error_dmesg.log)
[xrun_error_logger.log](https://github.com/thesofproject/sof/files/3319058/xrun_error_logger.log)

tlauda, 2019-06-24:
@ranj063 Could you share FW logs? What is the sequence from kernel perspective on pause/release? What stream bits are you setting?

@YvonneYang2 In your logs I see that you are doing stop/start instead of pause/release. It looks like there is some kind of host pointer mismatch after such sequence.

ranj063, 2019-06-24:
@tlauda the firmware logs are in https://github.com/thesofproject/linux/issues/1022

For the sequence, please refer to https://github.com/thesofproject/linux/blob/0087d4afe31bad40a0f365b774547fd69124d58c/sound/soc/sof/intel/hda-stream.c#L215

tlauda, 2019-06-24:
@ranj063 Logs in thesofproject/linux#1022 are from the older version of FW without couple of crucial fixes.

ranj063, 2019-06-24:
@fredoh9 could you please attach the logs from the GLK Ubuntu with the latest codebase for SOF and sof-dev?

ranj063, 2019-06-25:
@tlauda here's the logger output:
[glk-log.txt](https://github.com/thesofproject/sof/files/3323213/glk-log.txt)

ranj063, 2019-06-25:
and here's the dmesg:
[glk-dmesg.txt](https://github.com/thesofproject/sof/files/3323215/glk-dmesg.txt)

YvonneYang2, 2019-06-25:
I did two ways to check if both can reproduce the issue before I caught the logs, so there may be play/pause in the dmesg.
I catch the new logs with the test recipe https://github.com/thesofproject/sof/issues/1578#issuecomment-504840027 on glk ubuntu as below, please check them. The logs is classified by the different operations.  The stop/resume log only covers the stop/resume operation. Then do "dmesg -C", reproduced the issue by pause/resume operation, caught the pause/resume log.

[stop_resume_xrun_dmesg.log](https://github.com/thesofproject/sof/files/3323206/stop_resume_xrun_dmesg.log)
[stop_resume_xrun_logger.log](https://github.com/thesofproject/sof/files/3323207/stop_resume_xrun_logger.log)
[play_pause_xrun_dmesg.log](https://github.com/thesofproject/sof/files/3323208/play_pause_xrun_dmesg.log)
[play_pause_xrun_logger.log](https://github.com/thesofproject/sof/files/3323209/play_pause_xrun_logger.log)

tlauda, 2019-06-25:
@ranj063 Kernel doesn't restart capture host stream on release. In the logs we can see that only the first host copy is successful and after incrementing the pointers data isn't consumed by Host DMA. I've done such experiment with slim driver and I get the same result, when I don't start again Host DMA.

ranj063, 2019-06-25:
@tlauda the PAUSE_RELEASE logic in the kernel is exactly the same as that for START. But the thing that is different is that we do not do stream hw_params() before PAUSE_RELEASE. do you think we need to send the hw_params IPC before restarting the DMA after PAUSE_RELEASE? 
We do not seem to have any issues with playback though. So I'm not sure whats missing here. @kv2019i any thoughts?

tlauda, 2019-06-25:
@ranj063 We don't need to set hw_params() before PAUSE_RELEASE. Maybe you could debug and see if the correct capture stream is running? It's weird that it's working for playback.

kv2019i, 2019-06-25:
> @ranj063 We don't need to set hw_params() before PAUSE_RELEASE. 

Ack on that... setting hw_params() again would be against ALSA API as the DMA pointers would be reset.

ranj063, 2019-06-25:
thanks @kv2019i . @tlauda ack. let me debug a bit more today and get back with details about whether the  stream info has changed during pause_release

ranj063, 2019-06-25:
@tlauda i can confirm that the host DMA is restarted during PAUSE_RELEASE for the same stream capture stream that was started.

ranj063, 2019-06-26:
> @ranj063 We don't need to set hw_params() before PAUSE_RELEASE. Maybe you could debug and see if the correct capture stream is running? It's weird that it's working for playback.

@tlauda its not just weird that it is working for playback, it is also working with arecord on APL

tlauda, 2019-06-26:
@ranj063 Are you telling me that it only fails for GLK and not for APL? Are the kernel and FW the same for both platforms?

ranj063, 2019-06-26:
> @ranj063 Are you telling me that it only fails for GLK and not for APL? Are the kernel and FW you are using the same for both platforms?

@tlauda yes. The kernel, FW and topology are the same in fact as I'm testing the nocodec mode.

keqiaozhang, 2019-06-27:
I can reproduce this issue on WHL-I2S, the reproduction rate is about 10%.

tlauda, 2019-06-27:
@keqiaozhang Please attach FW logs.

keqiaozhang, 2019-06-27:
@tlauda 

**Trace log on WHL**
```
    0      2         PIPE 2.11    159919282.968750         5.833333 src/audio/pipeline.c:378    pipeline_prepare()
    0      2          DAI 2.10    159919287.968750         5.000000      src/audio/dai.c:407    dai_prepare()
    0      2         PIPE 2.11    159919293.385417         5.416667 src/audio/pipeline.c:604    pipeline_trigger()
    0      2          DAI 2.10    159919298.177083         4.791667      src/audio/dai.c:470    dai_comp_trigger(), command = 1
    0      2          DAI 2.10    159919302.708333         4.531250      src/audio/dai.c:481    dai_comp_trigger(), START
    0      2          SSP         159919307.291667         4.583333 ers/intel/cavs/ssp.c:798    ssp_trigger() cmd 1
    0      2          SSP         159919312.760417         5.468750 ers/intel/cavs/ssp.c:735    ssp_start()
    0      1          DAI 2.10    159924442.604167      5129.843750      src/audio/dai.c:567    dai_check_for_xrun() error: overrun due to no data available
    0      1         COMP         159924449.583333         6.979167 of/audio/component.h:748    comp_overrun() error: dev->comp.id = 10, sink->free = 256, copy_bytes = 0
    0      2         PIPE 2.11    159924457.083333         7.500000 src/audio/pipeline.c:604    pipeline_trigger()
    0      2          DAI 2.10    159924462.343750         5.260417      src/audio/dai.c:470    dai_comp_trigger(), command = 8
    0      2          DAI 2.10    159924467.031250         4.687500      src/audio/dai.c:528    dai_comp_trigger(), XRUN
    0      2          DAI 2.10    159924471.458333         4.427083      src/audio/dai.c:534    dai_comp_trigger(), PAUSE/STOP
    0      2          DMA         159924475.989583         4.531250 src/drivers/dw/dma.c:442    dw_dma_stop(): dma 0 channel 1 stop
    0      2          SSP         159924485.364583         9.375000 ers/intel/cavs/ssp.c:798    ssp_trigger() cmd 0
    0      2          SSP         159924546.406250        61.041668 ers/intel/cavs/ssp.c:769    ssp_stop(), RX stop
    0      2          SSP         159924551.770833         5.364583 ers/intel/cavs/ssp.c:788    ssp_stop(), SSP port disabled
    0      1         PIPE         159924559.322917         7.552083 src/audio/pipeline.c:781    pipeline_copy() error: ret = -61, start->comp.id = 10, dir = 0
    0      1         PIPE 2.11    159924566.822917         7.500000 src/audio/pipeline.c:895    pipeline_xrun_recover()
```
[logger-whl.log](https://github.com/thesofproject/sof/files/3334120/logger-whl.log)

Jiangxinx, 2019-06-27:
The reproduction rate is 100% on BYT platform.

***sof-logger***
```
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      1          DAI 2.17    190302175.208333 190302176.000000      src/audio/dai.c:567    dai_check_for_xrun() error: overrun due to no data available
    0      1         COMP         190302182.083333         6.875000 of/audio/component.h:728    comp_overrun() error: dev->comp.id = 17, sink->free = 0, copy_bytes = 0
    0      1         PIPE         190302217.031250        34.947918 src/audio/pipeline.c:781    pipeline_copy() error: ret = -61, start->comp.id = 17, dir = 0
    0      1         PIPE 2.18    190302227.708333        10.677083 src/audio/pipeline.c:895    pipeline_xrun_recover()
```

***log file***
[logger.log](https://github.com/thesofproject/sof/files/3334123/logger.log)

tlauda, 2019-06-27:
@keqiaozhang Attached logger-whl.log is empty.

keqiaozhang, 2019-06-27:
Sorry,  trace log is updated.
[logger-whl.log](https://github.com/thesofproject/sof/files/3334190/logger-whl.log)

tlauda, 2019-06-27:
@keqiaozhang Could you try to get logs from the start? Here we are in xrun loop and I'm not sure what caused the initial error.

tlauda, 2019-06-27:
@mengdonglin @Jiangxinx @keqiaozhang @ranj063 
1. BYT - fixed in #1597 (please verify).
2. WHL - need logs from the start to know what exactly has caused this xrun loop.
3. GLK - debug in progress.

keqiaozhang, 2019-06-28:
> @keqiaozhang Could you try to get logs from the start? Here we are in xrun loop and I'm not sure what caused the initial error.

@tlauda 
This is all trace output, looks like it overflowed. I will try to disable the XRUN recovery to capture the first XRUN.

keqiaozhang, 2019-06-28:
@tlauda 
The full trace log is attached. be noticed that the xrun recovery is disabled by 
```
diff --git a/src/include/sof/audio/pipeline.h b/src/include/sof/audio/pipeline.h
index 9eaf9f227e41..78dddf68ef70 100644
--- a/src/include/sof/audio/pipeline.h
+++ b/src/include/sof/audio/pipeline.h
@@ -24,7 +24,7 @@
  * It should remain enabled in the situation when the
  * recovery is delegated to the outside of firmware.
  */
-#define NO_XRUN_RECOVERY 0
+#define NO_XRUN_RECOVERY 1

 /* pipeline tracing */
 #define trace_pipe(format, ...) \
```
***Trace***
```
    0      2          DMA          75598870.781250         7.343750 src/drivers/dw/dma.c:367    dw_dma_release(): dma 0 channel 1 release
    0      2          SSP          75598878.750000         7.968750 ers/intel/cavs/ssp.c:798    ssp_trigger() cmd 3
    0      2          SSP          75598884.270833         5.520833 ers/intel/cavs/ssp.c:735    ssp_start()
    0      1          DAI 2.10     75809015.208333    210130.937500      src/audio/dai.c:567    dai_check_for_xrun() error: overrun due to no data available
    0      1         COMP          75809022.447917         7.239583 of/audio/component.h:748    comp_overrun() error: dev->comp.id = 10, sink->free = 256, copy_bytes = 0
    0      2         PIPE 2.11     75809029.843750         7.395833 src/audio/pipeline.c:604    pipeline_trigger()
    0      2          DAI 2.10     75809034.895833         5.052083      src/audio/dai.c:470    dai_comp_trigger(), command = 8
    0      2          DAI 2.10     75809039.791667         4.895833      src/audio/dai.c:528    dai_comp_trigger(), XRUN
    0      2          DAI 2.10     75809044.010417         4.218750      src/audio/dai.c:534    dai_comp_trigger(), PAUSE/STOP
    0      2          DMA          75809048.437500         4.427083 src/drivers/dw/dma.c:442    dw_dma_stop(): dma 0 channel 1 stop
    0      2          SSP          75809058.437500        10.000000 ers/intel/cavs/ssp.c:798    ssp_trigger() cmd 0
    0      2          SSP          75809119.322917        60.885418 ers/intel/cavs/ssp.c:769    ssp_stop(), RX stop
    0      2          SSP          75809124.739583         5.416667 ers/intel/cavs/ssp.c:788    ssp_stop(), SSP port disabled
    0      1         PIPE          75809132.447917         7.708333 src/audio/pipeline.c:781    pipeline_copy() error: ret = -61, start->comp.id = 10, dir = 0
    0      1         PIPE 2.11     75809140.156250         7.708333 src/audio/pipeline.c:973    pipeline_task(): xrun recover failed! pipeline will be stopped!
    0      2          IPC          77223881.822917   1414741.625000    src/ipc/handler.c:439    ipc: comp 6 -> trigger cmd 0x60000
    0      2         PIPE 2.11     77223889.895833         8.072917 src/audio/pipeline.c:604    pipeline_trigger()
    0      2         HOST          77223894.791667         4.895833     src/audio/host.c:263    host_trigger()
    0      2       VOLUME          77223900.052083         5.260417 udio/volume/volume.c:521    volume_trigger()
    0      2          DAI 2.10     77223905.104167         5.052083      src/audio/dai.c:470    dai_comp_trigger(), command = 2
    0      1         COMP          77223909.947917         4.843750 rc/audio/component.c:152    comp_set_state() error: wrong state = 1, COMP_TRIGGER_PAUSE
    0      1          IPC          77223916.979167         7.031250 src/audio/pipeline.c:631    pipeline_trigger() error: ret = -22, host->comp.id = 6, cmd = 2
    0      1          IPC          77223924.687500         7.708333    src/ipc/handler.c:473    ipc: comp 6 trigger 0x60000 failed -22
```
[logger-whl-no-xrun-recovery.log](https://github.com/thesofproject/sof/files/3337699/logger-whl-no-xrun-recovery.log)

tlauda, 2019-06-28:
@keqiaozhang Thanks! How many pauses do you need to do to make it reproduce?

keyonjie, 2019-06-28:
good, @keqiaozhang, maybe worth to add some log to check if the channel is really opened at this case, something like:
```
diff --git a/src/drivers/dw/dma.c b/src/drivers/dw/dma.c
index 07c22e2b..84cbc83e 100644
--- a/src/drivers/dw/dma.c
+++ b/src/drivers/dw/dma.c
@@ -1285,6 +1285,11 @@ static int dw_dma_avail_data_size(struct dma *dma, unsigned int channel)
        if (size < 0)
                size += chan->ptr_data.buffer_bytes;
 
+       if (!size)
+               trace_dwdma("dw_dma_avail_data_size() "
+                           "size is 0! Channel enable = %d.",
+                           (dw_read(dma, DW_DMA_CHAN_EN) &
+                            DW_CHAN(channel)) ? 1 : 0);
        return size;
 }

```

keqiaozhang, 2019-07-01:
@keyonjie  @tlauda 
I retested this bug with latest top master branch, the reproduction rate is < 1%.
here is the trace log after applying above patch(run recovery is also disabled).
***logger***：
 ```
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      1          DAI 2.10    741098556.927083 741098560.000000      src/audio/dai.c:567    dai_check_for_xrun() error: overrun due to no data available
    0      1         COMP         741098563.125000         6.197917 of/audio/component.h:735    comp_overrun() error: dev->comp.id = 10, sink->free = 512, copy_bytes = 0
    0      1         PIPE         741098673.489583       110.364586 src/audio/pipeline.c:784    pipeline_copy() error: ret = -61, start->comp.id = 10, dir = 0
    0      1         PIPE 2.11    741098681.145833         7.656250 src/audio/pipeline.c:976    pipeline_task(): xrun recover failed! pipeline will be stopped!
    0      1         COMP         741774154.270833    675473.125000 rc/audio/component.c:152    comp_set_state() error: wrong state = 1, COMP_TRIGGER_PAUSE
    0      1          IPC         741774161.354167         7.083333 src/audio/pipeline.c:634    pipeline_trigger() error: ret = -22, host->comp.id = 6, cmd = 2
    0      1          IPC         741774169.010417         7.656250    src/ipc/handler.c:482    ipc: comp 6 trigger 0x60000 failed -22
    0      1         COMP         741789843.177083     15674.166992 rc/audio/component.c:126    comp_set_state() error: wrong state = 1, COMP_TRIGGER_RELEASE
    0      1          IPC         741789850.156250         6.979167 src/audio/pipeline.c:634    pipeline_trigger() error: ret = -22, host->comp.id = 6, cmd = 3
    0      1          IPC         741789857.552083         7.395833    src/ipc/handler.c:482    ipc: comp 6 trigger 0x70000 failed -22
    0      1         COMP         742564133.645833    774276.125000 rc/audio/component.c:152    comp_set_state() error: wrong state = 1, COMP_TRIGGER_PAUSE
    0      1          IPC         742564140.729167         7.083333 src/audio/pipeline.c:634    pipeline_trigger() error: ret = -22, host->comp.id = 6, cmd = 2
    0      1          IPC         742564148.020833         7.291667    src/ipc/handler.c:482    ipc: comp 6 trigger 0x60000 failed -22
    0      1         COMP         742579930.156250     15782.135742 rc/audio/component.c:126    comp_set_state() error: wrong state = 1, COMP_TRIGGER_RELEASE
    0      1          IPC         742579937.187500         7.031250 src/audio/pipeline.c:634    pipeline_trigger() error: ret = -22, host->comp.id = 6, cmd = 3
    0      1          IPC         742579944.635417         7.447917    src/ipc/handler.c:482    ipc: comp 6 trigger 0x70000 failed -22
    0      1         COMP         743229882.812500    649938.187500 rc/audio/component.c:162    comp_set_state() error: wrong state = 5, COMP_TRIGGER_RESET
```
***logger -t***
```
    0      2          SSP         740589663.645833         7.656250 ers/intel/cavs/ssp.c:798    ssp_trigger() cmd 3
    0      2          SSP         740589669.166667         5.520833 ers/intel/cavs/ssp.c:735    ssp_start()
    0      2          DMA         741098550.729167    508881.562500 src/drivers/dw/dma.c:1292   dw_dma_avail_data_size() size is 0! Channel enable = 1.
    0      1          DAI 2.10    741098556.927083         6.197917      src/audio/dai.c:567    dai_check_for_xrun() error: overrun due to no data available
    0      1         COMP         741098563.125000         6.197917 of/audio/component.h:735    comp_overrun() error: dev->comp.id = 10, sink->free = 512, copy_bytes = 0
    0      2         PIPE 2.11    741098570.625000         7.500000 src/audio/pipeline.c:607    pipeline_trigger()
    0      2          DAI 2.10    741098575.677083         5.052083      src/audio/dai.c:470    dai_comp_trigger(), command = 8
    0      2          DAI 2.10    741098580.468750         4.791667      src/audio/dai.c:528    dai_comp_trigger(), XRUN
    0      2          DAI 2.10    741098584.687500         4.218750      src/audio/dai.c:534    dai_comp_trigger(), PAUSE/STOP
    0      2          DMA         741098589.375000         4.687500 src/drivers/dw/dma.c:442    dw_dma_stop(): dma 0 channel 1 stop
    0      2          SSP         741098599.166667         9.791667 ers/intel/cavs/ssp.c:798    ssp_trigger() cmd 0
    0      2          SSP         741098660.208333        61.041668 ers/intel/cavs/ssp.c:769    ssp_stop(), RX stop
    0      2          SSP         741098665.781250         5.572917 ers/intel/cavs/ssp.c:788    ssp_stop(), SSP port disabled
    0      1         PIPE         741098673.489583         7.708333 src/audio/pipeline.c:784    pipeline_copy() error: ret = -61, start->comp.id = 10, dir = 0
    0      1         PIPE 2.11    741098681.145833         7.656250 src/audio/pipeline.c:976    pipeline_task(): xrun recover failed! pipeline will be stopped!
    0      2          IPC         741774125.885417    675444.750000    src/ipc/handler.c:448    ipc: comp 6 -> trigger cmd 0x60000
    0      2         PIPE 2.11    741774133.854167         7.968750 src/audio/pipeline.c:607    pipeline_trigger()
    0      2         HOST         741774139.010417         5.156250     src/audio/host.c:286    host_trigger()
    0      2       VOLUME         741774144.375000         5.364583 udio/volume/volume.c:521    volume_trigger()
[... cut ...]
    0      2         PIPE 2.11    742564113.645833        17.500000 src/audio/pipeline.c:607    pipeline_trigger()
    0      2         HOST         742564118.645833         5.000000     src/audio/host.c:286    host_trigger()
    0      2       VOLUME         742564123.958333         5.312500 udio/volume/volume.c:521    volume_trigger()
    0      2          DAI 2.10    742564129.010417         5.052083      src/audio/dai.c:470    dai_comp_trigger(), command = 2
```
[xrun-logger-t.log](https://github.com/thesofproject/sof/files/3344168/xrun-logger-t.log)

tlauda, 2019-07-01:
@ranj063 For GLK and WHL please verify if deasserting L1SEN bit helps.

ranj063, 2019-07-01:
@tlauda we can confirm disabling L1SEN fixes the problem on GLK. But given that this issue is not reproducible on other platforms, how should we go about doing this? @lgirdwood what do you think?

jajanusz, 2019-07-01:
@ranj063 As @keqiaozhang wrote there is low repro rate on WHL. Maybe he can stress test it.

ranj063, 2019-07-01:
@jajanusz sure, I dont have a WHL with I2S reworks to test at my end. I'll work with @keqiaozhang tonight to test it.

But I wonder if this issue is limited to I2S DAI's. We havent seen it in any of the HDA DAI capture scenarios. I am wondering what would be the correct fix. 
1. To disable L1SEN for all platforms or
2. To disable L1SEN for all platforms with I2S DAI's

YvonneYang2, 2019-07-08:
Test on GLK with test recipe as below, I/O error caused by "ctrl + z / fg" is not reproduced. Pause/Resume can still reproduce the bug.

Test recipe:
Firmware: 27cdfbdb https://github.com/thesofproject/sof/commits/master
Kernel: fdfcca4c https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-glk-da7219.tplg same as firmware
platform： GLK

[pause_dmesg.log](https://github.com/thesofproject/sof/files/3366550/pause_dmesg.log)
[pause_logger.log](https://github.com/thesofproject/sof/files/3366551/pause_logger.log)

tlauda, 2019-07-08:
@ClarexZhou Temporary workaround for this issue is https://github.com/thesofproject/linux/pull/1070.

YvonneYang2, 2019-07-08:
Check on GLK with https://github.com/thesofproject/linux/pull/1070, issue not reproduced.

Test recipe:
Firmware: 27cdfbd https://github.com/thesofproject/sof/commits/master
Kernel: fdfcca4c https://github.com/thesofproject/linux/commits/topic/sof-dev + https://github.com/thesofproject/linux/pull/1070 
Topology: file: tools/topology/sof-glk-da7219.tplg same as firmware
platform： GLK

mengdonglin, 2019-07-08:
@Jiangxinx @keqiaozhang Could you please check other platforms can reproduce this issue with thesofproject/linux#1070?

Jiangxinx, 2019-07-09:
I can still reproduce this issue with  thesofproject/linux#1070 on BYT platform.
***Test recipe:***
Firmware: 27cdfbd https://github.com/thesofproject/sof/commits/master
Kernel: 1da8b5fa https://github.com/thesofproject/linux/commits/topic/sof-dev + thesofproject/linux#1070
Topology file: sof-byt-rt5651.tplg & sof-byt-nocodec.tplg
platform：BYT

***Output***
```
ubuntu@cd-byt-mb-nocodec-04:~$ arecord -Dhw:0,0 -c2 -fS16_LE 1.wav -vv -i
Recording WAVE '1.wav' : Signed 16 bit Little Endian, Rate 8000 Hz, Stereo
Warning: rate is not accurate (requested = 8000Hz, got = 48000Hz)
         please, try the plug plugin 
Hardware PCM card 0 'sof-nocodec' device 0 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16320
  period_size  : 4080
  period_time  : 85000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4080
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 16320
  silence_threshold: 0
  silence_size : 0
  boundary     : 9187343239835811840
  appl_ptr     : 0
  hw_ptr       : 0
                arecord: pcm_read:2103: read error: Input/output error
```

***sof-logger***
```
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DAI 2.16    105743464.895833 105743464.000000      src/audio/dai.c:567 	dai_check_for_xrun() error: overrun due to no data available
    0      1         COMP         105743471.666667         6.770833 of/audio/component.h:735 	comp_overrun() error: dev->comp.id = 16, sink->free = 768, copy_bytes = 0
    0      1         PIPE         105743505.989583        34.322918 src/audio/pipeline.c:784 	pipeline_copy() error: ret = -61, start->comp.id = 16, dir = 0
    0      1         PIPE 2.17    105743516.614583        10.625000 src/audio/pipeline.c:976 	pipeline_task(): xrun recover failed! pipeline will be stopped!
    0      1         COMP         106248529.635417    505013.031250 rc/audio/component.c:162 	comp_set_state() error: wrong state = 5, COMP_TRIGGER_RESET
```

ranj063, 2019-07-09:
@Jiangxinx didnt you verify that https://github.com/thesofproject/sof/pull/1597 fixes this issue for BYT?

Jiangxinx, 2019-07-10:
@ranj063 I have verified #1597 and can fix this issue,but it reproduced again from 20190706:
Firmware: 27cdfbdb https://github.com/thesofproject/sof/commits/master
Kernel: fdfcca4c https://github.com/thesofproject/linux/commits/topic/sof-dev

The previous version (20190705) can work fine:
Firmware: fcd72767 https://github.com/thesofproject/sof/commits/master
Kernel: 86b820f2  https://github.com/thesofproject/linux/commits/topic/sof-dev

After my bisect test, I think it was regressed by PR#1605

keyonjie, 2019-07-10:
@Jiangxinx what #1605 have done is only disable Xrun recovery, so in this test case, even you revert #1605, you should still see errors like this in sof-logger:
```
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DAI 2.16    105743464.895833 105743464.000000      src/audio/dai.c:567 	dai_check_for_xrun() error: overrun due to no data available
    0      1         COMP         105743471.666667         6.770833 of/audio/component.h:735 	comp_overrun() error: dev->comp.id = 16, sink->free = 768, copy_bytes = 0
```

Jiangxinx, 2019-07-10:
@keyonjie I double checked  #1605, sof-logger can still occur errors after revert #1605 , but capture will work normally, not see any Input/output error.

keyonjie, 2019-07-10:
> @keyonjie I double checked #1605, sof-logger can still occur errors after revert #1605 , but capture will work normally, not see any Input/output error.

@Jiangxinx Thanks, that means Xrun still happened at pause-release, but the xrun recovery can help to recover it at this case.

keqiaozhang, 2019-07-23:
This issue is not reproducible on WHL-I2S after applying thesofproject/linux#1070.

mengdonglin, 2019-07-29:
@Jiangxinx Please help to check if https://github.com/thesofproject/linux/pull/1070 can fix this issue on Baytrail.

tlauda, 2019-07-29:
@mengdonglin It can't. Problems on byt and cAVS platforms are totally unrelated.

YvonneYang2, 2019-07-31:
As comment https://github.com/thesofproject/sof/issues/1578#issuecomment-509135444 said, thesofproject/linux#1070 can fix the issue on glk.

mengdonglin, 2019-08-05:
@tlauda Could you please analyze #1597 and #1605 and share some insights why it seems #1597 can fix this issue and #1605 introduced it again?
Please see validation comments: https://github.com/thesofproject/sof/issues/1578#issuecomment-509878507

tlauda, 2019-08-08:
@mengdonglin @YvonneYang2 Fixed on BYT with #1715. GLK debug still in progress.

Jiangxinx, 2019-08-09:
Can not be reproduced with  #1715 on BYT platform.

mengdonglin, 2019-08-10:
@tlauda thank you!  Please notify us when your new https://github.com/thesofproject/sof/pull/1722 is merged.

Jiangxinx, 2019-08-21:
Can not be reproduced with #1722 on BYT rt5682 platform.

YvonneYang2, 2019-08-26:
Can still reproduce the bug on glk platform with #1722

[4 more replies]

closed 2019-09-27 by plbossart