thesofproject/sof #1591: [FEATURE] Allow DMIC to accept 24/32 fifo bits config

wwittbrx, 2019-06-26:
**Is your feature request related to a problem? Please describe.**

This feature is required for keyword detection pipelines with 24/32 format. Currently when trying to configure DMIC this way we get error: fifo_bits EINVAL.

**Describe the solution you'd like**

Add support for configuring DMIC for 24/32 format capture.

singalsu, 2019-06-27:
The DMIC HW supports only S16_LE and S32_LE compatible data formats so the conversion to S24_4LE would need to happen in FW.

lgirdwood, 2019-06-29:
@singalsu @wwittbrx can be done by adding volume in the topology. i.e. 32bit DMIC with volume converting 32 -> 24.

tlauda, 2019-07-06:
Currently it will be supported by adding volume and later on we will do pcm conversion in some other component (after converter and volume separation).

closed 2019-07-06 by tlauda