thesofproject/sof #1593: [BUG][BYT] Input/output error when capture a audio for 3 minutes via HDA Analog on byt-rt5651 and nocodec platform.

Jiangxinx, 2019-06-27:
**Describe the bug**
capture a audio for 3 minutes via HDA Analog on byt-rt5651 and nocodec platform :
overrun!!! (at least 140.888 ms long)
arecord: pcm_read:2103: read error: Input/output error

**To Reproduce**
arecord -Dhw:0,0 -fS16_LE -c2 -r48000 test.wav -vv -d 180

**Reproduction Rate**
100%

**behavior**
```
Recording WAVE '/home/ubuntu/sofcitest/./out/sanity-test/capture-3mins.sh/record-hw:0,0-c2-r48000-fs16_le-d180-1561594575.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
HW Params of device "hw:0,0":
--------------------
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S16_LE S24_LE S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: [16 32]
FRAME_BITS: [32 64]
CHANNELS: 2
RATE: 48000
PERIOD_TIME: [500 85000]
PERIOD_SIZE: [24 4080]
PERIOD_BYTES: [192 16320]
PERIODS: [2 4]
BUFFER_TIME: [1000 340000]
BUFFER_SIZE: [48 16320]
BUFFER_BYTES: [192 65472]
TICK_TIME: ALL
--------------------
overrun!!! (at least 140.888 ms long)
arecord: pcm_read:2103: read error: Input/output error
```

**sof-logger**
```
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      1          DMA         124192735.052083 124192736.000000 src/drivers/dw/dma.c:291    dw_dma_start() error: dma 1 channel 2 not ready ena 0x6 status 0x3
    0      1         HOST         124192747.395833        12.343750     src/audio/host.c:469    host_buffer_cb() error: dma_copy() failed, ret = 4294967280
    0      1          DAI 2.17    124195694.583333      2947.187500      src/audio/dai.c:567    dai_check_for_xrun() error: overrun due to no data available
    0      1         COMP         124195701.458333         6.875000 of/audio/component.h:728    comp_overrun() error: dev->comp.id = 17, sink->free = 0, copy_bytes = 0
    0      1         PIPE         124195736.458333        35.000000 src/audio/pipeline.c:781    pipeline_copy() error: ret = -61, start->comp.id = 17, dir = 0
    0      1         PIPE 2.18    124195747.083333        10.625000 src/audio/pipeline.c:895    pipeline_xrun_recover()
    0      1         WAIT         124696051.041667    500303.968750       src/lib/wait.c:96     ewt
    0      1          DMA         124696058.072917         7.031250 src/drivers/dw/dma.c:460    dw_dma_stop() error: dma 1 channel 2 timeout
```

***bisect result***
Caused by master (#1587).

**Environment**
Firmware: fb105575 https://github.com/thesofproject/sof/commits/master
Kernel: 0087d4af https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: sof-byt-rt5651.tplg & sof-byt-nocodec.tplg

**log file**
[logger.log](https://github.com/thesofproject/sof/files/3332902/logger.log)

mengdonglin, 2019-06-27:
@Jiangxinx BYT has no HD-A analog support in HW level. Seems to be a typo. You're using I2S analog in no codec mode, right?

Jiangxinx, 2019-06-27:
@mengdonglin sorry, i have changed the title.

PR #1595 (2019-06-28): host: ignore host_trigger() for one shot copy mode
Ignores host_trigger(), when host component is
configured in one shot copy mode. In such mode
the DMA is started every time in copy and stopped
automatically.

Fixes #1593.
closed 2019-06-28 by jajanusz