thesofproject/sof #1594: [BUG] Analog capture not work when do S3 stress test with headset connected and sound setting opened.

ClarexZhou, 2019-06-27:
**Describe the bug**
Headset capture not work anymore when suspend & wake up stress test with sound setting opened. 

**To Reproduce**
1. Headset is connected
2. Open Settings->Sound, select Input table.
3. 'sudo rtcwake -m mem -s 30' to enter S3 and wake up.
4.  speaker to headset mic, check 'Input level' for headset in sound setting works
5. Repeat step3 and step 4 for more than 30 times.

**Reproduction Rate**
1/30 

**Expected behavior**
Headset capture should always works after wake up

**Impact**
Headset capture not work, zero data is recorded. Dmesg keeps out ipc rx: 0x90020000: GLB_TRACE_MSG  sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG repeatly.

**Environment**
FW: 6cc8da10 https://github.com/thesofproject/sof/tree/cml-hda-dmic-001-drop-stable
Kernel: https://github.com/thesofproject/linux/tree/release/sof-v5.0 (top commit: 1cd1a89)  + PR1060 
TPLG: sof-hda-generic-2ch.tplg, same branch with FW.

Dmesg:
```
Jun 26 05:45:02 u kernel: [ 1347.721952] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
Jun 26 05:45:02 u kernel: [ 1347.721960] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
Jun 26 05:45:02 u kernel: [ 1347.722184] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
Jun 26 05:45:02 u kernel: [ 1347.722195] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
Jun 26 05:45:02 u kernel: [ 1347.722392] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
Jun 26 05:45:02 u kernel: [ 1347.722401] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40020000: GLB_PM_MSG: CTX_RESTORE
Jun 26 05:45:02 u kernel: [ 1347.722517] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40020000: GLB_PM_MSG: CTX_RESTORE
Jun 26 05:45:02 u kernel: [ 1347.722603] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 5:0
Jun 26 05:45:02 u kernel: [ 1347.722637] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 6:0
Jun 26 05:45:02 u kernel: [ 1347.722649] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 7:0
Jun 26 05:45:02 u kernel: [ 1347.826309] usb 1-6: reset high-speed USB device number 2 using xhci_hcd
Jun 26 05:45:02 u kernel: [ 1347.856298] nvme nvme0: 8/0/0 default/read/poll queues
Jun 26 05:45:02 u kernel: [ 1347.909539] ata1: SATA link down (SStatus 4 SControl 300)
Jun 26 05:45:02 u kernel: [ 1348.221503] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
Jun 26 05:45:02 u kernel: [ 1348.221527] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
Jun 26 05:45:02 u kernel: [ 1348.721745] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
Jun 26 05:45:02 u kernel: [ 1348.721779] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
Jun 26 05:45:02 u kernel: [ 1349.327973] acpi LNXPOWER:04: Turning OFF
Jun 26 05:45:02 u kernel: [ 1349.328132] OOM killer enabled.
Jun 26 05:45:02 u kernel: [ 1349.328134] Restarting tasks ... 
Jun 26 05:45:02 u kernel: [ 1349.331061] sof-audio-pci 0000:00:1f.3: hda: prepare stream dir 1
Jun 26 05:45:02 u kernel: [ 1349.331075] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
Jun 26 05:45:02 u kernel: [ 1349.331211] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
Jun 26 05:45:02 u kernel: [ 1349.331231] sof-audio-pci 0000:00:1f.3: format_val=49, rate=48000, ch=2, format=10
Jun 26 05:45:02 u kernel: [ 1349.331268] sof-audio-pci 0000:00:1f.3: pcm: prepare stream 0 dir 1
Jun 26 05:45:02 u kernel: [ 1349.331270] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 0 dir 1
Jun 26 05:45:02 u kernel: [ 1349.331292] sof-audio-pci 0000:00:1f.3: period_bytes:0x3fc0
Jun 26 05:45:02 u kernel: [ 1349.331293] sof-audio-pci 0000:00:1f.3: periods:4
Jun 26 05:45:02 u kernel: [ 1349.331305] sof-audio-pci 0000:00:1f.3: stream_tag 3
Jun 26 05:45:02 u kernel: [ 1349.331311] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
Jun 26 05:45:02 u kernel: [ 1349.331471] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
Jun 26 05:45:02 u kernel: [ 1349.331474] sof-audio-pci 0000:00:1f.3: pcm: stream dir 1, posn mailbox offset is 790528
Jun 26 05:45:02 u kernel: [ 1349.331519] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 1
Jun 26 05:45:02 u kernel: [ 1349.331525] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
Jun 26 05:45:02 u kernel: [ 1349.331974] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
Jun 26 05:45:02 u kernel: [ 1349.331979] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=1
Jun 26 05:45:02 u kernel: [ 1349.332000] done.
Jun 26 05:45:02 u kernel: [ 1349.347054] thermal thermal_zone4: failed to read out thermal zone (-61)
Jun 26 05:45:02 u systemd-sleep[2668]: System resumed.
Jun 26 05:45:02 u kernel: [ 1349.369124] PM: suspend exit
Jun 26 05:45:02 u systemd[1]: Started Suspend.
Jun 26 05:45:02 u systemd[1]: sleep.target: Unit not needed anymore. Stopping.
Jun 26 05:45:02 u systemd[1]: Stopped target Sleep.
Jun 26 05:45:02 u systemd[1]: tlp-sleep.service: Unit not needed anymore. Stopping.
[... 140 lines cut ...]
    0      2          DAI 2.13    552001721.718750         4.531250      src/audio/dai.c:447 	dai_prepare()
    0      2         PIPE 2.14    552001726.614583         4.895833 src/audio/pipeline.c:631 	pipeline_trigger()
    0      2          DAI 2.13    552001731.093750         4.479167      src/audio/dai.c:510 	dai_comp_trigger(), command = 1
    0      2          DAI 2.13    552001735.312500         4.218750      src/audio/dai.c:521 	dai_comp_trigger(), START
    0      2          DMA         552002905.677083      1170.364624 intel/cavs/hda-dma.c:555 	hda-dmac: 6 channel 1 -> start
    0      2          DMA         552002910.520833         4.843750 intel/cavs/hda-dma.c:391 	hda-dmac: 6 channel 1 -> enable
    0      2       BUFFER         552004157.447917      1246.927124   src/audio/buffer.c:118 	comp_update_buffer_produce(), no bytes to produce
    0      1         COMP         552004163.177083         5.729167 rc/audio/component.c:252 	comp_get_copy_limits() error: source component buffer has not enough data available
    0      2         COMP         552004169.010417         5.833333 of/audio/component.h:731 	comp_underrun(), ((dev->comp.id << 16) | source->avail) = 655360, ((min_bytes << 16) | copy_bytes) = 0
    0      2         PIPE 2.14    552004173.229167         4.218750 src/audio/pipeline.c:631 	pipeline_trigger()
    0      2          DAI 2.13    552004177.760417         4.531250      src/audio/dai.c:510 	dai_comp_trigger(), command = 8
    0      2          DAI 2.13    552004182.239583         4.479167      src/audio/dai.c:568 	dai_comp_trigger(), XRUN
    0      2          DAI 2.13    552004186.250000         4.010417      src/audio/dai.c:574 	dai_comp_trigger(), PAUSE/STOP
    0      2          DMA         552004190.677083         4.427083 intel/cavs/hda-dma.c:650 	hda-dmac: 6 channel 1 -> stop
    0      1       VOLUME         552004198.385417         7.708333 udio/volume/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         552004204.375000         5.989583 src/audio/pipeline.c:808 	pipeline_copy() error: ret = -5, start->comp.id = 13, dir = 0
    0      1         PIPE 2.14    552004211.354167         6.979167 src/audio/pipeline.c:920 	pipeline_xrun_recover()
    0      2         PIPE 2.14    552004217.083333         5.729167 src/audio/pipeline.c:402 	pipeline_prepare()
    0      2          DAI 2.13    552004221.614583         4.531250      src/audio/dai.c:447 	dai_prepare()
    0      2         PIPE 2.14    552004226.562500         4.947917 src/audio/pipeline.c:631 	pipeline_trigger()
    0      2          DAI 2.13    552004231.041667         4.479167      src/audio/dai.c:510 	dai_comp_trigger(), command = 1
```

[sof-logger-s3-mantis.log](https://github.com/thesofproject/sof/files/3334195/sof-logger-s3-mantis.log)
[mantis-s3-headset-fail-2.log](https://github.com/thesofproject/sof/files/3334196/mantis-s3-headset-fail-2.log)

RanderWang, 2019-06-28:
I reproduced the XRUN recovery case by modifying hda_dma_avail_data_size.
Just let hda_dma_avail_data_size always return 0 after capture works 10 seconds,
At this time, FW would recover XRUN and arecord just seems hang but not stop.
then do S3 and resume from S3, you will find so many XRUN recovery messages.

hda_dma_avail_data_size will only returns 0 when status & DGCS_BNE == 0

````
	status = host_dma_reg_read(chan->dma, chan->index, DGCS);

	if (status & DGCS_BF)
		return chan->buffer_bytes;

	if (!(status & DGCS_BNE))
		return 0;
````

RanderWang, 2019-06-28:
I also did some changes in codecs and sof kernel driver to generate infinite XRUN recovery, but I didn't get any XRUN.

(1) don't program  codec for capture stream when resuming from S3.  Codec will not send data to controller with this change. But no XRUN occurred.
 (2) don't program controller for capture stream when resuming from S3. Still no XRUN occurred.

Until now, our QA still cant reproduce the bug.

ClarexZhou, 2019-06-28:
Finally got bug reproduced.  Bug can be reproduced via below steps:
1.	Boot up with headset connected
2.	Open sound setting quickly after unit boot up
3.	Go to Input page, highlight headset, check capture not work. But no repeat messages send in dmesg. (same issue with #1055 Headset arecord not work when plug in headset as soon as wake up from S3 (2/12))
4.	Keep sound setting opened, enter S3 and wake up, check dmesg.   Repeated message keep flashing in dmesg issue is reproduced.

After step3, if close sound setting, then do S3,  the repeated message flashing issue not occurred. -> Open sound setting and highlight Headset input, then do S3 again, still the repeated message flashing issue not occurred.

tlauda, 2019-06-28:
@ClarexZhou Do you have FW logs?

ClarexZhou, 2019-07-01:
Attach logs.

sof-logger 
···
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1         PIPE 2.14     57460664.635417  57460664.000000 src/audio/pipeline.c:920 	pipeline_xrun_recover()
    0      1         COMP          57463117.447917      2452.812500 rc/audio/component.c:252 	comp_get_copy_limits() error: source component buffer has not enough data available
    0      1       VOLUME          57463151.979167        34.531250 udio/volume/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE          57463157.864583         5.885417 src/audio/pipeline.c:808 	pipeline_copy() error: ret = -5, start->comp.id = 13, dir = 0
    0      1         PIPE 2.14     57463164.895833         7.031250 src/audio/pipeline.c:920 	pipeline_xrun_recover()
    0      1         COMP          57465617.343750      2452.447998 rc/audio/component.c:252 	comp_get_copy_limits() error: source component buffer has not enough data available
    0      1       VOLUME          57465650.729167        33.385418 udio/volume/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE          57465656.614583         5.885417 src/audio/pipeline.c:808 	pipeline_copy() error: ret = -5, start->comp.id = 13, dir = 0
    0      1         PIPE 2.14     57465663.697917         7.083333 src/audio/pipeline.c:920 	pipeline_xrun_recover()
    0      1         COMP          57468117.760417      2454.062500 rc/audio/component.c:252 	comp_get_copy_limits() error: source component buffer has not enough data available
    0      1       VOLUME          57468153.489583        35.729168 udio/volume/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE          57468159.479167         5.989583 src/audio/pipeline.c:808 	pipeline_copy() error: ret = -5, start->comp.id = 13, dir = 0
    0      1         PIPE 2.14     57468166.406250         6.927083 src/audio/pipeline.c:920 	pipeline_xrun_recover()
    0      1         COMP          57470617.395833      2450.989502 rc/audio/component.c:252 	comp_get_copy_limits() error: source component buffer has not enough data available
    0      1       VOLUME          57470652.812500        35.416668 udio/volume/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE          57470658.802083         5.989583 src/audio/pipeline.c:808 	pipeline_copy() error: ret = -5, start->comp.id = 13, dir = 0
    0      1         PIPE 2.14     57470665.729167         6.927083 src/audio/pipeline.c:920 	pipeline_xrun_recover()
    0      1         COMP          57473117.812500      2452.083252 rc/audio/component.c:252 	comp_get_copy_limits() error: source component buffer has not enough data available
    0      1       VOLUME          57473153.020833        35.208332 udio/volume/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE          57473159.010417         5.989583 src/audio/pipeline.c:808 	pipeline_copy() error: ret = -5, start->comp.id = 13, dir = 0
    0      1         PIPE 2.14     57473166.093750         7.083333 src/audio/pipeline.c:920 	pipeline_xrun_recover()
    0      1         COMP          57475617.812
···

[sof-logger-t_afterstep4.log](https://github.com/thesofproject/sof/files/3343636/sof-logger-t_afterstep4.log)
[dmesgAfterstep4.log](https://github.com/thesofproject/sof/files/3343637/dmesgAfterstep4.log)
[sof-logger_afterstep4.log](https://github.com/thesofproject/sof/files/3343638/sof-logger_afterstep4.log)

tlauda, 2019-07-01:
@ClarexZhou Please get FW log with XRUN disabled as @keqiaozhang did it with different bug: https://github.com/thesofproject/sof/issues/1578#issuecomment-506611258.

RanderWang, 2019-07-01:
@tlauda  @jajanusz 
The FW log maybe not useful for you because it is full of XRUN. I have an idea: if XRUN recovery occurs infinite, can we just stop recovery after first continuous  recoveries ? So that we can get few log and it is more easy to get the first XRUN point.

For this case, we found that for a special operation,  capture didn't work before S3 and infinite XRUN happened after S3. Now I am working on why capture didn't work. I think you could wait for my finding and go on other work.

ClarexZhou, 2019-07-01:
The dmesg keeps flushing issue can't be reproduced after disabled xrun recovery.

sof-logger:
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1         COMP           1768107.395833   1768107.375000 rc/audio/component.c:252 	comp_get_copy_limits() error: source component buffer has not enough data available
    0      1       VOLUME           1768143.697917        36.302082 udio/volume/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE           1768149.947917         6.250000 src/audio/pipeline.c:808 	pipeline_copy() error: ret = -5, start->comp.id = 13, dir = 0
    0      1         PIPE 2.14      1768157.187500         7.239583 src/audio/pipeline.c:998 	pipeline_task(): xrun recover failed! pipeline will be stopped!
```

[sof-logger__xrunRecoveryDisbale.log](https://github.com/thesofproject/sof/files/3344641/sof-logger__xrunRecoveryDisbale.log)
[dmesg_xrunRecoveryDisbale_dmesg.log](https://github.com/thesofproject/sof/files/3344642/dmesg_xrunRecoveryDisbale_dmesg.log)
[sof-logger_all_xrunRecoveryDisbale.log](https://github.com/thesofproject/sof/files/3344643/sof-logger_all_xrunRecoveryDisbale.log)

jajanusz, 2019-07-01:
@RanderWang Looks more like kernel issue. FW is not spamming anything, you just fell into some loop on your side (maybe there are some interrupts that trigger it on your side but I doubt it). There are no messages generated (by FW) but you keep rereading them, can you check what causes it?

jajanusz, 2019-07-01:
Probably the same issue as: https://github.com/thesofproject/linux/issues/1064

RanderWang, 2019-07-03:
@jajanusz  @tlauda Thanks for your advice. My finding was changed in my debug progress and the final reason is: wrong link dma channel is set to FW after S3.

tlauda, 2019-07-03:
Kernel issue, fix in https://github.com/thesofproject/linux/pull/1072.

closed 2019-07-03 by tlauda