thesofproject/sof #1599: [BUG][BYT]Aplay Input/output error on BYT-RT5651 and nocodec platform

Jiangxinx, 2019-07-01:
**Describe the bug**
aplay: pcm_write:2011: write error: Input/output error  on BYT-RT5651 and nocodec platform

**To Reproduce**
aplay -Dhw:0,0 -fS16_LE -c2 -r 48000 /dev/zero -vv -i

**Reproduction Rate**
100%

**Screenshots or console output**
```
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 0 'sof-nocodec' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16368
  period_size  : 4080
  period_time  : 85000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4080
  period_event : 0
  start_threshold  : 16368
  stop_threshold   : 16368
  silence_threshold: 0
  silence_size : 0
  boundary     : 9214364837600034816
  appl_ptr     : 0
  hw_ptr       : 0
#+                                                 | 00%aplay: pcm_write:2011: write error: Input/output error
```

**sof-logger**
```
    0      2       BUFFER         1951560788.072917         3.020833   src/audio/buffer.c:94    comp_update_buffer_produce(), no bytes to produce
    0      2       BUFFER         1951560790.156250         2.083333   src/audio/buffer.c:168   comp_update_buffer_consume(), no bytes to consume
    0      2       BUFFER         1951561777.656250       987.500000   src/audio/buffer.c:94    comp_update_buffer_produce(), no bytes to produce
    0      2       BUFFER         1951561779.687500         2.031250   src/audio/buffer.c:168   comp_update_buffer_consume(), no bytes to consume
    0      2       BUFFER         1951561782.968750         3.281250   src/audio/buffer.c:168   comp_update_buffer_consume(), no bytes to consume
    0      2       BUFFER         1951561785.000000         2.031250   src/audio/buffer.c:94    comp_update_buffer_produce(), no bytes to produce
    0      2       BUFFER         1951561788.072917         3.072917   src/audio/buffer.c:94    comp_update_buffer_produce(), no bytes to produce
    0      2       BUFFER         1951561790.156250         2.083333   src/audio/buffer.c:168   comp_update_buffer_consume(), no bytes to consume
    0      2       BUFFER         1951562777.604167       987.447937   src/audio/buffer.c:94    comp_update_buffer_produce(), no bytes to produce
    0      2       BUFFER         1951562779.687500         2.083333   src/audio/buffer.c:168   comp_update_buffer_consume(), no bytes to consume
    0      2       BUFFER         1951562782.916667         3.229167   src/audio/buffer.c:168   comp_update_buffer_consume(), no bytes to consume
    0      2       BUFFER         1951562785.000000         2.083333   src/audio/buffer.c:94    comp_update_buffer_produce(), no bytes to produce
    0      2       BUFFER         1951562788.020833         3.020833   src/audio/buffer.c:94    comp_update_buffer_produce(), no bytes to produce
    0      2       BUFFER         1951562790.104167         2.083333   src/audio/buffer.c:168   comp_update_buffer_consume(), no bytes to consume
    0      2       BUFFER         1951563777.604167       987.500000   src/audio/buffer.c:94    comp_update_buffer_produce(), no bytes to produce
    0      2       BUFFER         1951563779.687500         2.083333   src/audio/buffer.c:168   comp_update_buffer_consume(), no bytes to consume
    0      2       BUFFER         1951563782.968750         3.281250   src/audio/buffer.c:168   comp_update_buffer_consume(), no bytes to consume
    0      2       BUFFER         1951563785.000000         2.031250   src/audio/buffer.c:94    comp_update_buffer_produce(), no bytes to produce
    0      2       BUFFER         1951563788.072917         3.072917   src/audio/buffer.c:94    comp_update_buffer_produce(), no bytes to produce
    0      2       BUFFER         1951563790.156250         2.083333   src/audio/buffer.c:168   comp_update_buffer_consume(), no bytes to consume
    0      2          IPC         1951564576.718750       786.562500    src/ipc/handler.c:448   ipc: comp 1 -> trigger cmd 0x50000
    0      2         PIPE 1.10    1951564578.854167         2.135417 src/audio/pipeline.c:607   pipeline_trigger()
    0      2         HOST         1951564581.041667         2.187500     src/audio/host.c:286   host_trigger()
    0      2       VOLUME         1951564583.645833         2.604167 udio/volume/volume.c:521   volume_trigger()
    0      2        MIXER         1951564586.250000         2.604167    src/audio/mixer.c:204   mixer_trigger()
    0      2       VOLUME         1951564588.958333         2.708333 udio/volume/volume.c:521   volume_trigger()
    0      2          DAI 1.9     1951564591.562500         2.604167      src/audio/dai.c:470   dai_comp_trigger(), command = 0
    0      2          DAI 1.9     1951564593.750000         2.187500      src/audio/dai.c:534   dai_comp_trigger(), PAUSE/STOP
    0      2          DMA         1951564595.729167         1.979167 src/drivers/dw/dma.c:442   dw_dma_stop(): dma 1 channel 0 stop
    0      2          SSP         1951564768.229167       172.500000 intel/baytrail/ssp.c:567   ssp_trigger()
    0      2          SSP         1951564771.041667         2.812500 intel/baytrail/ssp.c:548   ssp_stop(), TX stop
    0      2          SSP         1951564773.750000         2.708333 intel/baytrail/ssp.c:557   ssp_stop(), SSP port disabled
    0      2          IPC         1951565174.010417       400.260406    src/ipc/handler.c:337   ipc: comp 1 -> free
    0      2         PIPE 1.10    1951565176.093750         2.083333 src/audio/pipeline.c:686   pipeline_reset()
    0      2         HOST         1951565178.281250         2.187500     src/audio/host.c:660   host_reset()
    0      2          DMA         1951565180.260417         1.979167 src/drivers/dw/dma.c:254   dw_dma_channel_put(): dma 1 channel 2 put
    0      2       VOLUME         1951565187.083333         6.822917 udio/volume/volume.c:662   volume_reset()
    0      2        MIXER         1951565189.687500         2.604167    src/audio/mixer.c:304   mixer_reset()
    0      2       VOLUME         1951565192.343750         2.656250 udio/volume/volume.c:662   volume_reset()
    0      2          DAI 1.9     1951565194.895833         2.552083      src/audio/dai.c:448   dai_reset()
```

**Bisect result**
unfinished 

**Environment**
Firmware: 2b093579 https://github.com/thesofproject/sof/commits/master
Kernel: 67e5fa8c https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: sof-byt-rt5651.tplg & sof-byt-nocodec.tplg

**log file**
[dmesg.log](https://github.com/thesofproject/sof/files/3345039/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3345040/logger.log)

xiulipan, 2019-07-02:
Bisect to be #1595 
https://sof-ci.01.org/sofpr/PR1595/build2373/devicetest/BYT_MB_NOCODEC/check_aplay_function.sh/check_aplay_function.sh.txt

@tlauda I see the fix and I think it is good for the issue.

closed 2019-07-02 by jajanusz

jajanusz, 2019-07-02:
@xiulipan So why did it pass when I was merging #1595 ? Were there some changes in CI?

xiulipan, 2019-07-02:
@jajanusz 
I think there is some issue in test case that try to get the wrong return value not from the aplay itself but from some other cmd.

PR #1601 (2019-07-02): dw-dma: change behavior for preload
Changes behavior for DMA_COPY_PRELOAD flag.
DW-DMA should just start transfer the same way
as for DMA_COPY_ONE_SHOT. It fixes playback
for platforms using DW-DMA for host transfers.

Fixes #1599.