thesofproject/sof #1610: [FEATURE] Separation of volume control and PCM conversion functionalities

lbetlej, 2019-07-03:
PCM format conversion and volume control are independent audio signal processing actions.  To have better functional separation of components the 2 functionalities of existing Volume component shall be separated. Low power use cases call for for FW components footprint optimization as well.

Requested change is to create 2 separate components : PCM converter and Volume control and implementation of PCM conversion functions as a library functions  (generic & optimized for HiFi co-processor) that can be used i.e. by Keyphrase Detection component.

lgirdwood, 2019-07-04:
@lbetlej these were historically converged since volume control would be on most pipelines and this would give any PCM format conversion needed downstream for free (and avoiding the need for an extra copy), but you are right this does not provide a library for other users (which we could use internally by volume). I would still keep volume performing PCM format conversion though. 
I think we can reduce footprint by using Kconfig for supported volume PCM formats.

ktrzcinx, 2020-04-15:
Done in https://github.com/thesofproject/sof/pull/2029

closed 2020-04-15 by ktrzcinx