thesofproject/sof #1614: [BUG] some glitches happen when testing WoV on WHL

keqiaozhang, 2019-07-08:
**Describe the bug**
Some glitches happen occasionally in the recorded wav when testing the wov.

**To Reproduce**
Run below command line repeatedly.
1. arecord -Dhw:0,8 -M -N -f s24_le --period-size=400 -r 16000 -c 2 wov.wav -d 10 -vvv
2. check the recorded wav file.
3. you may hear the glitches.

**Reproduction Rate**
Reproduction rate is about 5%

**Expected behavior**
no glitches and no noises.

**Impact**
noise can be heard.

**Environment**
1) Kernel: 
branch: topic/keyword-detection
commit/9c986c39d452e88cdf97bf3934b99265263a499e
2) Firmware
branch: cml-005-drop-stable
commit:
3) Platform: WHL-RVP/I2S mode.

singalsu, 2019-07-08:
@keqiaozhang Please attach an example wav so we can analyze what happens. I don't have a system myself to test this.

keqiaozhang, 2019-07-08:
@singalsu 
The wav file is attached.

singalsu, 2019-07-08:
@keqiaozhang Thanks!

Maybe there's repeatedly missing blocks of sound? The glitches seems to occur in 40 ms period long bursts. I'd suggest to record loud, less than 100 Hz sine wave to see if there's e.g. some regular pattern for the glitches.

Also it would be safe to apply e.g. +24 dB gain in the volume component PGA for keyword capture path. Now the signal is very low level.

mrajwa, 2019-07-09:
@keqiaozhang, I like Seppo suggestion about sine wave test. Can we instead of playing some music around use dmic injector and provide it plain sine wave with low frequency?

singalsu, 2019-07-09:
I'd recommend to get a good active speaker to lab. I have myself at work a small Genelec speaker and it is capable to drive a microphone near to it's limits. It's lot easier to do a quick test with a speaker since every board need to  be prepared for connection to DMIC injector. Of course DMIC injector is noiseless and highest quality but there's more effort.

keqiaozhang, 2019-07-11:
@mrajwa 
we don't have DMIC injector in PRC.  Could @wwittbrx  help on this?

mrajwa, 2019-07-19:
@keqiaozhang, what is the status, does the issue still occur with latest patches?

YvonneYang2, 2019-07-22:
Test it on whl-i2s with test recipe as below, issue not reproduced. Close the bug.

Test recipe:
Firmware: d6335a4 https://github.com/thesofproject/sof/commits/cml-005-drop-stable
Kernel: 9c986c3 https://github.com/thesofproject/linux/commits/topic/keyword_detect
Topology:sof-cml-rt5682.tplg link from sof-whl-rt5682-kwd.tplg
platform: whl-i2s

closed 2019-07-22 by YvonneYang2