thesofproject/sof #1615: [BUG] Keyword detection trigger failure on WHL

keqiaozhang, 2019-07-08:
**Describe the bug**
This issue only happens on WHL when doing the wov testing,  Sometimes detection cannot be triggered after clapping. 

**To Reproduce**
Run below test command repeatedly
1. arecord -Dhw:0,8 -M -N -f s24_le --period-size=400 -r 16000 -c 2 wov.wav -d 10 -vvv
2. clap your hands to trigger the detection.

**Reproduction Rate**
Reprodcution rate is about 20%

**Expected behavior**
detection can be triggered properly.

**Impact**
Failed to trigger the detection for WOV.

**Environment**
1) Kernel: 
branch: topic/keyword-detection
commit/9c986c39d452e88cdf97bf3934b99265263a499e
2) Firmware
branch: cml-005-drop-stable
commit:
3) Platform: WHL-RVP/I2S mode.
4) Reproducibility Rate. 20%

**Screenshots or console output**
```
 arecord -Dhw:0,8 -M -N -f s24_le --period-size=400 -r 16000 -c 2 wov.wav -d 10 -vvv
MobaXterm X11 proxy: Authorisation not recognised
xcb_connection_has_error() returned true
Recording WAVE 'wov.wav' : Signed 24 bit Little Endian, Rate 16000 Hz, Stereo
Hardware PCM card 0 'sof-sof_rt5682' device 8 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : MMAP_INTERLEAVED
  format       : S24_LE
  subformat    : STD
  channels     : 2
  rate         : 16000
  exact rate   : 16000 (16000/1)
  msbits       : 32
  buffer_size  : 6528
  period_size  : 408
  period_time  : 25500
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 408
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 6528
  silence_threshold: 0
  silence_size : 0
  boundary     : 7349874591868649472
  appl_ptr     : 0
  hw_ptr       : 0
mmap_area[0] = 0x7f3d9b912000,0,64 (32)
mmap_area[1] = 0x7f3d9b912000,32,64 (32)
```

mrajwa, 2019-07-09:
I will handle this one.

mrajwa, 2019-07-16:
@keqiaozhang, the release branch has been updated regarding this issue.

keyonjie, 2019-07-16:
+ @hualongfeng who might be able to help on this also.

keqiaozhang, 2019-07-17:
@mrajwa 
I tried the latest release branch, but this issue is still reproducible on both cml and whl-rvp.
Firmware info:
https://github.com/thesofproject/sof/commit/b85c320c4aa7898608d64bdd5caa73bbd59f9380

**trace**
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      1          MEM         23760000439.427086              nan      src/lib/alloc.c:246  error: 171 blocks needed for allocation but only 170 blocks are remaining
    0      1          KPB         23763005260.364586   3004821.000000      src/audio/kpb.c:792  kpb_init_draining(), period_limit: 31968 [bytes] and interval 37 [ms] 888000 ticks
    0      1          MEM         23768152778.229168   5147518.000000      src/lib/alloc.c:246  error: 171 blocks needed for allocation but only 170 blocks are remaining
    0      1          KPB         23771130621.354168   2977843.000000      src/audio/kpb.c:792  kpb_init_draining(), period_limit: 31968 [bytes] and interval 37 [ms] 888000 ticks
    0      1          MEM         23776820880.989586   5690259.500000      src/lib/alloc.c:246  error: 171 blocks needed for allocation but only 170 blocks are remaining
```
[logger-t.txt](https://github.com/thesofproject/sof/files/3400420/logger-t.txt)

keqiaozhang, 2019-07-18:
@mrajwa 
Looks like this issue only occurs when testing S24_LE format.  No such issue when testing S16_LE format.

mrajwa, 2019-07-19:
@keqiaozhang, I have provided patch for this on release branch. Please verify if it solved the issue.

hualongfeng, 2019-07-22:
I test the the latest release branch(d6335a4154) on cml chromebook, this issue is still reproducible.

The below is test script. And when the issue occur, the following are not trigger unless wait 3 second.

#!/bin/bash

rm -r ~/Desktop/test/*

for (( i=1; i <= 100; i++ ))
do
	arecord -Dhw:0,8 -M -N -f S24_LE --period-size=400 -r 16000 -c 2 ~/Desktop/test/wov$i.wav -d 10 -vvv
	sleep 1
done

YvonneYang2, 2019-07-22:
Test it on WHL-I2S with test recipe as below, issue still can reproduced.

The frequency :  1/5      1/7      1/2

Test recipe:
Firmware: d6335a4 https://github.com/thesofproject/sof/commits/cml-005-drop-stable
Kernel: 9c986c3 https://github.com/thesofproject/linux/commits/topic/keyword_dete

hualongfeng, 2019-07-23:
This is error tips on firmware branch(6335a4).
[error.txt](https://github.com/thesofproject/sof/files/3419684/error.txt)

mrajwa, 2019-07-25:
@keqiaozhang @YvonneYang2 @hualongfeng, it's because I missed some related patches, sorry about that. They are pushed now, so please test again.

YvonneYang2, 2019-07-26:
Check it on WHL-I2S with test recipe as below, run arecord 150 times, issue cannot reproduce.

Test recipe:
Firmware: 4f590d4 https://github.com/thesofproject/sof/commits/cml-005-drop-stable
Kernel: 9c986c3 https://github.com/thesofproject/linux/commits/topic/keyword_detect

keqiaozhang, 2019-07-26:
I still can reproduce this issue on cml with S24_LE format.

Test recipe:
Firmware: 4f590d4 https://github.com/thesofproject/sof/commits/cml-005-drop-stable
Kernel: 9c986c3 https://github.com/thesofproject/linux/commits/topic/keyword_detect

YvonneYang2, 2019-07-26:
Update https://github.com/thesofproject/sof/issues/1615#issuecomment-515319154. test 150 times with s16_le format cannot reproduce the bug, but can still reproduce the issue with s24_le format.

Reproduce rate: 1/17  
[log_17.txt](https://github.com/thesofproject/sof/files/3434782/log_17.txt)

mrajwa, 2019-07-26:
@keqiaozhang @YvonneYang2, logs doesn't say much, can I have dmesg. Also, how often does it happen? If there is no detection I assume you terminate arecord and start again, then can you trigger?

YvonneYang2, 2019-07-26:
Set the interval time between terminate arecord and start as 10s, issue occurred at 43rd. 
If no detection occurred, terminate it and start, can work again. Logs please see work_again.txt.

[logger43.log](https://github.com/thesofproject/sof/files/3435276/logger43.log)
[work_again_dmesg.txt](https://github.com/thesofproject/sof/files/3435277/work_again_dmesg.txt)
[work_again_logger.txt](https://github.com/thesofproject/sof/files/3435278/work_again_logger.txt)
[dmesg_42.log](https://github.com/thesofproject/sof/files/3435279/dmesg_42.log)
[dmesg_43.log](https://github.com/thesofproject/sof/files/3435280/dmesg_43.log)
[logger_42.log](https://github.com/thesofproject/sof/files/3435281/logger_42.log)

mrajwa, 2019-07-30:
@YvonneYang2, @keqiaozhang please retest with https://github.com/thesofproject/sof/tree/cml-007-drop-stable

YvonneYang2, 2019-07-31:
Issue still reproduced with test recipe as below. Meanwhile, arecord status will stuck during test. Sometimes, arecord stuck issue will block the trigger issue validation.

Reproduce rate:
Trigger issue: 1/11
arecord stuck: 1/5

Test recipe:
Firmware: 442cfb2 https://github.com/thesofproject/sof/commits/cml-007-drop-stable
Kernel: f267c3d https://github.com/thesofproject/linux/commits/topic/keyword_detect
Topology:file: sof-cml-rt5682.tplg link from sof-whl-rt5682-kwd.tplg
platform: whl-i2s

[arecord_stuck_error_logger.txt](https://github.com/thesofproject/sof/files/3450044/arecord_stuck_error_logger.txt)
[arecord_stuck_logger.txt](https://github.com/thesofproject/sof/files/3450045/arecord_stuck_logger.txt)
[arecord_stuck_dmesg.txt](https://github.com/thesofproject/sof/files/3450046/arecord_stuck_dmesg.txt)
[trigger_dmesg.txt](https://github.com/thesofproject/sof/files/3450049/trigger_dmesg.txt)
[trigger_logger.txt](https://github.com/thesofproject/sof/files/3450051/trigger_logger.txt)

YvonneYang2, 2019-07-31:
Arecord stuck issue will track in #1692

mrajwa, 2019-09-02:
This issue should be finally fixed by #1768, @keqiaozhang please verify on your side.

YvonneYang2, 2019-09-03:
Issue is blocked by #1775 , so cannot verify it now.

mrajwa, 2019-09-03:
@YvonneYang2, can you test cml-007-drop-stable, with #1768 merged into?

keqiaozhang, 2019-09-03:
@mrajwa 
I tried to manually apply your PR1768 on 007 stable branch, but failed, there're some hard conflicts, please check the src/audio/pipeline.c, "flags" is still in used and can not be removed, so I keep the flags definition and run test on WHL, this issue still can be reproduced with 100%, even the S16_LE format.

YvonneYang2, 2019-09-04:
Test it 200 times on WHL platform with recipe as below, issue cannot reproduce.

FW 90873c5 + #1689  + #1740  + #1768 
Kernel f95e4eb + PR1037

aiChaoSONG, 2019-09-04:
Test it 200 times on CML Chrome with recipe as below, issue cannot reproduce too.

FW: master branch 90873c5 + #1689 + #1740 + #1768
Kernel: topic/sof-dev branch f95e4eb + PR1037

mrajwa, 2019-09-04:
Good, can we close this issue now?

closed 2019-10-03 by tlauda