thesofproject/sof #1617: [BUG] Xrun on simultaneous recording for DMIC FIFO's with KD topology

wwittbrx, 2019-07-08:
**Describe the bug**
During simultaneous recording on both FIFO's for KD topology we get an xrun.

**To Reproduce**
run arecord -D hw:0,1 -f s32_le -r 48000 -c 4 record_normal.wav -d 10 & arecord -M -N -D hw:0,8 -f s24_le -c 2 -r 16000 -d 10 record_kd.wav

**Reproduction Rate**
100%

**Expected behavior**
FIFO A pipeline records audio for 10 seconds. FIFO B waits for trigger event, drains buffer and realtime data.

**Environment**
https://github.com/thesofproject/linux/commit/9c986c39d452e88cdf97bf3934b99265263a499e
https://github.com/thesofproject/sof/commit/b8e53bccb0a29fd48be475154b8861654695ae4f
https://github.com/thesofproject/sof/blob/cml-005-drop-stable/tools/topology/sof-cml-rt5682-kwd.m4

Ubuntu 19.04 on Hatch platform

**Log**
[wov_xrun.log](https://github.com/thesofproject/sof/files/3368975/wov_xrun.log)

keyonjie, 2019-07-09:
Good catch, @wwittbrx.

@keqiaozhang I believe we can duplicate this on WHL RVP also.

mrajwa, 2019-07-21:
Fixed, patch available on release branch. @wwittbrx please confirm if it is working now.

wwittbrx, 2019-07-22:
Confirmed fixed as of https://github.com/thesofproject/sof/commit/d6335a415431ac9f7e0af3f5322b8eb642b14083. Closing the issue.

closed 2019-07-22 by wwittbrx