thesofproject/sof #1632: [BUG] Pipelines with SRC component work only when converting from lower to higher sampling rates

zrombel, 2019-07-12:
**Describe the bug**
Pipelines with SRC component work only when converting from lower to higher sampling rates. For example if playback pipeline with SRC component is created, host side sampling rate is 16kHz and DAI side sampling rate is 48kHz pipeline works as expected. When host side sampling rate is 48kHz and DAI side sampling rate is 16kHz pipe line fail to copy data.

This bug is caused of incorrect propagation of frames_per_sched parameter trough components in pipelines. 

PR 842 https://github.com/thesofproject/sof/pull/842 is resolving this issue and should be included on master branch.

**To Reproduce**
Create playback pipeline containing SRC component with source sampling rate = 48000 and sink sampling rage 16000.

**Reproduction Rate**
Always

**Found**
SOF FW branch: master, commit: 09b416bb5a97d3f83be8e5122df258d2a187a006 and earlier.

tlauda, 2019-10-02:
@zrombel Please verify with master.

singalsu, 2019-10-07:
@zrombel @lgirdwood Thanks for reporting, I'll check this.

tlauda, 2019-10-10:
@singalsu I think it is fixed. There is #842 mentioned in bug description and it's already merged, so I think we're good.

singalsu, 2019-10-14:
@tlauda I'm having problems to replicate this. I've set up for testing capture SRC pipelines for both 48 kHz and 16 kHz DMIC DAI. Capture 48 -> 48 kHz via SRC works but e.g. 48 -> 16 kHz doesn't. It's maybe the reported issue. Capture for 16 ->16 kHz, 16 -> 32 kHz,  works OK.

In module src.c line 589 "params->rate = cd->source_rate;" looks suspicious. I think it should be instead "params->rate = cd->sink_rate;" for any topology. But changing it does not fix the FW crash that I get.

The stack dump suggests that the crash happens at dma_single_chan_domain_set().

tlauda, 2019-10-14:
@singalsu What is the pipeline and FW commit you are using and getting crash?

singalsu, 2019-10-14:
I'm using git master fetched today with a custom topology. We don't have in git any capture SRC topologies.

tlauda, 2019-10-14:
@singalsu Can you describe how does the pipeline look like and what are the params? I would like to reproduce it on my end.

singalsu, 2019-10-14:
I need to clean up for you a topology to test with. I'll email it tomorrow.

I'm currently using patched bxt pcm512x machine driver to introduce the DMIC BEs. The topology contains one playback SRC and two capture SRCs. To avoid the complexity I'll make a simple nocodec machine driver topology with single SRC capture pipe.

tlauda, 2019-10-16:
Should be fixed with #1947 and #1952.

singalsu, 2019-10-16:
@tlauda Thanks! I'll test the combination of those.

singalsu, 2019-10-16:
> Should be fixed with #1947 and #1952.

Yep, I'm happy to confirm that these both applied fix the SRC related problems!

jdrzaszcz, 2019-10-18:
Checked on slim driver with master commit (81cc31b4ab34dab1378a18f686214256dfe407e3) and src component works on playback and capture pipeline. With both conversion from lower to higher sampling rates and higher to lower.

closed 2019-10-18 by zrombel