thesofproject/sof #1640: [BUG]sof-ctl: failed to store the ABI header

keqiaozhang, 2019-07-16:
**Describe the bug**
The abi header cannot be stored by "sof-ctl -Dhw:0 -n 14 -b -o wov.config"
only can store the 32 bytes detect test config. 

**To Reproduce**
sof-ctl -Dhw:0 -n 14 -b -o wov.config
od -x wov.config

**Reproduction Rate**
100%

**Expected behavior**
should get 64 bytes
00000000 4f53 0046 0000 0000 0020 0000 8000 0300
00000010 0000 0000 0000 0000 0000 0000 0000 0000
00000020 0020 0000 0000 0000 0000 0000 0834 0000
00000030 0003 3fff 0834 0000 0000 0000 0000 0000

but only get 32 bytes
00000020 0020 0000 0000 0000 0000 0000 0834 0000
00000030 0003 3fff 0834 0000 0000 0000 0000 0000

**Impact**
 ABI header cannot be saved.

**Environment**
Test recipe:
Kernel: https://github.com/thesofproject/linux/commit/9c986c39d452e88cdf97bf3934b99265263a499e
Firmware: https://github.com/thesofproject/sof/commit/2418d934a526e261a00eec0615a797016c930ea1

keyonjie, 2019-07-16:
@keqiaozhang Thanks for reporting it, I will figure out a patch to fix it soon.

keyonjie, 2019-07-16:
@keqiaozhang I just create a PR https://github.com/thesofproject/sof/pull/1641 to address this.

PR #1641 (2019-07-17): sof-ctl: fix ABI dumping doesn't work issue
This is aim to fix #1640

For dumping to output file without '-r', we need to dump ABI header,
here fix it.
closed 2019-07-17 by jajanusz