thesofproject/sof #1642: [BUG] detection trigger failure after history_depth is changed

keqiaozhang, 2019-07-16:
**Describe the bug**
I tried to test wov with different history depths,  it can be set with new value, but it doesn't work. only 2.1s draining data can be saved, After checking the code, I found that there's one line hard code to set the history_depth back to 2.1s in firmware. so I tried to remove it, but after this, the detection cannot be triggered. 
What I did is 
```
diff --git a/src/audio/detect_test.c b/src/audio/detect_test.c
index 5d97933bd575..0ad3958320b2 100644
--- a/src/audio/detect_test.c
+++ b/src/audio/detect_test.c
@@ -158,7 +158,7 @@ static void default_detect_test(struct comp_dev *dev,
                                 * Zero value means default config value
                                 * will be used.
                                 */
-                               cd->history_depth = 2100;
+                               cd->history_depth = 0;
                                detect_test_notify(dev);
                                cd->detected = 1;
                        }
```

**To Reproduce**
1. create a new detect configure to change the history_depth, the new configure file I used is:
$ od -x wov.config
```
0000000 0020 0000 0000 0000 0000 0000 0bb8 0000
0000020 0003 3fff 0bb8 0000 0000 0000 0000 0000
0000040
```
2. use sof-ctl tool to update it
 sof-ctl -Dhw:0 -n 14 -br -s wov.config

3. then run the wov test:
arecord -Dhw:0,8 -M -N -c 2 -f S16_LE -r 16000 wov.wav -vvv -d 10

4.  then you will run to the issue. cannot trigger the detection by clapping

**Reproduction Rate**
100%

**Expected behavior**
Keyword detection can work with various of history_depth. 

**Impact**
detection trigger failure.

**Environment**
Test recipe:
1) Kernel: https://github.com/thesofproject/linux/commit/9c986c39d452e88cdf97bf3934b99265263a499e
2) Firmware: https://github.com/thesofproject/sof/commit/2418d934a526e261a00eec0615a797016c930ea1

**Trace**
./sof-logger -l sof-cnl.ldc
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      1          MEM          81585146.822917  81585144.000000      src/lib/alloc.c:246    error: 171 blocks needed for allocation but only 170 blocks are remaining
    0      1          KPB          95238349.791667  13653203.000000      src/audio/kpb.c:722    kpb_init_draining() error: not enough data in history buffer
```

mrajwa, 2019-07-16:
@keqiaozhang, I don't think this it is an issue. First of all you have to make sure that as long as your preamble time is shorter than buffering time (2,1s always) You have to provide history depth less or equal to preamble time(*). If you don't, you will get obvious error which you hit already "kpb_init_draining() error: not enough data in history buffer". 

* Keep in mind that history_depth can no be bigger than 2,1s. So if your preamble time is five seconds you can only retrieve last 2,1 seconds from that.

keqiaozhang, 2019-07-17:
@mrajwa 
I got your point. but the history depth should be configurable. then we can test with various preamble times and history depths.

mrajwa, 2019-07-17:
@keqiaozhang, right. I will unlock this feature in the next patch on release branch.

keqiaozhang, 2019-07-22:
@mrajwa 
Can we set history_depth bigger than 2.1s?
I tried to set history depth=3.0s and preamble size=5.0s, it can trigger after 5.0s, but the draining data is very short(the data before the detection), less than 0.5s, please have a check.

keqiaozhang, 2019-07-26:
@mrajwa 
I still can reproduce above issue on latest code base, bug title is revised.

Test recipe:
Firmware: 4f590d4 https://github.com/thesofproject/sof/commits/cml-005-drop-stable
Kernel: 9c986c3 https://github.com/thesofproject/linux/commits/topic/keyword_detect

mrajwa, 2019-07-27:
> Can we set history_depth bigger than 2.1s?
No, 2.1 is the MAX time-size of the history buffer. So the limit of history depth is 2100 [ms].

> I tried to set history depth=3.0s and preamble size=5.0s, it can trigger after 5.0s, but the draining data is very short(the data before the detection), less than 0.5s, please have a check.

Wait a while, something dosn't add up here - how could you manage to trigger KWD if you had set history depth to incorrect, out of the bound, 3 seconds value? You should get 
`"kpb_init_draining() error: not enough data in history buffer".

How do you change history depth, what files do you modify and to what value? Show me exactly what are you doing please.
`

keqiaozhang, 2019-07-31:
I retested this bug with below recipe, but I run to this issue:
Tested with default history depth and preamble size, no issues.
But for example: set history depth=1.5s and preamble size=2.0s, it can be triggered, but no draining data(before detection) is recorded. it can only record the data after detection, even set the history depth and preamble size back to default 2.1s, still no draining data.

mrajwa, 2019-07-31:
> even set the history depth and preamble size back to default 2.1s, still no draining data.

Huh, how do you reset back these parameters? I assume you change topology file reboot and start again right?  If sof you should get as you said:

> Tested with default history depth and preamble size, no issues.

So something doesn't add up here

keqiaozhang, 2019-07-31:
Here is the detector config after changed: history depth=1.5s and preamble size=2.0s
$ sof-ctl -Dhw:0 -n 14 -br
```
Control size is 304.
Retrieving configuration for device hw:0 control numid=14.
Success.
hdr: magic 0x00464f53
hdr: type 0
hdr: size 32 bytes
hdr: abi 3:9:0
00000000 0020 0000 0000 0000 0000 0000 07d0 0000
00000010 0003 3fff 05dc 0000 0000 0000 0000 0000
```
from the sof-logger, the draining time is 0ms
```
    0      2          KPB         384491244.166667         5.416667      src/audio/kpb.c:838    kpb_init_draining(): requested draining of 0 [ms] from history buffer
```

[logger-aft.txt](https://github.com/thesofproject/sof/files/3450662/logger-aft.txt)
[dmesg.txt](https://github.com/thesofproject/sof/files/3450667/dmesg.txt)

mrajwa, 2019-07-31:
`kpb_init_draining(): requested draining of 0 [ms]` explains why you are getting no draining data.
Looks like your config is not properly aligned with 

> struct sof_detect_test_config

keqiaozhang, 2019-07-31:
Thanks Marcin, after updating my test script. this issue is gone. let's close this bug.

closed 2019-07-31 by keqiaozhang