thesofproject/sof #1645: [BUG] sof-logger -t is empty when run in 2nd time.

ClarexZhou, 2019-07-17:
**Describe the bug**
Sometimes empty log is output when run "sudo sof-logger -l /etc/sof/sof-*.ldc -t"

**To Reproduce**
1. sudo sof-logger -l /etc/sof/sof-*.ldc -t
2. ctrl+c to exit
3. sudo sof-logger -l /etc/sof/sof-*.ldc -t

**Reproduction Rate**
3/3

**Expected behavior**
sof logger should always output.

**Actual result**
sof logger is empty when print 2nd time.

**Environment**
linux (topic/sof-dev) 04b9bfe5
sof (master)  62035b22
tplg: sof-hda-generic-2ch.tplg, same with sof.
platform: CML HDA notebook

**sof logger info and screenshot**
```
ubuntu@sof-dell-cml-hda-06:~$ sudo sof-logger -l /etc/sof/sof-*.ldc -t
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      2          DMA          64673411.145833  64673412.000000 intel/cavs/hda-dma.c:461 	hda-dmac: 4 channel 0 -> get
    0      2          DMA          64673577.812500       166.666672 intel/cavs/hda-dma.c:686 	hda-dmac: 4 channel 0 -> config
    0      2          DMA          64673633.958333        56.145832 intel/cavs/hda-dma.c:530 	hda-dmac: 4 channel 0 -> start
    0      2          DMA          64673637.864583         3.906250 intel/cavs/hda-dma.c:352 	hda-dmac: 4 channel 0 -> enable
    0      2          IPC          64674785.625000      1147.760376    src/ipc/handler.c:894 	ipc: hdr 0x30010000 tx (76) > rx (28)
    0      2          IPC          64674789.375000         3.750000    src/ipc/handler.c:897 	ipc: pipe 1 comp 1 -> new (type 1)
    0      2         HOST          64674795.000000         5.625000     src/audio/host.c:342 	host_new()
    0      2          DMA          64674811.666667        16.666666 intel/cavs/hda-dma.c:831 	hda-dmac :5 -> probe
    0      2          DMA          64674828.697917        17.031250        src/lib/dma.c:125 	dma_get() ID 5 sref = 1 busy channels 0
    0      2          IPC          64675009.375000       180.677078    src/ipc/handler.c:894 	ipc: hdr 0x30010000 tx (84) > rx (28)
    0      2          IPC          64675013.020833         3.645833    src/ipc/handler.c:897 	ipc: pipe 1 comp 2 -> new (type 5)
    0      2       VOLUME          64675018.333333         5.312500 udio/volume/volume.c:144 	volume_new()
    0      2       VOLUME          64675036.666667        18.333334 udio/volume/volume.c:216 	vol->initial_ramp = 250, vol->ramp = 0, vol->min_value = 0, vol->max_value = 65536
    0      2          IPC          64675190.885417       154.218750    src/ipc/handler.c:927 	ipc: pipe 1 buffer 3 -> new (0x300 bytes)
    0      2       BUFFER          64675196.041667         5.156250   src/audio/buffer.c:27  	buffer_new()
    0      2          IPC          64675362.864583       166.822922    src/ipc/handler.c:927 	ipc: pipe 1 buffer 4 -> new (0x300 bytes)
    0      2       BUFFER          64675367.916667         5.052083   src/audio/buffer.c:27  	buffer_new()
    0      2          IPC          64675544.062500       176.145828    src/ipc/handler.c:894 	ipc: hdr 0x30010000 tx (80) > rx (28)
    0      2          IPC          64675547.708333         3.645833    src/ipc/handler.c:897 	ipc: pipe 1 comp 5 -> new (type 2)
    0      2          DAI          64675553.750000         6.041667      src/audio/dai.c:156 	dai_new()
    0      2          DAI          64675567.812500        14.062500        src/lib/dai.c:69  	dai_get(), d = 0xbe04d000, sref = 1
    0      2          DMA          64675572.239583         4.427083 intel/cavs/hda-dma.c:831 	hda-dmac :7 -> probe
    0      2          DMA          64675589.479167        17.239584        src/lib/dma.c:125 	dma_get() ID 7 sref = 1 busy channels 0
    0      2          IPC          64675714.270833       124.791664    src/ipc/handler.c:955 	ipc: pipe 1 -> new
    0      2         PIPE          64675721.822917         7.552083 src/audio/pipeline.c:46  	pipeline_new()
    0      2          IPC          64675870.937500       149.114578    src/ipc/handler.c:644 	ipc: pm core mask 0x1 -> enable
    0      2          IPC          64675997.812500       126.875000    src/ipc/handler.c:894 	ipc: hdr 0x30010000 tx (76) > rx (28)
    0      2          IPC          64676001.458333         3.645833    src/ipc/handler.c:897 	ipc: pipe 2 comp 9 -> new (type 1)
    0      2         HOST          64676008.333333         6.875000     src/audio/host.c:342 	host_new()
    0      2          DMA          64676024.427083        16.093750        src/lib/dma.c:125 	dma_get() ID 4 sref = 2 busy channels 1
    0      2          IPC          64676146.302083       121.875000    src/ipc/handler.c:894 	ipc: hdr 0x30010000 tx (84) > rx (28)
    0      2          IPC          64676150.000000         3.697917    src/ipc/handler.c:897 	ipc: pipe 2 comp 10 -> new (type 5)
    0      2       VOLUME          64676156.927083         6.927083 udio/volume/volume.c:144 	volume_new()
    0      2       VOLUME          64676175.208333        18.281250 udio/volume/volume.c:216 	vol->initial_ramp = 250, vol->ramp = 0, vol->min_value = 0, vol->max_value = 2072775
    0      2          IPC          64676288.802083       113.593750    src/ipc/handler.c:927 	ipc: pipe 2 buffer 11 -> new (0x300 bytes)
    0      2       BUFFER          64676295.677083         6.875000   src/audio/buffer.c:27  	buffer_new()
    0      2          IPC          64676443.072917       147.395828    src/ipc/handler.c:927 	ipc: pipe 2 buffer 12 -> new (0x300 bytes)
    0      2       BUFFER          64676449.895833         6.822917   src/audio/buffer.c:27  	buffer_new()
    0      2          IPC          64676607.135417       157.239578    src/ipc/handler.c:894 	ipc: hdr 0x30010000 tx (80) > rx (28)
    0      2          IPC          64676610.781250         3.645833    src/ipc/handler.c:897 	ipc: pipe 2 comp 13 -> new (type 2)
    0      2          DAI          64676618.385417         7.604167      src/audio/dai.c:156 	dai_new()
    0      2          DAI          64676632.343750        13.958333        src/lib/dai.c:69  	dai_get(), d = 0xbe04d048, sref = 1
    0      2          DMA          64676636.718750         4.375000 intel/cavs/hda-dma.c:831 	hda-dmac :6 -> probe
[... 484 lines cut ...]
    0      2         PIPE 7.42     71617634.895833        11.875000 src/audio/pipeline.c:606 	pipeline_trigger()
    0      2         HOST          71617638.802083         3.906250     src/audio/host.c:288 	host_trigger()
    0      2          DMA          71617643.125000         4.322917 intel/cavs/hda-dma.c:625 	hda-dmac: 5 channel 0 -> stop
    0      2       VOLUME          71617647.552083         4.427083 udio/volume/volume.c:530 	volume_trigger()
    0      2          DAI 7.41     71617652.031250         4.479167      src/audio/dai.c:472 	dai_comp_trigger(), command = 0
    0      2          DAI 7.41     71617655.833333         3.802083      src/audio/dai.c:536 	dai_comp_trigger(), PAUSE/STOP
    0      2          DMA          71617659.583333         3.750000 intel/cavs/hda-dma.c:625 	hda-dmac: 7 channel 0 -> stop
    0      2          IPC          71617881.510417       221.927078    src/ipc/handler.c:338 	ipc: comp 37 -> free
    0      2         PIPE 7.42     71617893.385417        11.875000 src/audio/pipeline.c:685 	pipeline_reset()
    0      2         HOST          71617897.447917         4.062500     src/audio/host.c:675 	host_reset()
    0      2       VOLUME          71617904.687500         7.239583 udio/volume/volume.c:671 	volume_reset()
    0      2          DAI 7.41     71617909.218750         4.531250      src/audio/dai.c:450 	dai_reset()
    0      2          IPC          71618173.333333       264.114594    src/ipc/handler.c:527 	ipc: dai 3,4 -> config 
    0      2          DAI          71618177.916667         4.583333        src/lib/dai.c:69  	dai_get(), d = 0xbe04d120, sref = 2
    0      2          DAI          71618182.083333         4.166667        src/lib/dai.c:94  	dai_put(), dai = 0xbe04d120, sref = 1
    0      2          DAI          71618193.958333        11.875000      src/audio/dai.c:644 	config comp 41 pipe 7 dai 4 type 3
    0      2          DAI 7.41     71618197.812500         3.854167      src/audio/dai.c:717 	dai_config(), channel = -1
    0      2          IPC          71618492.083333       294.270844    src/ipc/handler.c:527 	ipc: dai 3,4 -> config 
    0      2          DAI          71618496.354167         4.270833        src/lib/dai.c:69  	dai_get(), d = 0xbe04d120, sref = 2
    0      2          DAI          71618500.520833         4.166667        src/lib/dai.c:94  	dai_put(), dai = 0xbe04d120, sref = 1
    0      2          DAI          71618512.187500        11.666667      src/audio/dai.c:644 	config comp 41 pipe 7 dai 4 type 3
^C
ubuntu@sof-dell-cml-hda-06:~$ sudo sof-logger -l /etc/sof/sof-*.ldc -t
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
```

xiulipan, 2019-07-17:
@ranj063 @ClarexZhou 
I think this is not a issue.

In Linux driver, when we read out from trace for once. We will move the buf position so that if there is no more dma trace, we will have empty trace as we have read out them already.

xiulipan, 2019-07-17:
did some test and make a correction:

In Linux driver, when we read out from trace for once in **PM suspend**. We will try to read out from some invalid buffer and flush the existing trace.

@ClarexZhou If you try to get trace with an aplay running, you will always get data. But if the device is idle and you run twice, you will get nothing.

@ranj063 if we need to disable the cache when we are in PM suspend?

kv2019i, 2019-07-17:
@ClarexZhou Ack on comment from @xiulipan . This is the expected behaviour in trace (-t) mode. User-space client (i.e. sof-logger) is responsible to store the trace. Once read, kernel trace buffer will be empty. 

This was changed in Q2 to fix issues of getting duplicated/missing traces when running sof-logger over runtime PM suspend/resume cycles.

@xiulipan What you describe in second comment does not sound right. Can you detail what you did exactly. If SOF is in runtime-suspend and you start sof-logger, it will read any traces that are left in the buffer. If you close and run again, you should not get anything anymore, until device is restarted. This is the expected behaviour.

xiulipan, 2019-07-18:
@kv2019i 
Thanks for the clarification. Then I think this issue is not a bug now?

ranj063, 2019-07-19:
@xiulipan yes, this is expected behaviour. It was agreed upon already to have it working this way. The way to work around the issue is to save the trace to a file and then use sof-logger with the save trace file.

ClarexZhou, 2019-07-22:
close for not a bug

closed 2019-07-22 by ClarexZhou