thesofproject/sof #1648: [BUG]detection trigger failure with GCC toolchain

hualongfeng, 2019-07-18:
**Describe the bug**
I tried to test the issue 1615, but it can not be triggered after clapping.

**To Reproduce**
Run below test command

    1. arecord -Dhw:0,8 -M -N -f s24_le --period-size=400 -r 16000 -c 2 wov.wav -d 10 -vvv
    2. clap your hands to trigger the detection.
**Reproduction Rate**
always not trigger

**Expected behavior**
detection can be triggered properly.

**Impact**
Failed to trigger the detection for WOV.

**Environment**
1) Kernel:
    branch: topic/keyword-detect
    commit/9c986c39d452e88cdf97bf3934b99265263a499e
2) Firmware
    branch: top cml-005-drop-stable
    commit: b8e53bc
3) Platform: WHL-RVP/I2S mode.
4) Reproducibility Rate. always not.

mrajwa, 2019-09-02:
This issue should be fixed by #1768, @hualongfeng please verify.

lgirdwood, 2019-09-08:
@hualongfeng ping, can you update.

keqiaozhang, 2019-09-09:
@YvonneYang2 
Please help to verify #1768.

YvonneYang2, 2019-09-09:
Test it on whl with FW GCC toolchain compiled , issue cannot reproduce.

Test recipe:
FW: master branch 90873c5 + #1689 + #1740 + #1768
Kernel: topic/sof-dev branch f95e4eb + PR1037

lgirdwood, 2019-09-09:
@YvonneYang2 I've renamed this as it's not a compile failure.

mrajwa, 2019-09-09:
@lgirdwood, I think we can close this issue as it seems to be fixed now. Do you Agree?

YvonneYang2, 2019-09-10:
Test it on whl with test recipe as below, arecord with s24_le format will hang after trigger it successfully . 

Reproduced rate :100%

Test recipe:
FW: master branch 92b23fa8 (pr1689 has already been merged)+ #1740  + #1768  
Kernel: topic/sof-dev branch 21f22f5 + PR1037

**error logger**
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          MEM          50935791.041667  50935792.000000      src/lib/alloc.c:273 	error: 171 blocks needed for allocation but only 170 blocks are remaining
    0      1          EDF          56090606.927083   5154816.000000 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56115605.416667     24998.490234 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56140605.312500     24999.896484 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56165605.312500     25000.000000 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56190605.260417     24999.947266 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56215605.156250     24999.896484 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56240606.145833     25000.990234 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56265605.364583     24999.218750 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56290605.000000     24999.634766 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56315605.156250     25000.156250 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56340605.260417     25000.103516 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56365605.364583     25000.103516 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56390606.302083     25000.937500 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56415605.364583     24999.062500 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56440605.312500     24999.947266 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56465605.208333     24999.896484 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56490605.156250     24999.947266 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56515605.156250     25000.000000 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56540606.145833     25000.990234 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56565605.260417     24999.115234 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56590605.156250     24999.896484 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56615605.364583     25000.208984 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56640605.000000     24999.634766 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56665605.312500     25000.312500 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56690606.250000     25000.937500 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56715605.208333     24998.958984 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56740605.364583     25000.156250 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
    0      1          EDF          56765605.156250     24999.791016 edule/edf_schedule.c:125 	schedule_edf_task(), task already queued or running 3, deadline = 101855771 
[... cut ...]
    0      1         PIPE          57165677.343750        52.447918 src/audio/pipeline.c:802 	pipeline_copy() error: ret = -61, start->comp.id = 40, dir = 0
    0      1         PIPE 8.41     57165684.218750         6.875000 src/audio/pipeline.c:985 	pipeline_task(): xrun recover failed! pipeline will be stopped!
```
[dmesg.log](https://github.com/thesofproject/sof/files/3593352/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3593353/logger.log)
[error_logger.log](https://github.com/thesofproject/sof/files/3593354/error_logger.log)

mrajwa, 2019-09-10:
@YvonneYang2 well this is what testing active developing branch means - you are prone to new bugs and this is exactly what happens here. I think @tlauda is working on this already.

mrajwa, 2019-09-10:
Update, fix for edf scheduler issue has already been merged into master.

YvonneYang2, 2019-09-25:
Check it on whl-i2s with test recipe as below, issue cannot reproduce. So close the bug.

Test recipe:
Firmware: d42c019 https://github.com/thesofproject/sof/commits/master
Kernel: c0f21e6 https://github.com/thesofproject/linux/commits/topic/sof-dev 
tplg:   sof-whl-rt5682-kwd.tplg rename sof-cml-rt5682.tplg

closed 2019-09-25 by YvonneYang2