thesofproject/sof #1649: [BUG] Checkpatch is out-of-date

jajanusz, 2019-07-18:
**Describe the bug**
F.e. our checkpatch will not allow this
```
prtinf("super very long line that is exceeding 80 characters limit..........................................",
          some_ arg ...)
```
While checkpatch in kernel repo will allow string to exceed 80chars limit (as long as it's last thing in the line).

I guess there are more irritating things, that were fixed over time in kernel repo and were not ported here.

**To Reproduce**
Run checkpatch on example patch.

**Reproduction Rate**
100%

**Expected behavior**
I expect checkpatch to work more like in kernel repo.

**Impact**
Annoyance

mrajwa, 2019-07-18:
@jajanusz, but as for this 80 character limit - does it complain about 80 character straight with no line break or 80 in total? If it is the former case then I would say its OK.

jajanusz, 2019-07-18:
> @jajanusz, but as for this 80 character limit - does it complain about 80 character straight with no line break or 80 in total? If it is the former case then I would say its OK.

I'm talking about string literal without break, for example 80+ chars in 1 textual, not logical line.
While it may make sense for many arguments then it's not practical for long error messages. It makes grepping/searching inconvenient and string-splitting is actually banned in kernel guidelines - our checkpatch encourages us to go against this guideline.

tlauda, 2019-07-20:
@jajanusz Can we also get rid of some of the not applicable warnings? This one is very annoying: WARNING: Please update ABI in accordance with http://semver.org.

jajanusz, 2019-07-23:
> @jajanusz Can we also get rid of some of the not applicable warnings? This one is very annoying: WARNING: Please update ABI in accordance with http://semver.org.

This one is good reminder, problem with it is that checkpatch is too dumb to know that you already applied what it suggests and you cannot get rid of this warn...

tlauda, 2019-07-23:
@jajanusz That's why I think it eliminates itself from being useful. We shouldn't automatically enforce the rule we cannot verify.

jajanusz, 2019-07-23:
@tlauda It's not even in kernel repo, so I guess @lgirdwood added that, but I guess it would be better to just auto-label it with ABI (similar to how CODWOWNERS work). Someone would need to add it to CI, here's some inspiration -> https://tech.lendinghome.com/auto-labeling-pull-requests-118c84f188d3

PR #2039 (2019-11-04): checkpatch: update to match kernel 5.4-rc4
Merge kernel's checkpatch with custom features added for SOF.
SOF follows kernel code style, so checkpatch should be also up-to-date.

Fixes #1649
closed 2019-11-04 by jajanusz