thesofproject/sof #1650: [BUG] Playback silence for formats 8kHz 16b/16b 1ch and 24kHz 16b/16b 1ch

zrombel, 2019-07-18:
**Description**
Issue concerns two audio formats:
- 8kHz 16b/16b 1ch
- 24kHz 16b/16b 1ch
For those formats there was a need of setting period_source and period_sink parameters in pipeline components at minimal value of 4. 

After commit 528c72baefa299dd46f593f91246ac7f76c3047a there are two issues:

1) Silent samples at the beginning of playback on pipeline HOST -> BUF -> DAI
[image]

2) Silence played on pipeline HOST -> BUF -> VOL -> BUF -> DAI
[image]

**Reproduction Rate**
Always

**Environment**
1) Branch master commit 528c72baefa299dd46f593f91246ac7f76c3047a and later 
2) Bug concerns all platforms

bkokoszx, 2019-07-22:
In our current flow of propagating prepare across the simple pipeline (HOST->DAI) we have respectively:
1. host_params()
2. dai_params().
Both of this functions invoke buffer_set_size function on the same buffer. In host_params() we perform buffer alignment to 32 B (for above format and period=4 buffer size is equal ALIGN_UP_32B(8 * 2B) * 4 = 128B). On the other hand in dai_params() we do not have any buffer alignment (buffer size is equal to 8 * 2 * 4 = 64B). Difference in buffer size calculation provides a mismatch, which is a root cause of this issue.

In my opinion we can perform similar buffer alignment on dai side. Additionally we should not allow other components like volume in comp_prepare() change sink buffer size if sink component is connected to dai (only check if sink buffer size is sufficient). Here is an example: 
https://github.com/bkokoszx/sof/commits/dw-dma-buffer-alignment

@juimonen 
What's your opinion?

zrombel, 2019-07-22:
@bkokoszx changes fixes this bug. I have checked it with pass-trough pipelines, pipelines with volume and mixer, so it should be merged into master.

PR #1663 (2019-07-27): comp: add comp_set_sink_buffer() function
Component in comp_prepare() should change sink buffer
only when sink component is not connected to dma. 
comp_set_sink_buffer() checks whether sink component
is not connected to dma and only in that case tries to
resize sink buffer.

This PR also add dai buffer alignment in order to provide the same
alignment as in the host side. Otherwise it may appear buffer 
mismatch between host and dai (in simple host->dai pipeline)
resulting in issue #1650

PR fixes #1650
closed 2019-07-27 by tlauda