thesofproject/sof #1659: [BUG] Topology load failure on CML

keqiaozhang, 2019-07-22:
**Describe the bug**
Tplg failed to load on CML ChromeBook. This is more like a regression, no such issue on cml-005-drop-stable branch.

**To Reproduce**
check the dmesg after system boot.

**Reproduction Rate**
100%.

**Expected behavior**
No failure after booting

**Impact**
failed to create the audio pipelines

**Environment**
1) kernel: https://github.com/thesofproject/linux/commit/44c31df18ca1b04d0018542b31d58587193604cb
2) firmware:
https://github.com/thesofproject/sof/commit/a513ac7251b16aa9be0f443af091c40f7b8cf084
3) firmware binary: sof-cnl.ri
4) tplg binary: sof-cml-rt5682-max98357a.tplg

**dmesg**
```
[    2.439687] sof-audio-pci 0000:00:1f.3: buffer BUF8.0: size 5120 caps 0x71
[    2.439689] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[    2.439788] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x30200000 size 20
[    2.439790] sof-audio-pci 0000:00:1f.3: error: buffer BUF8.0 load failed
[    2.439792] sof-audio-pci 0000:00:1f.3: error: DSP failed to add widget id 0 type 31 name : BUF8.0 stream none reply -12
[    2.439794] sof-audio-pci 0000:00:1f.3: ASoC: failed to load widget BUF8.0
[    2.439802] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[    2.439805] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1000f0f successful
[    2.439807] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    2.439810] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.439813] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.439815] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    2.439818] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.439821] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.439823] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    2.439826] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.439829] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.439831] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    2.439835] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.439838] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.439839] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    2.439842] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.439845] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.439847] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    2.439851] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.439854] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.439855] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    2.439856] sof-audio-pci 0000:00:1f.3: error: tplg component load failed -12
[    2.439861] sof-audio-pci 0000:00:1f.3: error: failed to load DSP topology -22
[    2.439862] sof-audio-pci 0000:00:1f.3: ASoC: failed to probe component -22
[    2.439873] sof_rt5682 sof_rt5682: ASoC: failed to instantiate card -22
[    2.439891] sof_rt5682: probe of sof_rt5682 failed with error -22
```

[dmesg.txt](https://github.com/thesofproject/sof/files/3416393/dmesg.txt)

ranj063, 2019-07-22:
@keqiaozhang so you;re saying the same kernel works with the cml-005 branch?

keqiaozhang, 2019-07-23:
> @keqiaozhang so you;re saying the same kernel works with the cml-005 branch?

Yes, same kernel and tplg binary works with cml-005 branch.

keyonjie, 2019-07-23:
@keqiaozhang maybe master is not ready for cml chromebook? We released them with cml-0045, cml-005, cml-006, but may be some commits are not submitted to master yet.

Can you provide the sof-logger logs also, looks the buffer allocation failed there.

keqiaozhang, 2019-07-23:
@keyonjie 
sof-logger is not available after this issue. No traces output.

keqiaozhang, 2019-07-30:
@mrajwa 
Any updates on this bug?  This is a blocker for new release in this week.

mrajwa, 2019-07-30:
@keqiaozhang I think it is fixed, please try with master branch or https://github.com/thesofproject/sof/tree/cml-007-drop-stable

keqiaozhang, 2019-07-31:
Confirmed that this issue is fixed on both cml-007-drop-stable and master branches.
master: https://github.com/thesofproject/sof/commit/c6dd4797874db228e78285fbe80cd8b77e62536d
cml-007-drop-stable: https://github.com/thesofproject/sof/commit/442cfb2f2bbc05cd1617cd205aeeb18d9dfb699a

closed 2019-07-31 by keqiaozhang