thesofproject/sof #1665: [BUG][APL][WHL][ICL] Glitches on stream when another stream is being started

zrombel, 2019-07-23:
**Description**
When stream is being stared a glitch can be found on already running stream.
Example of three playback pass-through pipelines:
Sending trigger start to all pipelines:
2019-07-23 13:34:55,303 | Sending StreamTrigStart (60040000, 00000000)
2019-07-23 13:34:55,709 | Sending StreamTrigStart (60040000, 00000000)
2019-07-23 13:34:55,975 | Sending StreamTrigStart (60040000, 00000000)
Time delta between first and last TrigStart ipc is 672ms.
And on wave recorded on first pipeline glitch can be seen after approximately 660ms from start:
[image]

[image]

**To Reproduce**
Start consecutive streams on multi pipeline topology.

**Reproduction Rate**
~30%

**Environment**
1) Branch: Master, commit: a513ac7251b16aa9be0f443af091c40f7b8cf084 
2) All platforms

tlauda, 2019-10-02:
@zrombel Pleas verify with master.

wenqingfu, 2019-10-24:
@zrombel ping for update...

zrombel, 2019-10-25:
Bug cannot be reproduce on FW commit 89ef0 of master brach. Closing Issue.

closed 2019-10-25 by zrombel