thesofproject/sof #1674: [BUG][CFL] System freeze when capture via both 2 DMIC at same time.

ClarexZhou, 2019-07-26:
**Describe the bug**
 System freeze when capture via both 2 DMIC at same time.

**To Reproduce**
1. Open one terminal, run 'arecord -Dhw:0,6 -r48000 -c2 -fs32_le -vv -i 1.wav'
2. Open one terminal, run 'arecord -Dhw:0,7 -r48000 -c2 -fs16_le -vv -i 2.wav'

**frequency**
100%

**Actual result**
System freeze

**Expected result**
Both capture works OK.

**Environment**
linux (topic/sof-dev): f745362c
sof (master): fd6208d6
Topology: file: sof-hda-generic-2.tplg same path as firmware
platform: CFL

only below message is got for system is freeze.
[freeze2Dmic_soflogger.txt](https://github.com/thesofproject/sof/files/3435057/freeze2Dmic_soflogger.txt)
[freeze2Dmic_dmesg.txt](https://github.com/thesofproject/sof/files/3435058/freeze2Dmic_dmesg.txt)

xiulipan, 2019-07-26:
@jajanusz @lyakh 
I checked the logs and found before Jul.24 all test case is passed. But from Jul.25 the test case is not pass. Linux is not changed between these two days.

Only un-test PR that merged on Jul.24 is https://github.com/thesofproject/sof/pull/1592
The test case is test by CI on Jun.28 and not re-test.

Could you try to revert https://github.com/thesofproject/sof/commit/825646595580ff181520d2c5173575bb42d5188d and test again?

xiulipan, 2019-07-26:
@ClarexZhou @jajanusz @lyakh 
It seems this issue is already fixed by #1669

dengyangchao, 2019-07-29:
Test it on CFL with test recipe as below, issue not reproduced. Close the bug.
Test recipe:
Firmware: f7809a2e
Kernel: f745362c
Topology:sof-hda-generic-2.tplg same path as firmware
platform: CFL

closed 2019-07-29 by ClarexZhou