thesofproject/sof #1675: [BUG] DMIC capture silent on WHL laptop with 1.3 signed firmware

plbossart, 2019-07-26:
**Describe the bug**

with the 1.3 signed firmware, the sound captured w/ DMICs is inaudible. With earlier versions it was low but still audible

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug) 

change the IP as needed
scp sof-cnl-signed-intel.ri root@192.168.1.185:/lib/firmware/intel/sof/
ssh root@192.168.1.185 'cd /lib/firmware/intel/sof; ln -s sof-cnl-signed-intel.ri sof-cnl.ri'
scp sof-hda-generic-4ch.tplg root@192.168.1.185:/lib/firmware/intel/sof-tplg/
ssh root@192.168.1.185 'cd /lib/firmware/intel/sof-tplg; ln -s sof-hda-generic-4ch.tplg sof-hda-generic.tplg'

Use SOF kernel or Mark Brown's tree

**Reproduction Rate**

100%

**Expected behavior**

loud-enough and crisp input

**Impact**

showstopper, DMICs are the only capture.

**Environment**

already provided

Side note: why do we have a DMIC32 and DMIC16 inputs? this is completely misleading, the DMIC32 operates at 48kHz on 16 bits and the other on 16KHz 16 bits.

plbossart, 2019-07-26:
see wave file attached and screenshot. The DC removal is still out of whack
[plb.wav.gz](https://github.com/thesofproject/sof/files/3437287/plb.wav.gz)
[image]

plbossart, 2019-08-07:
@singalsu can you please take a look now that you are back?

singalsu, 2019-08-08:
@plbossart I'll start with this, thanks for reporting & reminding. I have a suitable laptop to test here.

Concerning that attached waveform plot I'd no more increase the unmute ramp time to conceal the PCM negative values drift. It would be better to place IIR EQ component with high-pass into topology. Then the unmute ramp could be even shortened from current length. The slow DC offset removal settling is not configurable to any faster in cAVS so it needs to be addressed by FW.

singalsu, 2019-08-08:
@plbossart Can you check the alsamixer settings. The topology has volume controllers for the both DMIC capture devices so the gain could be adjusted up. I suspect the gain is in default 0 dB state for them.

I wasn't able to use the tool for plotting topologies for this so I can't name the PGAx.y instances to check. Something in this topology doesn't work with the old tool.

But since we need the gains to be sane out of the box we need to set a default gain somewhere, e.g. +20 dB (it's a typical on/ff switchable gain for ADCs/CODECs to use for analog microphones).

To address the low gain there could be several options

1. Configure a gain somewhere in UCM for DMIC capture devices

2. If possible set in sof-hda-generic.m4 topology default volume for DMIC capture to provide the gain

3. If we decide to use EQ to properly address the DC seen in the beginning it could be programmed for a fixed gain and volume at 0 dB would produce sufficiently loud capture.

singalsu, 2019-08-09:
[image]

singalsu, 2019-08-09:
In this case PGA10.0 and PGA11.0 should get e.g. +20 dB default gain if not done via adding EQ to pipes.

singalsu, 2019-08-13:
I just created SOF PR #1725. It's addressing these issues with high-pass EQ (IIR). Here are 16 kHz and 48 kHz demo recordings with 0 dB PGA10.0 and PGA11.0 settings with WHL laptop.

[rec16k_eq_demo.wav.gz](https://github.com/thesofproject/sof/files/3497991/rec16k_eq_demo.wav.gz)
[rec48k_eq_demo.wav.gz](https://github.com/thesofproject/sof/files/3497993/rec48k_eq_demo.wav.gz)

singalsu, 2019-08-19:
PR #1725 is now merged so this can be closed.

closed 2019-08-19 by singalsu