thesofproject/sof #1676: [FEATURE] Need pipeline/DAIs with dynamic configurations

plbossart, 2019-07-26:
**Is your feature request related to a problem? Please describe.**

Outputs such as HDMI may support of variety of sample rates and formats, as reported by the receiver with the ELD/EDID information.
The current SOF firmware+topologies assume fixed configurations which don't allow us to support such configurations.

**Describe the solution you'd like**

We need a mechanism where the topologies do not have expose sample/formats and where the firmware adjust the DAI output rate. This could mean no checks, and the host driver is supposed to make the checks, or we explicitly describe a mask of supported capabilities.

**Describe alternatives you've considered**

none

**Additional context**

Limiting the HDMI output to 48kHz/2ch is clearly a bad idea. Users will soon be forced to use SOF instead of the legacy HDAudio driver when DMICs are required, this means they need to get iso-capabilities on the HDMI front.

kkurbjun, 2021-11-19:
I am running with SOF on my laptop and have been looking into why I cannot get multi-channel audio or HDMI bitstream passthrough when SOF is loaded.  If I boot with the legacy HDA code ("options snd-intel-dspcfg dsp_driver=1" ) I am able to select 5.1 profiles with pulseaudio/pipewire but when I let SOF load I only get the option for 2 channel PCM over HDMI.

As you called out the legacy HDA driver doesn't work with the built-in microphone using DMIC so right now I have the option of no mic or no multi channel audio through HDMI.

Is this enhancement the reason I cannot drive anything other than 2 channel PCM over HDMI or should I be looking elsewhere on how to get multi-channel HDMI working when SOF is loaded?

plbossart, 2021-11-19:
@kkurbjun the primary reason is that the topology files for SOF/HDaudio platforms only use 2 channels. There is no real restriction in the kernel driver proper, it will just follow what the topology indicates. It's something we are planning to fix but likely in 2022 at this point. Thanks for your understanding.

kv2019i, 2023-11-28:
@jsarha @ujfalusi This is now fixed with chain-dma support, right?

jsarha, 2023-11-28:
> @jsarha @ujfalusi This is now fixed with chain-dma support, right?

Yes it is. 

However, it requires the chain-dma to be in use. Chain-DMA is the default behavior now, but the in the topology its possible to revert back to the old DSP pass-through behaviour. If that is done, the multichannel (or DTS/DD/AC3/etc. passthrough) does not work.

jsarha, 2023-11-28:
@plbossart ,  do you agree that this issue is now solved?

Of course it would be nice to be able to play multichannel audio through DSP and do some processing, but that is a different issue, isn't it? Not even sure what would be the usecase here.

kkurbjun, 2023-11-28:
Is there a way I should be testing this?  I am running the v2023.09.1 firmware release on Fedora 38 and I still only see a stereo output option via HDMI through gnome settings.

kv2019i, 2023-11-28:
@kkurbjun Unfortunately, in the official binaries, this is currently only available for Intel Meteor Lake and newer. This requires a move to new IPC version, and quite recent Linux kernel, so at least for now we are not planning to move older platforms. We did release opt-in binaries for SOF2.5 that has this feature for TGL/ADL/RPL. but as you can see from the notes, the requirements are quite complex -> https://github.com/thesofproject/sof-bin/tree/main/v2.5.x

jsarha, 2023-11-28:
> @jsarha can you remind me how the ELD information is handled and used to allow/prevent specific frequencies/channel counts from being used when opening a PCM device?

I've tested that it works, and if I am not mistaken, the magic happens in hdac_hdmi_pcm_open(), where it calls snd_pcm_hw_constraint_eld(). But at this hour I am not going to track the full call path.