thesofproject/sof #1677: [BUG][BYT] Failed while do playback and capture at same time on BYT MinnowBoard platform: I/O error (100%)

Jiangxinx, 2019-07-29:
**Describe the bug**
Failed while do playback and capture at same time on MinnowBoard BYT-NOCODEC and BYT-rt5651 platform: I/O error (100%). Dmesg and sof-logger have not any abvious error traces.
I think it is caused by PR #1053 that I had reproduced on the 20190726.

**To Reproduce**
1.
terminal 1:
```
ubuntu@cd-byt-mb-rt5651-02:~$ aplay -Dhw:0,0 -fs16_LE -c2 -r 48000 audio-files/raw/test-c2-r48000-fs16_le-d60.raw -vv -i
Playing raw data 'audio-files/raw/test-c2-r48000-fs16_le-d60.raw' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 0 'sof-bytcr-rt5651' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16368
  period_size  : 4080
  period_time  : 85000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4080
  period_event : 0
  start_threshold  : 16368
  stop_threshold   : 16368
  silence_threshold: 0
  silence_size : 0
  boundary     : 9214364837600034816
  appl_ptr     : 0
  hw_ptr       : 0
############                            +          | 79%
```
terminal 2:
```
ubuntu@cd-byt-mb-rt5651-02:~$ arecord -Dhw:0,0 -fs16_LE -c2 -r 48000 1.wav -vv -i
Recording WAVE '1.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 0 'sof-bytcr-rt5651' device 0 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16320
  period_size  : 4080
  period_time  : 85000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4080
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 16320
  silence_threshold: 0
  silence_size : 0
  boundary     : 9187343239835811840
  appl_ptr     : 0
  hw_ptr       : 0
#+                                                 | 00%
```

2.
terminal 1:   ctrl+c stop the playback
```
Aborted by signal Interrupt...
#########                               +          | 79%
```
terminal 2:
```
arecord: pcm_read:2103: read error: Input/output error
```

**Reproduction Rate**
100%

**Expected behavior**
Playback from terminal 1 can not interrupt capture from  terminal 2.

**Environment**
Sof: f7809a2e on topic/sof-master
Kernel: f745362c on topic/sof-dev
platform: apl-up2 & BYT

**log file**
[dmesg.log](https://github.com/thesofproject/sof/files/3440579/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3440580/logger.log)

tlauda, 2019-08-08:
@Jiangxinx I cannot reproduce this bug with current master 62b59ce76244b6d8c7745dfc1ee889b4669c6ceb. Could you confirm whether it's still there?

dengyangchao, 2019-08-20:
issue can reproduce on **GLK-Chrome** 

**To reproduce**
1. arecord -Dhw:0,1 -r48000 -c2 -fs16_le 1.wav -vv -i
2. aplay -Dhw:0,1 -r48000 -c2 -fs16_le /dev/zero -vv -i
3. pause-release arecord
4. use "ctrl+c" to stop aplay
5. arecord shows i/o error

**Frequency**
100%

**Test recipe**
Firmware: 5476ed7 https://github.com/thesofproject/sof/commits/master
Kernel: a600576c https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-glk-da7219.tplg same as firmware

[dmesg.log](https://github.com/thesofproject/sof/files/3519493/dmesg.log)

YvonneYang2, 2019-08-21:
> issue can reproduce on **GLK-Chrome**
> 
> **To reproduce**
> 
> 1. arecord -Dhw:0,1 -r48000 -c2 -fs16_le 1.wav -vv -i
> 2. aplay -Dhw:0,1 -r48000 -c2 -fs16_le /dev/zero -vv -i
> 3. pause-release arecord
> 4. use "ctrl+c" to stop aplay
> 5. arecord shows i/o error

Issue as https://github.com/thesofproject/sof/issues/1677#issuecomment-522917198 said on glk cannot be reproduced when apply patch https://github.com/thesofproject/linux/pull/1070

Jiangxinx, 2019-08-21:
Both #1070  and #1722 can not work for this issue on byt rt5682 platform.

PR #1757 (2019-08-26): byt: fix simultaneous playback and capture
Fixes simultaneous playback and capture on byt/cht platforms.
It's a temporary fix and will be replaced with the proper solution
soon.

Fixes #1677.
closed 2019-08-26 by tlauda