thesofproject/sof #1680: [FEATURE] Tools - Need FW binary release script

lgirdwood, 2019-07-29:
**Is your feature request related to a problem? Please describe.**
A script is needed to take FW binaries and topologies (with version/git tag suffix information) and copy them to correct directories within the Linux FW repo. This will include creating any soft links to reflect the correct version (so that driver stings dont need to be changed).

FW binaries would also have to be copied into correct directory depending on key used to sign FW.

**Describe the solution you'd like**
script with some flags

lgirdwood, 2019-07-29:
@mengdonglin I will do this manually for v1.3, but this needs @xiulipan to create something here before v1.4

lgirdwood, 2020-01-27:
@mengdonglin @fkaczmar @lbetlej lets move this into v1.6 territory since we are still discussing the binary packaging with other maintainers.

lgirdwood, 2020-08-18:
@xiulipan any update. I understand we have merged a few updates now ?

xiulipan, 2020-08-19:
@lgirdwood @mengdonglin We did not got the binary from release. Do we need to add some in previous release?

lgirdwood, 2020-08-20:
I think this is done now in sof-bin.

closed 2020-08-20 by lgirdwood