thesofproject/sof #1686: [FEATURE] Tools - needs FW binary introspection script

lgirdwood, 2019-07-30:
**Is your feature request related to a problem? Please describe.**
A script is needed that can read FW binaries and dump manifest and FW boot metadata.

**Describe the solution you'd like**
A simple python script.

jajanusz, 2019-07-31:
@mmaka1 @lgirdwood It looks like https://github.com/thesofproject/sof-ecl-plugins is already capable of it. I don't like that duplication of tools. Also this one is going to have some parts that could be common with rimage. I guess there will be more tools coming like this. Maybe we should rewrite rimage to python, this way f.e. all manifest/metadata code will be shared between tools (and even ecl plugin would be able to execute them if you gonna embed jython there, what I'd do anyway if we are going to add Kconfig capability to ecl plugin).

jajanusz, 2019-12-13:
Added in https://github.com/thesofproject/sof/pull/2160

closed 2019-12-13 by jajanusz