thesofproject/sof #1690: [BUG] XRUN during stop-release

aiChaoSONG, 2019-07-31:
**Describe the bug**
Resuming Playback/Capture from stop will cause underrun/overrun, and the aplay/arecord never continues playing/recording, it seems that they are waiting for something.

**To Reproduce**
1. run command `aplay -Dplughw:0,0 -c2 -f dat -i -vvv Titanium_20.wav`. or `arecord -Dplughw:0,0 -c2 -fdat -i -vvv  headset.wav` 
2. Press `CTRL + Z`.
3. Press `fg`.

**Reproduction Rate**
100% on CML Chrome with Ubuntu OS.

**Expected behavior**
aplay and arecord should continue playing or recording without xrun.

**Impact**
showstopper

**Environment**
1) Firmware: cml-007-drop-stable    `9dfb699a`
2) Kernel: topic/sof-dev     `eaa1eb1c`
3)Topology: sof-cml-rt5682-max98357a.tplg
4) CML Chrome

aiChaoSONG, 2019-08-01:
This issue persists with **sof-dev kernel`eaa1eb1c`** and  **master sof`67a48ca8`** today(Aug 1st).

tlauda, 2019-08-01:
@aiChaoSONG Provide FW logs please.

aiChaoSONG, 2019-08-02:
This bug can only be reproduced with aplay/arecord running under zsh,  under bash, running aplay/arecord will cause xrun,  but apaly/arecord can resume, and xrun has no influence on audio quality.

closed 2019-08-02 by aiChaoSONG