thesofproject/sof #1692: [BUG][WHL]Arecord hung after trigger successfully.

YvonneYang2, 2019-07-31:
**Describe the bug**
Arecord will hang after trigger it. Only reboot can recover the issue.

**To Reproduce**
1. Run command "arecord -Dhw:0,8 -M -N -r 16000 -c2 -f s24_le -vvv 1.wav"
2. Trigger it.

**Reproduction Rate**
80%

**Expected behavior**
Arecord successfully after trigger arecord.

**Impact**
Arecord hung after trigger it

**Environment**
Firmware: 442cfb2 https://github.com/thesofproject/sof/commits/cml-007-drop-stable
Kernel: f267c3d https://github.com/thesofproject/linux/commits/topic/keyword_detect
Topology:file: sof-cml-rt5682.tplg link from sof-whl-rt5682-kwd.tplg
platform: whl-i2s

**logger**
```
    0      2       VOLUME          66817883.854167         4.427083 udio/volume/volume.c:405 	volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 0, value = 65536
    0      2       VOLUME          66817888.854167         5.000000 udio/volume/volume.c:405 	volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 1, value = 65536
    0      2          IPC          66818155.364583       266.510406    src/ipc/handler.c:628 	ipc: pm -> restore
    0      2          IPC          66819229.010417      1073.645874    src/ipc/handler.c:221 	ipc: comp 34 -> params
    0      2         PIPE 8.41     66819262.239583        33.229168 src/audio/pipeline.c:315 	pipeline_params()
    0      2         HOST          66819267.760417         5.520833     src/audio/host.c:502 	host_params()
    0      2          DMA          66819277.552083         9.791667 intel/cavs/hda-dma.c:462 	hda-dmac: 4 channel 1 -> get
    0      2          DMA          66819283.437500         5.885417 intel/cavs/hda-dma.c:687 	hda-dmac: 4 channel 1 -> config
    0      2       VOLUME          66819292.343750         8.906250 udio/volume/volume.c:247 	volume_params()
    0      2          DAI 8.40     66819297.968750         5.625000      src/audio/dai.c:356 	dai_params()
    0      2         PIPE 8.41     66819306.927083         8.958333 src/audio/pipeline.c:379 	pipeline_prepare()
    0      2         HOST          66819312.031250         5.104167     src/audio/host.c:632 	host_prepare()
    0      2          KPB          66819329.166667        17.135416      src/audio/kpb.c:338 	kpb_prepare()
    0      2          KPB          66819334.635417         5.468750      src/audio/kpb.c:178 	kpb_allocate_history_buffer()
    0      1          MEM          66820020.729167       686.093750      src/lib/alloc.c:255 	error: 171 blocks needed for allocation but only 170 blocks are remaining
    0      2          KPB          66820041.093750        20.364584      src/audio/kpb.c:204 	kpb new memory block: 65280
    0      2          KPB          66820101.250000        60.156250      src/audio/kpb.c:204 	kpb new memory block: 203520
    0      2          KPB          66820105.520833         4.270833      src/audio/kpb.c:251 	kpb_allocate_history_buffer(): allocated 268800 bytes
    0      2          KPB          66820110.260417         4.739583      src/audio/kpb.c:1137	kpb_clear_history_buffer()
    0      2       VOLUME          66820692.083333       581.822937 udio/volume/volume.c:590 	volume_prepare()
    0      2          DAI 8.40     66820716.354167        24.270834      src/audio/dai.c:420 	dai_prepare()
    0      2          IPC          66821272.604167       556.250000    src/ipc/handler.c:221 	ipc: comp 42 -> params
    0      2         PIPE 9.47     66821295.833333        23.229166 src/audio/pipeline.c:315 	pipeline_params()
    0      2     SELECTOR          66821302.656250         6.822917 /selector/selector.c:164 	selector_params()
    0      2         PIPE 9.47     66821308.229167         5.572917 src/audio/pipeline.c:379 	pipeline_prepare()
    0      2      KEYWORD          66821313.072917         4.843750 /audio/detect_test.c:704 	test_keyword_prepare()
    0      2     SELECTOR          66821328.333333        15.260417 /selector/selector.c:350 	selector_prepare()
    0      2     SELECTOR          66821333.385417         5.052083 /selector/selector.c:379 	selector_prepare(): source->params.channels = 2
```
[dmesg.txt](https://github.com/thesofproject/sof/files/3450986/dmesg.txt)
[logger.log](https://github.com/thesofproject/sof/files/3450987/logger.log)

mrajwa, 2019-07-31:
@YvonneYang2, logs look preatty fine. If that issue happens, you presumably terminate arecord and start again - now what happens? Are you able to trigger again and record with no problem or it no longer works.

PS: I can not reproduce this issue on neither CML nor WHL

YvonneYang2, 2019-08-01:
Retest it again, when issue happened, terminate arecord and start again cannot recover it, will still reproduce the issue after trigger it.(logs please see again_xxx.log) Only reboot can recover it. 

Logs as below 
[dmesg.log](https://github.com/thesofproject/sof/files/3454605/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3454606/logger.log)
[logger_error.log](https://github.com/thesofproject/sof/files/3454607/logger_error.log)
[again_dmesg.log](https://github.com/thesofproject/sof/files/3454608/again_dmesg.log)
[again_logger.log](https://github.com/thesofproject/sof/files/3454609/again_logger.log)
[again_logger_error.log](https://github.com/thesofproject/sof/files/3454611/again_logger_error.log)

mrajwa, 2019-08-01:
Logs, again are clear. Can you try, before test run 

> echo on > /sys/bus/pci/devices/0000\:00\:1f.3/power/control 

and send logs again if they reveal sth new.

mrajwa, 2019-08-27:
@YvonneYang2 @keqiaozhang I think we can close this one?

keqiaozhang, 2019-08-28:
Confirmed that this issue is caused by a known issue with Host DMA(more like a hardware issue) and can be fixed by resetting or disabling the LI support("L1 Support Enable(L1SEN)" bit). So this is not a sof issue, let's close it.

closed 2019-08-28 by keqiaozhang