thesofproject/sof #1700: [BUG][WHL]volume adjust not work when change volume based on sof-cml-demux-rt5682 tplg

YvonneYang2, 2019-08-02:
**Describe the bug**
Volume cannot be changed during playback when using sof-cml-demux-rt5682 tplg.

**To Reproduce**
1. Aplay with command "aplay -Dhw:0,0 -r48000 -c2 -fs16_le California_Dreaming_48k_16bit.wav -vv -i"
2. Open alsamixer, adjust PGA1.1 1 volume.

**Reproduction Rate**
100%

**Expected behavior**
Wav volume should be changed

**Impact**
Wav volume cannot be changed.

**Environment**
Kernel: f267c3d https://github.com/thesofproject/linux/commits/topic/keyword_detect
Firmware: 442cfb2 https://github.com/thesofproject/sof/commits/cml-007-drop-stable
tplg : sof-cml-rt5682-kwd.tplg link from sof-cml-demux-rt5682.tplg
platform: whl-i2s

**dmesg**
```
[  126.749086] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[  126.749099] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  126.749327] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  126.749339] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  126.749571] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  126.749585] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  126.749810] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  126.749821] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  126.750040] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  126.750053] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50030000: GLB_COMP_MSG: SET_DATA
[  126.750151] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x50030000 size 125
[  126.757550] sof-audio-pci 0000:00:1f.3: error: set/get ctrl ipc comp 9
[  126.764076] sof-audio-pci 0000:00:1f.3: error: failed kcontrol value set for widget: 2
[  126.771975] sof-audio-pci 0000:00:1f.3: error: restoring kcontrols after resume
[  126.779265] sof-audio-pci 0000:00:1f.3: error: failed to restore pipeline after resume -22
[  126.787509] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 0
[  126.787510] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[  126.787511] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[  126.787512] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[  126.787868] sof-audio-pci 0000:00:1f.3: rate_min: 48000 rate_max: 48000
[  126.787869] sof-audio-pci 0000:00:1f.3: channels_min: 2 channels_max: 2
[  126.895787] sof-audio-pci 0000:00:1f.3: rate_min: 48000 rate_max: 48000
[  126.895789] sof-audio-pci 0000:00:1f.3: channels_min: 2 channels_max: 2
[  126.895973] sof-audio-pci 0000:00:1f.3: rate_min: 48000 rate_max: 48000
[  126.895974] sof-audio-pci 0000:00:1f.3: channels_min: 2 channels_max: 2
[  126.895978] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 0 dir 0
[  126.895980] sof-audio-pci 0000:00:1f.3: generating page table for 00000000310cc214 size 0xffc0 pages 16
[  126.895987] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  126.896005] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  126.896009] sof-audio-pci 0000:00:1f.3: period_bytes:0x3fc0
[  126.896010] sof-audio-pci 0000:00:1f.3: periods:4
[  126.896020] sof-audio-pci 0000:00:1f.3: stream_tag 1
[  126.896024] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  126.896198] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  126.896200] sof-audio-pci 0000:00:1f.3: pcm: stream dir 0, posn mailbox offset is 790528
[  126.896913] div[0]=1
[  126.896914] div[1]=2
[  126.923153] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[  126.923392] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x14001e successful
[  126.923398] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  126.923552] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  127.247834] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  127.247873] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  127.747819] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  127.747856] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  157.179972] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[  157.180526] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20140000 successful
[  157.180536] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  157.180881] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  157.180890] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  157.181024] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  157.181267] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
[  157.747785] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  157.747826] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
```

[dmesg.log](https://github.com/thesofproject/sof/files/3460403/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3460404/logger.log)

aiChaoSONG, 2019-10-16:
issue can be produced on Hatch with sof-cml-demux-rt5682.tplg and sof-cml-demux-rt5682-max98357a.tplg

wenqingfu, 2019-10-24:
@fkaczmar to reassign

tlauda, 2019-10-25:
@wenqingfu @aiChaoSONG #2009 should fix this problem. Remember that for whl platform you need to link sof-whl-demux topology. CML and WHL are using different SSPs.

aiChaoSONG, 2019-10-25:
I will check this issue with your PR on Monday, better to be merged, if everything's fine, I will close this issue.

tlauda, 2019-10-30:
@aiChaoSONG Have you confirmed it's working now?

dengyangchao, 2019-10-31:
tried  #2009 on CML Chrome
boot up the device and occur DSP panic, also hw params install failed

TPLG: sof-cml-demux-rt5682-max98357a.tplg  rename to sof-cml-rt5682-max98357a.tplg

<details>
 <summary>Dmesg</summary>

````
[   40.132058] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 3 dir 0
[   40.132062] sof-audio-pci 0000:00:1f.3: generating page table for 0000000003363ba9 size 0x4b00 pages 5
[   40.132076] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   40.132100] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   40.132106] sof-audio-pci 0000:00:1f.3: period_bytes:0x12c0
[   40.132109] sof-audio-pci 0000:00:1f.3: periods:4
[   40.132127] sof-audio-pci 0000:00:1f.3: stream_tag 2
[   40.132139] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   40.132250] sof-audio-pci 0000:00:1f.3: error : DSP panic!
[   40.132266] sof-audio-pci 0000:00:1f.3: panic: dsp_oops_offset 788480 offset 788480
[   40.132284] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   40.132390] sof-audio-pci 0000:00:1f.3: error: runtime exception
[   40.132396] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[   40.132402] sof-audio-pci 0000:00:1f.3: error: panic at :0
[   40.132408] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[   40.132415] sof-audio-pci 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[   40.132422] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x00000044 PS       0x00060b25 SAR     0x00000000
[   40.132429] sof-audio-pci 0000:00:1f.3: EPC1     0xbe00ee9c EPC2     0xbe01cce5 EPC3     0x00000000 EPC4    0x00000000
[   40.132435] sof-audio-pci 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   40.132442] sof-audio-pci 0000:00:1f.3: EPS2     0x00060b20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[   40.132448] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[   40.132454] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe05eb30
[   40.132471] sof-audio-pci 0000:00:1f.3: 0xbe05eb30: be00ee9c 00000000 00000000 00000000
[   40.132488] sof-audio-pci 0000:00:1f.3: 0xbe05eb34: 00000000 00000000 be056300 be05ee20
[   40.132500] sof-audio-pci 0000:00:1f.3: 0xbe05eb38: c382ba00 f886f1ae 9195b828 ffffa442
[   40.132508] sof-audio-pci 0000:00:1f.3: 0xbe05eb3c: 9447d500 ffffa442 93fc4a00 ffffa442
[   40.132516] sof-audio-pci 0000:00:1f.3: 0xbe05eb40: a76bbb12 ffffffff 00000000 00000000
[   40.132523] sof-audio-pci 0000:00:1f.3: 0xbe05eb44: 92b89c00 ffffa442 9447d500 ffffa442
[   40.132530] sof-audio-pci 0000:00:1f.3: 0xbe05eb48: 00000000 00000000 00000000 00000000
[   40.132537] sof-audio-pci 0000:00:1f.3: 0xbe05eb4c: 00000000 00000000 00000000 00000000
[   40.132542] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   40.639428] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60010000 size 108
[   40.639439] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[   40.639449] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   40.639549] sof-audio-pci 0000:00:1f.3: error: runtime exception
[   40.639553] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[   40.639557] sof-audio-pci 0000:00:1f.3: error: panic at :0
[   40.639562] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[   40.639566] sof-audio-pci 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[   40.639572] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x00000044 PS       0x00060b25 SAR     0x00000000
[   40.639577] sof-audio-pci 0000:00:1f.3: EPC1     0xbe00ee9c EPC2     0xbe01cce5 EPC3     0x00000000 EPC4    0x00000000
[   40.639582] sof-audio-pci 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   40.639587] sof-audio-pci 0000:00:1f.3: EPS2     0x00060b20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[   40.639591] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[   40.639595] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe05eb30
[   40.639603] sof-audio-pci 0000:00:1f.3: 0xbe05eb30: be00ee9c 00000000 00000000 00000000
[   40.639608] sof-audio-pci 0000:00:1f.3: 0xbe05eb34: 00000000 00000000 be056300 be05ee20
[   40.639614] sof-audio-pci 0000:00:1f.3: 0xbe05eb38: daab2d00 91f99033 9195b828 ffffa442
[   40.639619] sof-audio-pci 0000:00:1f.3: 0xbe05eb3c: 80e939e8 ffffbab7 00000000 00000000
[   40.639624] sof-audio-pci 0000:00:1f.3: 0xbe05eb40: 933289e8 ffffa442 933289f0 ffffa442
[   40.639629] sof-audio-pci 0000:00:1f.3: 0xbe05eb44: 92f95400 ffffa442 9195b828 ffffa442
[   40.639634] sof-audio-pci 0000:00:1f.3: 0xbe05eb48: 84de6a00 ffffa442 daab2d00 0000006c
[   40.639639] sof-audio-pci 0000:00:1f.3: 0xbe05eb4c: 00000000 00000000 00000000 00000000
[   40.639656] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[   40.639660] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   40.639672] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   40.639677] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[... cut ...]
    0      2          DMA          67118600.781250         5.885417 intel/cavs/hda-dma.c:587 	hda-dmac: 5 channel 1 -> config
    0      2       VOLUME          67118608.645833         7.864583 udio/volume/volume.c:244 	volume_params()

````
</details>

[amixer.txt](https://github.com/thesofproject/sof/files/3792385/amixer.txt)
[dmesg.log](https://github.com/thesofproject/sof/files/3792386/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3792387/logger.log)
[logger-error.log](https://github.com/thesofproject/sof/files/3792388/logger-error.log)

tlauda, 2019-10-31:
@dengyangchao Original issue have been reported with sof-cml-demux-rt5682.tplg (without max98357a). Please verify with such topology. For this particular problem you have separate issue #1973.

aiChaoSONG, 2019-10-31:
@dengyangchao  The issue you come up with may be causes by the speaker pipeline,  please retry with `sof-cml-demux-rt5682.tplg` rather than `sof-cml-demux-rt5682-max98357a.tplg`.  And please close this issue if it is fixed.

dengyangchao, 2019-10-31:
@tlauda tried #2009 and tplg use:```sof-cml-demux-rt5682.tplg ``` 
and aplay via"aplay -Dhw:0,0 -fs16_le -c2 -r 48000 -vv -i wav/48California_Dreaming.wav "
there is no sound output, also can't detect headset,  and sound setting shows" Dummy Output"

add info:

```
numid=51,iface=MIXER,name='Headphone Jack Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=off
```

```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1           SA         170226371.458333 170226368.000000      src/lib/agent.c:57  	validate(), ll drift detected, delta = 27073
```

PCM List
```
**** List of PLAYBACK Hardware Devices ****
card 0: sofsofrt5682 [sof-sof_rt5682], device 0: Port1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsofrt5682 [sof-sof_rt5682], device 2: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsofrt5682 [sof-sof_rt5682], device 3: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsofrt5682 [sof-sof_rt5682], device 4: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```

[amixer.txt](https://github.com/thesofproject/sof/files/3792862/amixer.txt)
[dmesg.log](https://github.com/thesofproject/sof/files/3792864/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3792865/logger.log)
[logger-error.log](https://github.com/thesofproject/sof/files/3792866/logger-error.log)
[pcm-list.txt](https://github.com/thesofproject/sof/files/3792867/pcm-list.txt)

PR #2009 (2019-11-04): demux: return config on COMP_CMD_GET_DATA
Returns current config on COMP_CMD_GET_DATA. It fixes the problem
with wrong demux configuration after suspend. Kernel doesn't cache
initial kcontrol config and overwrites it with configuration returned
from component.

Fixes #1700.