thesofproject/sof #1708: [BUG] Unable to reset the priority and core tokens in tplg

keqiaozhang, 2019-08-07:
**Describe the bug**
I tried to reset the priority and DSP coreID tokens to test the smp support, but failed to change them. The priority and core are still "0"  in conf file after changing the pipeline to different priority and core ID.

**To Reproduce**
eg:
1. Change the pipeline 1 as priority 1 and dispatch it to core 1
```
diff --git a/tools/topology/sof-apl-pcm512x.m4 b/tools/topology/sof-apl-pcm512x.m4
index 66c0047d23da..dbf136387811 100644
--- a/tools/topology/sof-apl-pcm512x.m4
+++ b/tools/topology/sof-apl-pcm512x.m4
@@ -36,7 +36,7 @@ dnl     frames, deadline, priority, core)
 # Schedule 48 frames per 1000us deadline on core 0 with priority 0
 PIPELINE_PCM_ADD(sof/pipe-low-latency-playback.m4,
        1, 0, 2, s32le,
-       48, 1000, 0, 0)
+       48, 1000, 1, 1)

```
2. build the tplg and check the conf file
```
#
# DAI pipeline - always use 0 for DAIs
#
SectionVendorTuples."PIPELINE.1.SSP5.OUT_tuples" {
        tokens "sof_sched_tokens"
        tuples."word" {
                SOF_TKN_SCHED_PERIOD            "1000"
                SOF_TKN_SCHED_PRIORITY          "0"
                SOF_TKN_SCHED_CORE              "0"
                SOF_TKN_SCHED_FRAMES            "48"
                SOF_TKN_SCHED_TIME_DOMAIN       ""
        }
}
```
3.  priority and dsp coreID cannot be changed

**Reproduction Rate**
100%

**Expected behavior**
The priority of pipeline can be changed and pipeline can be dispatched to other dspcore.

**Impact**
smp doesn't support.

**Environment**
1) SOF: master/9e2aa3d3acf

paulstelian97, 2019-08-07:
From a manual run through the m4, it appears that the section you wanted to see the priority changes comes from DAI_ADD near line 70, not where you change it. Indeed, if I change the scheduling parameters of THAT call I will notice them in the .conf file.

Usually scheduling doesn't come from the PCMs but from the actual hardware interface, I guess.

keqiaozhang, 2019-08-07:
Thanks @paulstelian97 
I just realized it. only change the priority and core in DAI_ADD will take effect. These 2 tokens in PCM_ADD seem useless.

juimonen, 2019-08-07:
@keqiaozhang so close this one?

keqiaozhang, 2019-08-08:
@juimonen . This issue can be closed.

closed 2019-08-08 by keqiaozhang