thesofproject/sof #1719: [BUG] IPC error when stopping the WoV pipeline

keqiaozhang, 2019-08-09:
**Describe the bug**
Each time when stopping the WoV pipeline, there're some IPC errors show in dmesg and trace logger, but  no functional impact. 

**To Reproduce**
1. arecord -Dhw:0,8 -M -N -r 16000 -c2 -f s24_le -vvv -d 10 wov.wav
2. after that, check the dmesg and trace.

**Reproduction Rate**
100%

**Expected behavior**
no IPC error when trigger stop.

**Impact**
more like a warning, no functional impact.

**Environment**
1) driver: topic/keyword_detect /6664b4
2) firmware: origin/cml-007-drop-stable /0207c7
3)  tplg:sof-cml-rt5682-max98357a.tplg

** dmesg **
```
[  769.630427] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  769.630692] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60050000 size 12
[  769.630702] sof-audio-pci 0000:00:1f.3: error: failed to trigger DETECT9.0
[  769.630707] sof-audio-pci 0000:00:1f.3: error: failed to trigger widget DETECT9.0
[  769.630716] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
```
** trace **
```
   0      2          KPB         7804142457.500000         5.260417      src/audio/kpb.c:321   kpb_trigger()
    0      1         COMP         7804142462.239584         4.739583 rc/audio/component.c:164   comp_set_state() error: wrong state = 1, COMP_TRIGGER_STOP
    0      1          IPC         7804142469.583334         7.343750 src/audio/pipeline.c:635   pipeline_trigger() error: ret = -22, host->comp.id = 50, cmd = 0
    0      1          IPC         7804142477.187500         7.604167    src/ipc/handler.c:484   ipc: comp 50 trigger 0x50000 failed -22
    0      2          IPC         7804142779.114584       301.927094    src/ipc/handler.c:339   ipc: comp 50 -> free
```

[dmesg.txt](https://github.com/thesofproject/sof/files/3485625/dmesg.txt)
[logger-t.txt](https://github.com/thesofproject/sof/files/3485631/logger-t.txt)

lgirdwood, 2019-08-19:
@mrajwa @keyonjie fyi - does this mean anything to you guys ?

mrajwa, 2019-08-22:
@lgirdwood, @keqiaozhang yes I am aware of it. The problem is that sometimes we stop KWD pipelines when it is performing its long-last tasks (draining is a good example here). During such tasks KWD pipeline can not be reset right away. Instead we record reset request, stop current task gracefully, and finally reset. The source of IPC error is the busy state we return. I will try to come up with more "warning-like" approach.

lgirdwood, 2019-08-30:
@mrajwa ok, it sounds like we can also imrpove the error handling too. I.e. we can return -EAGAIN so that host SW knows to try again (and give time for draining to complete).

wwittbrx, 2019-10-25:
This is fixed on 89ef0af07612c0bcf1a85fec3871d0c31e60087b, closing

closed 2019-10-25 by wwittbrx