thesofproject/sof #1723: [BUG][BYT] Can not hear  the sound on byt rt5682 platform.

Jiangxinx, 2019-08-13:
**Describe the bug**
Can not hear  the sound on byt rt5682 platform.

**To Reproduce**
aplay -Dhw:0,0 -fS16_LE -c2 -r 48000 audio-files/raw/test-c2-r48000-fs16_le-d60.raw -vv -i

**Reproduction Rate**
100%

**Environment**
Firmware: c6f1b08f https://github.com/thesofproject/sof/commits/master
Kernel: 4730d842  https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: sof-byt-rt5682.tplg

**alsamixer**
```
state.sofsofrt5682 {
	control.1 {
		iface MIXER
		name 'DAC1 Playback Volume'
		value.0 87
		value.1 87
		comment {
			access 'read write'
			type INTEGER
			count 2
			range '0 - 86'
			dbmin -6525
			dbmax -75
			dbvalue.0 0
			dbvalue.1 0
		}
	}
	control.2 {
		iface MIXER
		name 'CBJ Boost Volume'
		value 0
		comment {
			access 'read write'
			type INTEGER
			count 1
			range '0 - 8'
			dbmin 0
			dbmax 5200
			dbvalue.0 0
		}
	}
	control.3 {
		iface MIXER
		name 'STO1 ADC Capture Switch'
		value.0 false
		value.1 false
		comment {
			access 'read write'
			type BOOLEAN
			count 2
		}
	}
	control.4 {
		iface MIXER
		name 'STO1 ADC Capture Volume'
		value.0 23
		value.1 23
		comment {
			access 'read write'
			type INTEGER
			count 2
			range '0 - 63'
			dbmin -1725
			dbmax 3000
			dbvalue.0 0
			dbvalue.1 0
		}
	}
	control.5 {
		iface MIXER
		name 'STO1 ADC Boost Gain Volume'
		value.0 2
		value.1 2
		comment {
			access 'read write'
			type INTEGER
			count 2
			range '0 - 3'
			dbmin 0
			dbmax 3600
			dbvalue.0 2400
			dbvalue.1 2400
		}
	}
	control.6 {
		iface MIXER
		name 'Headphone Jack Switch'
		value true
		comment {
			access 'read write'
			type BOOLEAN
			count 1
		}
	}
	control.7 {
		iface MIXER
		name 'Headset Mic Switch'
		value true
		comment {
			access 'read write'
			type BOOLEAN
			count 1
		}
	}
	control.8 {
		iface MIXER
		name 'Spk Switch'
		value true
		comment {
			access 'read write'
			type BOOLEAN
			count 1
		}
	}
	control.9 {
		iface MIXER
		name 'PGA1.0 1 PCM 0 Playback Volume'
		value.0 32
		value.1 32
		comment {
			access 'read write'
			type INTEGER
			count 2
			range '0 - 32'
			dbmin -9999999
			dbmax 0
			dbvalue.0 0
			dbvalue.1 0
		}
	}
	control.10 {
		iface MIXER
		name 'PGA1.1 1 Master Playback Volume'
		value.0 32
		value.1 32
		comment {
			access 'read write'
			type INTEGER
			count 2
			range '0 - 32'
			dbmin -9999999
			dbmax 0
			dbvalue.0 0
			dbvalue.1 0
		}
	}
	control.11 {
		iface MIXER
		name 'PGA2.0 2 PCM 0 Capture Volume'
		value.0 40
		value.1 40
		comment {
			access 'read write'
			type INTEGER
			count 2
			range '0 - 40'
			dbmin -9999999
			dbmax 1600
			dbvalue.0 1600
			dbvalue.1 1600
		}
	}
	control.12 {
		iface MIXER
		name 'PGA3.0 3 PCM 1 Playback Volume'
		value.0 32
		value.1 32
		comment {
			access 'read write'
			type INTEGER
			count 2
			range '0 - 32'
			dbmin -9999999
			dbmax 0
			dbvalue.0 0
			dbvalue.1 0
		}
	}
	control.13 {
		iface MIXER
		name 'RECMIX1L CBJ Switch'
		value true
		comment {
			access 'read write'
			type BOOLEAN
			count 1
		}
	}
	control.14 {
		iface MIXER
		name 'Stereo1 ADC L1 Mux'
		value 'DAC MIX'
		comment {
			access 'read write'
			type ENUMERATED
			count 1
			item.0 'DAC MIX'
			item.1 ADC
		}
	}
	control.15 {
		iface MIXER
		name 'Stereo1 ADC R1 Mux'
		value 'DAC MIX'
		comment {
			access 'read write'
			type ENUMERATED
			count 1
			item.0 'DAC MIX'
			item.1 ADC
		}
	}
	control.16 {
		iface MIXER
		name 'Stereo1 ADC L2 Mux'
		value 'DAC MIX'
		comment {
			access 'read write'
			type ENUMERATED
			count 1
			item.0 'DAC MIX'
			item.1 DMIC
		}
	}
	control.17 {
		iface MIXER
		name 'Stereo1 ADC R2 Mux'
		value 'DAC MIX'
		comment {
			access 'read write'
			type ENUMERATED
			count 1
			item.0 'DAC MIX'
			item.1 DMIC
		}
	}
	control.18 {
		iface MIXER
		name 'Stereo1 ADC L Mux'
		value 'ADC1 L'
		comment {
			access 'read write'
			type ENUMERATED
			count 1
			item.0 'ADC1 L'
			item.1 'ADC1 R'
		}
	}
	control.19 {
		iface MIXER
		name 'Stereo1 ADC R Mux'
		value 'ADC1 R'
		comment {
			access 'read write'
			type ENUMERATED
			count 1
			item.0 'ADC1 L'
			item.1 'ADC1 R'
		}
	}
	control.20 {
		iface MIXER
		name 'IF1_ADC Mux'
		value 'Slot 0'
		comment {
			access 'read write'
			type ENUMERATED
			count 1
			item.0 'Slot 0'
			item.1 'Slot 2'
			item.2 'Slot 4'
			item.3 'Slot 6'
		}
	}
	control.21 {
		iface MIXER
		name 'Stereo1 ADC MIXL ADC1 Switch'
		value true
		comment {
			access 'read write'
			type BOOLEAN
			count 1
		}
	}
	control.22 {
		iface MIXER
		name 'Stereo1 ADC MIXL ADC2 Switch'
		value true
		comment {
			access 'read write'
			type BOOLEAN
			count 1
		}
	}
	control.23 {
		iface MIXER
		name 'Stereo1 ADC MIXR ADC1 Switch'
		value true
		comment {
			access 'read write'
			type BOOLEAN
			count 1Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html
		}
	}
	control.24 {
		iface MIXER
		name 'Stereo1 ADC MIXR ADC2 Switch'
		value true
		comment {
			access 'read write'
			type BOOLEAN
			count 1
		}
	}
	control.25 {
		iface MIXER
		name 'IF1 01 ADC Swap Mux'
		value L/R
		comment {
			access 'read write'
			type ENUMERATED
			count 1
			item.0 L/R
			item.1 R/L
			item.2 L/L
			item.3 R/R
		}
	}
	control.26 {
		iface MIXER
		name 'IF1 23 ADC Swap Mux'
		value L/R
		comment {
			access 'read write'
			type ENUMERATED
			count 1
			item.0 L/R
			item.1 R/L
			item.2 L/L
			item.3 R/R
		}
	}
	control.27 {
		iface MIXER
		name 'IF1 45 ADC Swap Mux'
		value L/R
		comment {
			access 'read write'
			type ENUMERATED
			count 1
			item.0 L/R
			item.1 R/L
			item.2 L/L
			item.3 R/R
		}
	}
	control.28 {
		iface MIXER
		name 'IF1 67 ADC Swap Mux'
		value L/R
		comment {
			access 'read write'
			type ENUMERATED
			count 1
			item.0 L/R
			item.1 R/L
			item.2 L/L
			item.3 R/R
		}
	}
	control.29 {
		iface MIXER
		name 'IF2 ADC Swap Mux'
		value L/R
		comment {
			access 'read write'
			type ENUMERATED
			count 1
			item.0 L/R
			item.1 R/L
			item.2 L/L
			item.3 R/R
		}
	}
	control.30 {
		iface MIXER
		name 'ADCDAT Mux'
		value ADCDAT1
		comment {
			access 'read write'
			type ENUMERATED
			count 1
			item.0 ADCDAT1
			item.1 ADCDAT2
		}
	}
	control.31 {
		iface MIXER
		name 'DAC1 MIXL Stereo ADC Switch'
		value false
		comment {
			access 'read write'
			type BOOLEAN
			count 1
		}
	}
	control.32 {
		iface MIXER
		name 'DAC1 MIXL DAC1 Switch'
		value true
		comment {
			access 'read write'
			type BOOLEAN
			count 1
		}
	}
	control.33 {
		iface MIXER
		name 'DAC1 MIXR Stereo ADC Switch'
		value false
		comment {
			access 'read write'
			type BOOLEAN
			count 1
		}
	}
	control.34 {
		iface MIXER
		name 'DAC1 MIXR DAC1 Switch'
		value true
		comment {
			access 'read write'
			type BOOLEAN
			count 1
		}
	}
	control.35 {
		iface MIXER
		name 'DAC L1 Source'
		value DAC1
		comment {
			access 'read write'
			type ENUMERATED
			count 1
			item.0 'Stereo1 DAC Mixer'
			item.1 DAC1
		}
	}
	control.36 {
		iface MIXER
		name 'DAC R1 Source'
		value DAC1
		comment {
			access 'read write'
			type ENUMERATED
			count 1
			item.0 'Stereo1 DAC Mixer'
			item.1 DAC1
		}
	}
	control.37 {
		iface MIXER
		name 'Stereo1 DAC MIXL DAC L1 Switch'
		value true
		comment {
			access 'read write'
			type BOOLEAN
			count 1
		}
	}
	control.38 {
		iface MIXER
		name 'Stereo1 DAC MIXL DAC R1 Switch'
		value true
		comment {
			access 'read write'
			type BOOLEAN
			count 1
		}
	}
	control.39 {
		iface MIXER
		name 'Stereo1 DAC MIXR DAC L1 Switch'
		value true
		comment {
			access 'read write'
			type BOOLEAN
			count 1
		}
	}
	control.40 {
		iface MIXER
		name 'Stereo1 DAC MIXR DAC R1 Switch'
		value true
		comment {
			access 'read write'
			type BOOLEAN
			count 1
		}
	}
	control.41 {
		iface MIXER
		name 'HPOL Playback Switch'
		value false
		comment {
			access 'read write'
			type BOOLEAN
			count 1
		}
	}
	control.42 {
		iface MIXER
		name 'HPOR Playback Switch'
		value false
		comment {
			access 'read write'
			type BOOLEAN
			count 1
		}
	}
}
```

tlauda, 2019-08-13:
@Jiangxinx I am able to get loopback data with SSP Tx connected to the SSP Rx, so it looks like some kind of codec issue.

Jiangxinx, 2019-08-15:
@tlauda All of 5682 platform from my side have this problem, Do you know if 5682 codec platform need an additional amixer configuration?

lgirdwood, 2019-08-19:
@bardliao fyi - any obvious incorrect mixers ?

bardliao, 2019-08-20:
> @bardliao fyi - any obvious incorrect mixers ?

@Jiangxinx @lgirdwood Yes, please set below to true.
 'HPOL Playback Switch'
 'HPOR Playback Switch'

plbossart, 2019-08-20:
On 8/20/19 7:12 AM, Bard Liao wrote:
>     @bardliao <https://github.com/bardliao> fyi - any obvious incorrect
>     mixers ?
> 
> @Jiangxinx <https://github.com/Jiangxinx> @lgirdwood 
> <https://github.com/lgirdwood> Yes, please set below to true.
> 'HPOL Playback Switch'
> 'HPOR Playback Switch'

This just proves my point about only using maintained UCM files and not 
rely on asound.state...

Jiangxinx, 2019-08-21:
Passed after set: 'HPOL Playback Switch' and  'HPOR Playback Switch' to true.

closed 2019-08-21 by Jiangxinx