thesofproject/sof #1726: [BUG][BYT]DSP panic when do multiple pipeline playback test on byt nocodec platform.

Jiangxinx, 2019-08-14:
**Describe the bug**
DSP panic when do multiple pipeline playback test on byt nocodec platform.

**To Reproduce**
1.Open terminal1:  aplay -Dhw:0,0 -fS16_LE -c2 -r 48000 /dev/zero -vv -i
```
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 0 'sof-nocodec' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16368
  period_size  : 4080
  period_time  : 85000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4080
  period_event : 0
  start_threshold  : 16368
  stop_threshold   : 16368
  silence_threshold: 0
  silence_size : 0
  boundary     : 9214364837600034816
  appl_ptr     : 0
  hw_ptr       : 0
#+                                                 | 00%
```
Open terminal2:  aplay -Dhw:0,1 -fS16_LE -c2 -r 48000 /dev/zero -vv -i
```
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 0 'sof-nocodec' device 1 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 24000
  period_size  : 6000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 6000
  period_event : 0
  start_threshold  : 24000
  stop_threshold   : 24000
  silence_threshold: 0
  silence_size : 0
  boundary     : 6755399441055744000
  appl_ptr     : 0
  hw_ptr       : 0
#+                                                 | 00%
```

2.Then ctrl+c stop  terminal2's playback,and open terminal2's playback again.
terminal2:
```
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 0 'sof-nocodec' device 1 subdevice 0
[... 249 lines cut ...]
    0      1          IPC         279074331.875000        10.833333    src/ipc/handler.c:486 	ipc: comp 19 trigger 0x50000 failed -22
    0      1         COMP         279602153.697917    527821.812500 rc/audio/component.c:167 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_STOP
    0      1          IPC         279602162.239583         8.541667 src/audio/pipeline.c:639 	pipeline_trigger() error: ret = -22, host->comp.id = 1, cmd = 0
    0      1          IPC         279602173.072917        10.833333    src/ipc/handler.c:486 	ipc: comp 1 trigger 0x50000 failed -22
    0      1          SRC         285064837.395833   5462664.500000  src/audio/src/src.c:763 	No data to process.
    0      1         PIPE         285064844.947917         7.552083 src/audio/pipeline.c:789 	pipeline_copy() error: ret = -5, start->comp.id = 3, dir = 1
    0      1         PIPE 1.10    285064859.010417        14.062500 src/audio/pipeline.c:982 	pipeline_task(): xrun recover failed! pipeline will be stopped!
    0      1          IPC         285445811.979167    380952.968750    src/ipc/handler.c:1251	ipc: msg hdr for 0x600a0013 not found replace 0
    0      1          IPC         285477811.770833     31999.791016    src/ipc/handler.c:1251	ipc: msg hdr for 0x600a0013 not found replace 0
    0      1          IPC         285509811.979167     32000.208984    src/ipc/handler.c:1251	ipc: msg hdr for 0x600a0013 not found replace 0
    0      1          IPC         285541811.979167     32000.000000    src/ipc/handler.c:1251	ipc: msg hdr for 0x600a0013 not found replace 0
    0      1          SRC         286577569.635417   1035757.625000  src/audio/src/src.c:763 	No data to process.
    0      1         PIPE         286577577.135417         7.500000 src/audio/pipeline.c:789 	pipeline_copy() error: ret = -5, start->comp.id = 3, dir = 1
    0      1         PIPE 1.10    286577591.250000        14.114583 src/audio/pipeline.c:982 	pipeline_task(): xrun recover failed! pipeline will be stopped!
    0      2           SA         288268688.645833   1691097.375000      src/lib/agent.c:60  	validate(), idle longer than timeout, delta = 32452464
```

**Environment**
Firmware: c6f1b08 https://github.com/thesofproject/sof/commits/master
Kernel: b45315d5 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: sof-byt-nocodec.tplg

**log file**
[dmesg.log](https://github.com/thesofproject/sof/files/3500313/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3500314/logger.log)

YvonneYang2, 2019-08-19:
Only run command "aplay -Dhw:0,1 -r48000 -c2 -fs16_le /dev/zero -vv -i" can reproduce the issue.

**Reproduced rate**
100%

**Environment**
Firmware: ebf406d0 https://github.com/thesofproject/sof/commits/master
Kernel: a600576c https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology:file: tools/topology/sof-byt-nocodec.tplg

**dmesg**
```
[   90.270089] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   90.270391] sof-audio-acpi 80860F28:01: PCM: stream 1 dir 0 DMA position 0 DAI position 0
[   90.777296] sof-audio-acpi 80860F28:01: pcm: trigger stream 1 dir 0 cmd 0
[   90.777319] sof-audio-acpi 80860F28:01: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   90.778275] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   90.778294] sof-audio-acpi 80860F28:01: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   90.778415] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   90.778597] sof-audio-acpi 80860F28:01: pcm: close stream 1 dir 0
[   91.624193] sof-audio-acpi 80860F28:01: error : DSP panic!
[   91.624205] sof-audio-acpi 80860F28:01: panic: dsp_oops_offset 1329152 offset 1329152
[   91.624269] sof-audio-acpi 80860F28:01: error: can't enter idle
[   91.624275] sof-audio-acpi 80860F28:01: error: trace point 00000000
[   91.624280] sof-audio-acpi 80860F28:01: error: panic at .../work/sof.git/src/lib/agent.c=:61
[   91.624285] sof-audio-acpi 80860F28:01: error: DSP Firmware Oops
[   91.624292] sof-audio-acpi 80860F28:01: EXCCAUSE 0x0000003f EXCVADDR 0x00000000 PS       0x00060725 SAR     0x00000000
[   91.624298] sof-audio-acpi 80860F28:01: EPC1     0x00000000 EPC2     0x00000000 EPC3     0xff2ce8ff EPC4    0xff2ce8ff
[   91.624304] sof-audio-acpi 80860F28:01: EPC5     0xff2ce931 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   91.624310] sof-audio-acpi 80860F28:01: EPS2     0x00000000 EPS3     0x00060720 EPS4     0x00060720 EPS5    0x00060720
[   91.624316] sof-audio-acpi 80860F28:01: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x000000a2
[   91.624321] sof-audio-acpi 80860F28:01: stack dump from 0xff327c70
[   91.624330] sof-audio-acpi 80860F28:01: 0xff327c70: ff327c90 00000000 5bd6d719 00000000
[   91.624338] sof-audio-acpi 80860F28:01: 0xff327c74: 6567612f 632e746e 0000003d 00000000
[   91.624346] sof-audio-acpi 80860F28:01: 0xff327c78: 0a545e00 dbb7a5f0 345bc828 ffff9e88
[   91.624353] sof-audio-acpi 80860F28:01: 0xff327c7c: c04d9baf ffffffff 00000000 00000000
[   91.624361] sof-audio-acpi 80860F28:01: 0xff327c80: 370932a4 ffff9e88 b8ebb77c ffffffff
[   91.624369] sof-audio-acpi 80860F28:01: 0xff327c84: b8ebb860 ffffffff 0a545e00 dbb7a5f0
[   91.624376] sof-audio-acpi 80860F28:01: 0xff327c88: b8e86a6e ffffffff f6b87038 ffff9e87
[   91.624384] sof-audio-acpi 80860F28:01: 0xff327c8c: b9800205 ffffffff 00000000 00000000
[   91.624389] sof-audio-acpi 80860F28:01: error: waking up any trace sleepers
```
[dmesg.log](https://github.com/thesofproject/sof/files/3514712/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3514713/logger.log)

tlauda, 2019-08-26:
@YvonneYang2 Fixed by #1722. Could you recheck?

Jiangxinx, 2019-08-26:
@tlauda This issue still can be reproduced
```
[22762.145698] sof-audio-acpi 80860F28:01: period count 2 max 32
[22762.145701] sof-audio-acpi 80860F28:01: buffer max 8388608 bytes
[22762.146575] sof-audio-acpi 80860F28:01: rate_min: 48000 rate_max: 48000
[22762.146581] sof-audio-acpi 80860F28:01: channels_min: 2 channels_max: 2
[22762.146591] sof-audio-acpi 80860F28:01: pcm: hw params stream 1 dir 0
[22762.146598] sof-audio-acpi 80860F28:01: generating page table for 0000000066bc677a size 0x17700 pages 24
[22762.146603] sof-audio-acpi 80860F28:01: stream_tag 0
[22762.146614] sof-audio-acpi 80860F28:01: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[22762.147041] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[22762.147049] sof-audio-acpi 80860F28:01: pcm: stream dir 0, posn mailbox offset is 1329816
[22762.148351] sof-audio-acpi 80860F28:01: pcm: trigger stream 1 dir 0 cmd 1
[22762.148362] sof-audio-acpi 80860F28:01: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[22762.148460] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[22762.148603] sof-audio-acpi 80860F28:01: PCM: stream 1 dir 0 DMA position 0 DAI position 0
[22762.588095] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[22762.588118] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
[22762.657399] sof-audio-acpi 80860F28:01: pcm: trigger stream 1 dir 0 cmd 0
[22762.657422] sof-audio-acpi 80860F28:01: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[22762.658277] sof-audio-acpi 80860F28:01: error: ipc error for 0x60050000 size 12
[22762.658299]  Media Playback 1: ASoC: trigger FE failed -22
[22762.658351] sof-audio-acpi 80860F28:01: pcm: free stream 1 dir 0
[22762.658368] sof-audio-acpi 80860F28:01: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[22762.658501] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[22762.658621] sof-audio-acpi 80860F28:01: pcm: close stream 1 dir 0
[22762.700511] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 0 DMA position 11632 DAI position 448256
[22763.088251] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[22763.088273] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
[22763.208409] sof-audio-acpi 80860F28:01: pcm: trigger stream 0 dir 0 cmd 0
[22763.208432] sof-audio-acpi 80860F28:01: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[22763.208703] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[22763.208718] sof-audio-acpi 80860F28:01: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[22763.208850] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[22763.209005] sof-audio-acpi 80860F28:01: pcm: close stream 0 dir 0
[22763.588337] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[22763.588360] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
```

***Environment***
Firmware: 66da4e89 https://github.com/thesofproject/sof/commits/master
Kernel: 2a22b942 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology:file: tools/topology/sof-byt-nocodec.tplg

tlauda, 2019-08-26:
@Jiangxinx Do you still have panic? Because now it looks like problem with simultaneous streaming. Please also apply #1757 and check again.

Jiangxinx, 2019-08-27:
@tlauda I checked again on byt nocodec platform after #1757 was merged, can't reproduced the DSP panic error, however, the ipc error still exist:
```
[22762.145698] sof-audio-acpi 80860F28:01: period count 2 max 32
[22762.145701] sof-audio-acpi 80860F28:01: buffer max 8388608 bytes
[22762.146575] sof-audio-acpi 80860F28:01: rate_min: 48000 rate_max: 48000
[22762.146581] sof-audio-acpi 80860F28:01: channels_min: 2 channels_max: 2
[22762.146591] sof-audio-acpi 80860F28:01: pcm: hw params stream 1 dir 0
[22762.146598] sof-audio-acpi 80860F28:01: generating page table for 0000000066bc677a size 0x17700 pages 24
[22762.146603] sof-audio-acpi 80860F28:01: stream_tag 0
[22762.146614] sof-audio-acpi 80860F28:01: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[22762.147041] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[22762.147049] sof-audio-acpi 80860F28:01: pcm: stream dir 0, posn mailbox offset is 1329816
[22762.148351] sof-audio-acpi 80860F28:01: pcm: trigger stream 1 dir 0 cmd 1
[22762.148362] sof-audio-acpi 80860F28:01: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[22762.148460] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[22762.148603] sof-audio-acpi 80860F28:01: PCM: stream 1 dir 0 DMA position 0 DAI position 0
[22762.588095] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[22762.588118] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
[22762.657399] sof-audio-acpi 80860F28:01: pcm: trigger stream 1 dir 0 cmd 0
[22762.657422] sof-audio-acpi 80860F28:01: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[22762.658277] sof-audio-acpi 80860F28:01: error: ipc error for 0x60050000 size 12
[22762.658299]  Media Playback 1: ASoC: trigger FE failed -22
[22762.658351] sof-audio-acpi 80860F28:01: pcm: free stream 1 dir 0
[22762.658368] sof-audio-acpi 80860F28:01: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[22762.658501] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[22762.658621] sof-audio-acpi 80860F28:01: pcm: close stream 1 dir 0
[22762.700511] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 0 DMA position 11632 DAI position 448256
[22763.088251] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[22763.088273] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
[22763.208409] sof-audio-acpi 80860F28:01: pcm: trigger stream 0 dir 0 cmd 0
[22763.208432] sof-audio-acpi 80860F28:01: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[22763.208703] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[22763.208718] sof-audio-acpi 80860F28:01: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[22763.208850] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[22763.209005] sof-audio-acpi 80860F28:01: pcm: close stream 0 dir 0
[22763.588337] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[22763.588360] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
```

lgirdwood, 2019-08-30:
@Jiangxinx do things recover after the IPC error if you try again ?

Jiangxinx, 2019-09-02:
@lgirdwood Yes,it can be recover after I/O error,and this issue also can be reproduced on byt rt5682 platform today.

lgirdwood, 2019-09-06:
@tlauda looks like we are waiting on DMA emptying ?
```
    0      2          DMA          80057463.802083      1000.000000 src/drivers/dw/dma.c:1301	dw_dma_avail_data_size() size is 0! Channel enable = 0.
    0      2          DMA          80058463.802083      1000.000000 src/drivers/dw/dma.c:1301	dw_dma_avail_data_size() size is 0! Channel enable = 0.
    0      2          DMA          80059463.802083      1000.000000 src/drivers/dw/dma.c:1301	dw_dma_avail_data_size() size is 0! Channel enable = 0.
    0      2          DMA          80060463.802083      1000.000000 src/drivers/dw/dma.c:1301	dw_dma_avail_data_size() size is 0! Channel enable = 0.
    0      2          DMA          80061463.802083      1000.000000 src/drivers/dw/dma.c:1301	dw_dma_avail_data_size() size is 0! Channel enable = 0.
    0      2          DMA          80062463.802083      1000.000000 src/drivers/dw/dma.c:1301	dw_dma_avail_data_size() size is 0! Channel enable = 0.
```

juimonen, 2019-09-16:
I can reproduce this also. I don't see any obvious error's in the logs. For some reason it looks like that in the error case (start/stop stream 2 during simultaneous stream 1), the stream 1 data feed is disturbed and pipeline stops for xrun. Like the host dma from kernel->dsp is getting jammed for the existing stream->xrun->pipeline stop.

juimonen, 2019-09-16:
@tlauda so pipeline 0,1 src component _always_ gives me in the beginning: "No data to process. pipeline_task(): xrun recover failed! pipeline will be stopped!". But most of the time pipeline is not stopped, so streams are happily continuing to play... So could it be some timing with pipeline state, that every now and then the pipeline is actually able to stop -> hence we get this error? I can try to modify the src buffering or something to get rid of the message...

lgirdwood, 2019-09-16:
@juimonen I assume you are using XCC for this debug since it builds a faster SRC for the media pipeline.

juimonen, 2019-09-16:
@lgirdwood no I'm using gcc. In this use case src should not do any conversion, just copy through

juimonen, 2019-09-16:
@Jiangxinx @YvonneYang2 can you try this with https://github.com/thesofproject/sof/pull/1834
at least for me it seems to fix the issue.... sometimes this bug is pretty hard to reproduce

Jiangxinx, 2019-09-17:
@juimonen I tried #1834 on byt nocodec platform, and can't reproduce this issue.

Jiangxinx, 2019-09-18:
Fixed by #1834 .

closed 2019-09-18 by Jiangxinx

wenqingfu, 2019-09-18:
I found an occurrence in https://sof-ci.01.org/linuxpr/PR1161/build1870/devicetest/ again, but not sure the FW is with #1834 or not.

juimonen, 2019-09-18:
I think it is with #1834 as I don't see any src complain messages.  This could be another error somehow related to https://github.com/thesofproject/linux/pull/1219. I think we have some occasional jam in ending the stream, which gives input/output error in userspace. I have not found the root cause, but it seems somehow related to the draining/pm_wait... So my suspicion is that it is in upper layers from sof, but haven't figured out yet.