thesofproject/sof #1727: [BUG][CML Chrome] tplg sof-cml-rt5682-max98357a.tplg load failed

ClarexZhou, 2019-08-16:
**Describe the bug**
 tplg component load failed -12

**To Reproduce**
Boot up and check on CML chrome notebook.

**Reproduction Rate**

100%

**Expected behavior**
tplg load successfully

**Environment**

linux (topic/sof-dev):  b45315d5
sof (master): ebf406d0
tplg: sof-cml-rt5682-max98357a.tplg, same with firmware branch.
Platform: CML Chrome notebook.

Dmesg：
```
[    2.810378] sof-audio-pci 0000:00:1f.3: loaded host PCM4P
[    2.810380] sof-audio-pci 0000:00:1f.3:  config: periods snk 2 src 0 fmt 0
[    2.810385] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    2.810521] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    2.810525] sof-audio-pci 0000:00:1f.3: tplg: load control type 1 name : 6 Master Playback Volume
[    2.810531] sof-audio-pci 0000:00:1f.3: tplg: load kcontrol index 29 chans 2
[    2.810534] sof-audio-pci 0000:00:1f.3: tplg: ready widget id 29 pipe 6 type 6 name : PGA6.0 stream none
[    2.810537] sof-audio-pci 0000:00:1f.3:  config: periods snk 2 src 2 fmt 2
[    2.810542] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    2.810681] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    2.810685] sof-audio-pci 0000:00:1f.3: tplg: ready widget id 30 pipe 6 type 31 name : BUF6.0 stream none
[    2.810688] sof-audio-pci 0000:00:1f.3: buffer BUF6.0: size 768 caps 0x71
[    2.810691] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[    2.810815] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[    2.810819] sof-audio-pci 0000:00:1f.3: tplg: ready widget id 31 pipe 6 type 31 name : BUF6.1 stream none
[    2.810821] sof-audio-pci 0000:00:1f.3: buffer BUF6.1: size 768 caps 0x71
[    2.810824] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[    2.810961] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[    2.810966] sof-audio-pci 0000:00:1f.3: tplg: ready widget id 32 pipe 6 type 27 name : HDA2.OUT stream iDisp3
[    2.810972] sof-audio-pci 0000:00:1f.3: dai HDA2.OUT: type 3 index 2
[    2.810974] sof-audio-pci 0000:00:1f.3:  config: periods snk 0 src 2 fmt 2
[    2.810980] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    2.811120] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    2.811123] sof-audio-pci 0000:00:1f.3: tplg: connected widget HDA2.OUT -> DAI link iDisp3
[    2.811127] sof-audio-pci 0000:00:1f.3: tplg: ready widget id 33 pipe 6 type 32 name : PIPELINE.6.HDA2.OUT stream HDA2.OUT
[    2.811129] sof-audio-pci 0000:00:1f.3: tplg: pipeline id 6 comp 33 scheduling comp id 32
[    2.811132] sof-audio-pci 0000:00:1f.3: pipeline PIPELINE.6.HDA2.OUT: period 1000 pri 0 mips 5000 core 0 frames 0
[    2.811136] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[    2.811270] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[    2.811276] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[    2.811278] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40070000: GLB_PM_MSG: CORE_ENABLE
[    2.811372] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40070000: GLB_PM_MSG: CORE_ENABLE
[    2.811377] sof-audio-pci 0000:00:1f.3: tplg: ready widget id 34 pipe 7 type 23 name : PCM5P stream SRC Playback 5
[    2.811379] sof-audio-pci 0000:00:1f.3: loaded host PCM5P
[    2.811381] sof-audio-pci 0000:00:1f.3:  config: periods snk 3 src 0 fmt 0
[    2.811386] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    2.811520] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    2.811524] sof-audio-pci 0000:00:1f.3: tplg: ready widget id 35 pipe 7 type 34 name : SRC7.0 stream none
[    2.811527] sof-audio-pci 0000:00:1f.3: src SRC7.0: source rate 0 sink rate 48000
[    2.811529] sof-audio-pci 0000:00:1f.3:  config: periods snk 3 src 3 fmt 2
[    2.811534] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    2.811663] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    2.811667] sof-audio-pci 0000:00:1f.3: tplg: load control type 1 name : 7 Master Playback Volume
[    2.811673] sof-audio-pci 0000:00:1f.3: tplg: load kcontrol index 36 chans 2
[    2.811676] sof-audio-pci 0000:00:1f.3: tplg: ready widget id 36 pipe 7 type 6 name : PGA7.0 stream none
[    2.811679] sof-audio-pci 0000:00:1f.3:  config: periods snk 2 src 2 fmt 2
[    2.811684] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    2.811824] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[... 53 lines cut ...]
[    2.813346] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.813354] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.813358] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    2.813366] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.813375] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.813379] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    2.813387] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.813396] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.813400] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    2.813408] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.813416] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.813421] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    2.813426] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.813434] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.813438] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    2.813447] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.813455] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    2.813459] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    2.813462] sof-audio-pci 0000:00:1f.3: error: tplg component load failed -12
[    2.813470] sof-audio-pci 0000:00:1f.3: error: failed to load DSP topology -22
[    2.813473] sof-audio-pci 0000:00:1f.3: ASoC: failed to probe component -22
[    2.813490] sof_rt5682 sof_rt5682: ASoC: failed to instantiate card -22
[    2.813518] sof_rt5682: probe of sof_rt5682 failed with error -22
```
[dmesg.txt](https://github.com/thesofproject/sof/files/3508854/dmesg.txt)

ClarexZhou, 2019-08-19:
tplg can be loaded on CML chromebook after apply PR #1728

aiChaoSONG, 2019-08-19:
Firmware can be loaded normally after disabling Wov feature.
To disable Wov feature, go to **sof/tools/topology** folder, change **line 9** in **sof-cml-rt5682-max98357a.m4** from ```include(`sof-cml-rt5682-kwd.m4')``` to ```include(`sof-cml-rt5682.m4')```

lgirdwood, 2019-08-19:
@ClarexZhou what is the FW log ? I'm guessing we have run out of free memory ?

juimonen, 2019-08-19:
I checked the buffer sizes in topology conf file before and after my framecount patch and with the fix 
#1728 they are the same... I'm not sure if something has changes in the memory maps etc...?

aiChaoSONG, 2019-08-20:
FW trace log and error log are both empty

keyonjie, 2019-08-20:
Why the size of buf 8.0 is 0? Is it recent regression? @ClarexZhou @aiChaoSONG 
```
[    2.813180] sof-audio-pci 0000:00:1f.3: buffer BUF8.0: size 0 caps 0x71
```

juimonen, 2019-08-20:
@keyon yes it was a clear bug in the frame count calculation patch, nobody was too much testing kwd (kwd is almost only one using special buffer size). it is fixed with #1728 and the buffer sizes should be the same as before the change.

juimonen, 2019-08-20:
so do not test this without #1728... btw it should be merged, don't know why it is not...

aiChaoSONG, 2019-08-20:
after several test on different commit, commit `77b6db88` initially shows the bug

juimonen, 2019-08-20:
@aiChaoSONG so to see what is happening, apply #1728, build topologies again and see/attach here fw logs/dmesg

aiChaoSONG, 2019-08-20:
Firmware now can be loaded normally.
[soflogger.log](https://github.com/thesofproject/sof/files/3518837/soflogger.log)
[dmesg.log](https://github.com/thesofproject/sof/files/3518838/dmesg.log)

ClarexZhou, 2019-08-21:
Close for PR #1728 is merged.

closed 2019-08-21 by ClarexZhou