thesofproject/sof #1731: [BUG][ICL] DMIC I/O error when doing pause arecord then resume

dengyangchao, 2019-08-20:
**Describe the bug**
Pause arecord then resume, wait 5s, system shows I/O error

**More info**
when connected HDMI or DP issue can not reproduce

**To Reproduce**

1.Arecord via "arecord -Dhw:0,6 -r48000 -c4 -fs16_le 1.wav -vv -i"
2.Pause arecord
3.Resume asrecord
4.Wait 5s

**Reproduction Rate**
100%

**Expected results**
Resume playback normally after pause-release

**Actual result**
I/O error
[image]

**Environment**
Firmware: 5476ed7f https://github.com/thesofproject/sof/commits/master
Kernel: a600576c https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-hda-generic-4ch.tplg same as firmware
platform：ICL-RVP

**dmesg**
```
[  718.124732] sof-audio-pci 0000:00:1f.3: pcm: open stream 6 dir 1
[  718.124733] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[  718.124734] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[  718.124735] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[  718.124871] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 6 dir 1
[  718.124872] sof-audio-pci 0000:00:1f.3: generating page table for 000000000840c015 size 0xffc0 pages 16
[  718.124915] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  718.124955] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  718.124971] sof-audio-pci 0000:00:1f.3: period_bytes:0x3fc0
[  718.124972] sof-audio-pci 0000:00:1f.3: periods:4
[  718.125013] sof-audio-pci 0000:00:1f.3: stream_tag 2
[  718.125016] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  718.125586] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  718.125595] sof-audio-pci 0000:00:1f.3: pcm: stream dir 1, posn mailbox offset is 790528
[  718.125623] sof-audio-pci 0000:00:1f.3: nothing to do in IPC IRQ thread
[  718.127027] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 6 dir 1 cmd 1
[  718.127055] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2024001e successful
[  718.127056] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  718.127473] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  718.617356] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  718.617403] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  719.117691] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  719.117735] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  725.630584] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 6 dir 1 cmd 3
[  725.630611] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
[  725.630620] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[  725.631011] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[  726.617358] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  726.617396] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  749.418518] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 6 dir 1 cmd 4
[  749.418547] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2024001e successful
[  749.418549] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[  749.419023] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[  750.117344] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  750.117393] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  751.617923] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  751.617973] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  752.033241] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 6 dir 1 cmd 0
[  752.033369] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
[  752.033383] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  752.033615] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  752.033621] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  752.033828] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  752.034074] sof-audio-pci 0000:00:1f.3: pcm: close stream 6 dir 1
[  752.380688] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  752.380738] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  754.282507] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20140000 successful
[  754.330475] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  754.330724] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  754.330803] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[  754.330815] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[  754.330817] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[  754.330857] sof-audio-pci
```
[dmesg.log](https://github.com/thesofproject/sof/files/3519059/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3519060/logger.log)

dengyangchao, 2019-08-21:
issue still can reproduce, but can not reproduce when connecting DP or HDMI

**to Reproduce**
1.reboot system via "sudo reboot" and wait it to suspend status
2.Arecord via "arecord -Dhw:0,6 -r48000 -c4 -fs16_le 1.wav -vv -i"
3.Pause arecord
4.Resume asrecord wait 10s
Reproduce rate 100%

**More info**
1. connect DP or HDMI, then reboot via "sudo reboot" and wait it to suspend status
2. Arecord via "arecord -Dhw:0,6 -r48000 -c4 -fs16_le 1.wav -vv -i"
3. Pause arecord
4. Resume asrecord wait 10s
5. Repeat pause-release
Issue can not reproduce

**Test recipe**
Firmware: db138eba https://github.com/thesofproject/sof/commits/master
Kernel: 2a22b942 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-hda-generic-4ch.tplg same as firmware
platform：ICL-RVP

[dmesg.log](https://github.com/thesofproject/sof/files/3523688/dmesg.log)
[dmesg_with_hdmi.log](https://github.com/thesofproject/sof/files/3523689/dmesg_with_hdmi.log)

dabekjakub, 2019-09-09:
duplicate os #1760

closed 2019-09-09 by dabekjakub