thesofproject/sof #1733: [BUG] topology: sof-hda-generic-idisp.m4 prevents card probe

plbossart, 2019-08-21:
**Describe the bug**

on Up2 with the -idisp-generic topology, we see the following errors

````
[   59.430695] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for iDisp3_out
[   59.430700] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route iDisp3_out -> direct -> iDisp3 Tx
[   59.430704] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for iDisp2_out
[   59.430706] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route iDisp2_out -> direct -> iDisp2 Tx
[   59.430710] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for iDisp1_out
[   59.430713] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route iDisp1_out -> direct -> iDisp1 Tx
````
with recent changes in the kernel (' ASoC: soc-core: tidyup for snd_soc_dapm_add_routes()
'), these errors are now trapped and the card probe fails.

we need to fix the topology to remove those routes.

**To Reproduce**
Take an Up2 with no SSP outputs, only HDMI, and the latest topic/sof-dev kernel and SOF firmware.

**Reproduction Rate**
100%

**Expected behavior**
Topology deals with those routes, no warnings are shown and card probes

**Impact**
showstopper for these boards.

**Environment**
firmware:   db138eba8f0a5abd2f8b2a75d9bd235ab47fa0c9
kernel:   2a22b942ea414328b4e40331c09d72264331a7a1
topology file: sof-hda-generic-idisp.tplg

plbossart, 2019-08-21:
log for problematic configs:

````
[   24.437157] sof-audio-pci 0000:00:0e.0: warning: No matching ASoC machine driver found
[   24.437607] sof-audio-pci 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if 0x040100
[   24.437821] sof-audio-pci 0000:00:0e.0: use msi interrupt mode
[   24.441255] sof-audio-pci 0000:00:0e.0: bound 0000:00:02.0 (ops i915_globals_exit [i915])
[   24.445513] sof-audio-pci 0000:00:0e.0: hda codecs found, mask 4
[   24.445517] sof-audio-pci 0000:00:0e.0: using HDA machine driver skl_hda_dsp_generic now
[   24.458440] HDMI HDA Codec ehdaudio0D2: Max dais supported: 3
[   24.486139] sof-audio-pci 0000:00:0e.0: Firmware info: version 1:1:0-db138
[   24.486148] sof-audio-pci 0000:00:0e.0: Firmware: ABI 3:9:0 Kernel ABI 3:9:0
[   24.486357] sof-audio-pci 0000:00:0e.0: firmware boot complete
[   24.493473] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp1
[   24.493479] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp2
[   24.493481] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp3
[   24.493484] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link dmic01
[   24.493487] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link dmic16k
[   24.494349] sof-audio-pci 0000:00:0e.0: Topology: ABI 3:9:0 Kernel ABI 3:9:0
[   24.494364] sof-audio-pci 0000:00:0e.0: warning: widget type 7 name iDisp3 Tx not handled
[   24.494369] sof-audio-pci 0000:00:0e.0: warning: widget type 7 name iDisp2 Tx not handled
[   24.495222] sof-audio-pci 0000:00:0e.0: warning: widget type 7 name iDisp1 Tx not handled
[   24.497519] sof-audio-pci 0000:00:0e.0: ASoC: Parent card not yet available, widget card binding deferred
[   24.499116] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 5:0
[   24.499296] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 7:0
[   24.499484] skl_hda_dsp_generic skl_hda_dsp_generic: intel-hdmi-hifi1 <-> iDisp1 Pin mapping ok
[   24.499496] skl_hda_dsp_generic skl_hda_dsp_generic: intel-hdmi-hifi2 <-> iDisp2 Pin mapping ok
[   24.499509] skl_hda_dsp_generic skl_hda_dsp_generic: intel-hdmi-hifi3 <-> iDisp3 Pin mapping ok
[   24.499521] skl_hda_dsp_generic skl_hda_dsp_generic: dmic-hifi <-> DMIC01 Pin mapping ok
[   24.499531] skl_hda_dsp_generic skl_hda_dsp_generic: dmic-hifi <-> DMIC16k Pin mapping ok
[   24.499582] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> HDMI1 1 mapping ok
[   24.499619] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> HDMI2 2 mapping ok
[   24.499655] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> HDMI3 3 mapping ok
[   24.499673] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for iDisp3_out
[   24.499676] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route iDisp3_out -> direct -> iDisp3 Tx
[   24.499680] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for iDisp2_out
[   24.499682] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route iDisp2_out -> direct -> iDisp2 Tx
[   24.499686] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for iDisp1_out
[   24.499688] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route iDisp1_out -> direct -> iDisp1 Tx
[   24.499807] ------------[ cut here ]------------
[   24.499852] WARNING: CPU: 0 PID: 1251 at sound/core/control.c:466 snd_ctl_remove+0x109/0x120 [snd]
[   24.499853] Modules linked in: snd_soc_skl_hda_dsp(+) snd_soc_hdac_hdmi snd_soc_dmic snd_sof_pci snd_sof_intel_hda_common snd_soc_hdac_hda snd_sof_intel_hda snd_hda_ext_core snd_hda_codec snd_hwdep snd_hda_core snd_sof_acpi snd_sof_intel_byt snd_sof_intel_bdw snd_sof_intel_ipc snd_sof snd_sof_xtensa_dsp snd_soc_acpi_intel_match snd_soc_acpi snd_soc_max98090 snd_soc_max98357a snd_soc_wm8804_i2c snd_soc_wm8804 snd_soc_pcm512x_i2c snd_soc_pcm512x snd_soc_rt5682 snd_soc_rt5677_spi snd_soc_rt5670 snd_soc_rt5651 snd_soc_rt5645 snd_soc_rt5640 snd_soc_rl6231 snd_soc_rt286 snd_soc_rl6347a snd_soc_da7213 snd_soc_core snd_compress snd_pcm snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq snd_seq_device snd_timer snd soundcore x86_pkg_temp_thermal ax88179_178a usbnet i915 i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops mei_me mei drm efivarfs mmc_block sdhci_pci cqhci sdhci intel_lpss_pci xhci_pci intel_lpss mfd_core xhci_hcd
[   24.499891] CPU: 0 PID: 1251 Comm: systemd-udevd Not tainted 5.3.0-rc1-test+ #21
[   24.499892] Hardware name: AAEON UP-APL01/UP-APL01, BIOS UPA1AM40 08/06/2018
[   24.499897] RIP: 0010:snd_ctl_remove+0x109/0x120 [snd]
[   24.499899] Code: 4c 24 40 65 48 33 0c 25 28 00 00 00 75 0b 48 83 c4 48 5b 5d 41 5c 41 5d c3 e8 83 83 3b f8 48 c7 c7 60 3e 6b c0 e8 a0 c2 40 f8 <0f> 0b b8 ea ff ff ff eb cb 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f
[   24.499900] RSP: 0018:ffffa7f8408e7960 EFLAGS: 00010246
[   24.499902] RAX: 0000000000000024 RBX: ffff8d3739e02298 RCX: 0000000000000006
[   24.499903] RDX: 0000000000000000 RSI: 0000000000000096 RDI: ffff8d373ba163d0
[   24.499904] RBP: ffff8d371adee8d8 R08: 0000000000000342 R09: 0000000000000004
[   24.499905] R10: 0000000000000000 R11: 0000000000000001 R12: 0000000000000001
[   24.499906] R13: 0000000000000000 R14: ffff8d37140836c0 R15: ffff8d3739e03e98
[   24.499908] FS:  00007facba4988c0(0000) GS:ffff8d373ba00000(0000) knlGS:0000000000000000
[... cut ...]
[   24.500548] RSP: 002b:00007ffeeeb08aa8 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[   24.500552] RAX: ffffffffffffffda RBX: 0000558c3e340b10 RCX: 00007facbaf1d219
[   24.500555] RDX: 0000000000000000 RSI: 00007facbae01cad RDI: 0000000000000007
[   24.500558] RBP: 00007facbae01cad R08: 0000000000000000 R09: 0000000000000000
[   24.500560] R10: 0000000000000007 R11: 0000000000000246 R12: 0000000000000000
[   24.500561] R13: 0000558c3e343680 R14: 0000000000020000 R15: 0000558c3e340b10
[   24.500564] ---[ end trace 04393d9d0176f6d7 ]---
````

plbossart, 2019-08-21:
And while we are at it we should fix these topology files to align formats and capabilities, the generic and generic-idisp versions are just too different for no good reason.

plbossart, 2019-08-21:
@kv2019i @ranj063 FYI.

kv2019i, 2019-08-21:
FYI to @singalsu as well. He's been looking at a topology related crash as well.

singalsu, 2019-08-21:
@kv2019i I get this with WHL if I try to leave out digital and idisp stuff from topology. It looks similar:

```
[    1.701882] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for iDisp3 Tx
[    1.701960] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route iDisp3 Tx -> direct -> hifi3
[    1.702033] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for iDisp3_out
[    1.702097] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route iDisp3_out -> direct -> iDisp3 Tx
[    1.702173] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for iDisp2 Tx
[    1.702237] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route iDisp2 Tx -> direct -> hifi2
[    1.702336] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for iDisp2_out
[    1.702402] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route iDisp2_out -> direct -> iDisp2 Tx
[    1.702476] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for iDisp1 Tx
[    1.702552] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route iDisp1 Tx -> direct -> hifi1
[    1.702625] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for iDisp1_out
[    1.702689] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route iDisp1_out -> direct -> iDisp1 Tx
[    1.702796] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for Digital CPU Playback
[    1.702869] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route Digital CPU Playback -> direct -> Digital Codec Playback
[    1.702957] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for codec1_out
[    1.703022] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route codec1_out -> direct -> Digital CPU Playback
[    1.703112] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for codec2_out
[    1.703178] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route codec2_out -> direct -> Alt Analog CPU Playback
[    1.703274] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for Digital CPU Capture
[    1.703344] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route Digital CPU Capture -> direct -> codec1_in
[    1.703425] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no sink widget found for Digital CPU Capture
[    1.703504] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route Digital Codec Capture -> direct -> Digital CPU Capture
[    1.703592] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no sink widget found for codec2_in
[    1.703655] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route Alt Analog CPU Capture -> direct -> codec2_in
[    1.703908] sof-audio-pci 0000:00:1f.3: error: failed to find dai iDisp3 in sof_link_unload
[    1.703972] sof-audio-pci 0000:00:1f.3: error: failed to find dai iDisp2 in sof_link_unload
[    1.704033] sof-audio-pci 0000:00:1f.3: error: failed to find dai iDisp1 in sof_link_unload
[    1.704117] sof-audio-pci 0000:00:1f.3: error: failed to find dai Digital Playback and Capture in sof_link_unload
[    1.704230] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[    1.704239] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1000f0f successful
[    1.704243] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    1.704247] ------------[ cut here ]------------
[    1.704288] WARNING: CPU: 1 PID: 295 at sound/core/control.c:466 snd_ctl_remove+0x109/0x120 [snd]
[    1.704289] Modules linked in: snd_soc_skl_hda_dsp(+) snd_soc_hdac_hdmi snd_hda_codec_realtek snd_hda_codec_generic snd_soc_dmic snd_sof_pci snd_sof_intel_hda_common snd_soc_hdac_hda snd_intel_nhlt snd_sof_intel_hda snd_sof_intel_byt snd_soc_acpi_intel_match snd_sof_intel_ipc snd_sof snd_sof_xtensa_dsp snd_soc_acpi snd_hda_ext_core snd_soc_core snd_hda_codec snd_hwdep snd_hda_core snd_pcm snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq iwlmvm i915 snd_seq_device snd_timer i2c_algo_bit x86_pkg_temp_thermal snd drm_kms_helper intel_powerclamp syscopyarea sysfillrect sysimgblt fb_sys_fops drm soundcore iwlwifi intel_lpss_pci processor_thermal_device intel_lpss intel_soc_dts_iosf mei_me mfd_core mei int3400_thermal int3403_thermal acpi_thermal_rel int340x_thermal_zone intel_pch_thermal efivarfs xhci_pci xhci_hcd
[    1.704324] CPU: 1 PID: 295 Comm: systemd-udevd Not tainted 5.3.0-rc4-test+ #6
[    1.704326] Hardware name: Dell Inc. Inspiron 5490/, BIOS 86.1.4 04/08/2019
[    1.704331] RIP: 0010:snd_ctl_remove+0x109/0x120 [snd]
[    1.704335] Code: 4c 24 40 65 48 33 0c 25 28 00 00 00 75 0b 48 83 c4 48 5b 5d 41 5c 41 5d c3 e8 93 53 56 fb 48 c7 c7 60 6e 50 c0 e8 20 91 5b fb <0f> 0b b8 ea ff ff ff eb cb 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f
[    1.704337] RSP: 0018:ffffb3d640547970 EFLAGS: 00010246
[    1.704339] RAX: 0000000000000024 RBX: ffff9a77a601b498 RCX: 0000000000000006
[    1.704341] RDX: 0000000000000000 RSI: 0000000000000092 RDI: ffff9a77a82563d0
[    1.704342] RBP: ffff9a77a08cced8 R08: 00000000000004c1 R09: 0000000000000004
[    1.704344] R10: 0000000000000000 R11: 0000000000000001 R12: 0000000000000001
[    1.704345] R13: 0000000000000000 R14: ffff9a77a148c000 R15: ffff9a77a601b898
[    1.704348] FS:  00007f82e4d09680(0000) GS:ffff9a77a8240000(0000) knlGS:0000000000000000
[    1.704350] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[... cut ...]
[    1.704518] RSP: 002b:00007ffcaefc4868 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[    1.704521] RAX: ffffffffffffffda RBX: 000056134d395fe0 RCX: 00007f82e4827839
[    1.704522] RDX: 0000000000000000 RSI: 00007f82e4506145 RDI: 0000000000000007
[    1.704524] RBP: 00007f82e4506145 R08: 0000000000000000 R09: 00007ffcaefc4980
[    1.704525] R10: 0000000000000007 R11: 0000000000000246 R12: 0000000000000000
[    1.704526] R13: 000056134d387f80 R14: 0000000000020000 R15: 000056134d395fe0
[    1.704530] ---[ end trace b61a0cdef383dc1a ]---

```

xiulipan, 2019-08-23:
@plbossart 
Fixed with the https://github.com/thesofproject/sof/pull/1746

Test on my own UP2 and 3 PCM can work with sound output to HDMI monitor.

PR #1755 (2019-08-27): topology: add virtual routes for hda-idisp
Add missing virtual routes that will fail the probe.

Fix #1733
closed 2019-08-27 by tlauda