thesofproject/sof #1739: [WIP] [BUG] cavs: core 3 must be powered on for loads > 120 MCPS on icl

mmaka1, 2019-08-22:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?
Please also read https://thesofproject.github.io/latest/howtos/process/bug-tracking.html for further information on submitting bugs.

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug) 

**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
Does the reproduction rate vary with any other configuration or user action, if so please describe and show the new reproduction rate.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of 3 repositories: sof (firmware), linux (kernel driver) and soft (tools & topology).
2) Name of the topology file
3) Name of the platform(s) on which the bug is observed.
4) Reproducibility Rate. If you can only reproduce it randomly, it’s useful to report how many times the bug has been reproduced vs. the number of attempts it’s taken to reproduce the bug.

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

plbossart, 2019-08-22:
@mmaka1 care to provide more details? We see random IPC and underflows on ICL w/ SoundWire, if there is a connection.

lgirdwood, 2019-08-30:
@plbossart @ranj063 has anyone tried this yet ?

ranj063, 2019-08-30:
@lgirdwood I dont think we've tried it yet. I think Pierre asked for more details but @mmaka1 hasnt replied yet. 

At the moment the code loading sequence powers up all 4 cores before loading the FW. Not sure what Marcin means here and when Core 3 should be powered up precisely.

lbetlej, 2020-01-07:
@ranj063 I'll get back to you once SW/FW programming sequence is finalized for scenarios that require HP Ring Oscillator (400 MHz) clock. Could you confirm current implementation in the driver for initial boot:
1. Set SPA(0..PLATFORM_CORE_COUNT) bit 
2. Wait for CPA (0..PLATFORM_CORE_COUNT) bit
3. Wait for FW ready
4. SPA is not set to 0 for cores (1..PLATFORM_CORE_COUNT) -> all cores are up ?

tlauda, 2020-04-29:
@lbetlej Should we move this to linux repo, since it looks like additional work for kernel?