thesofproject/sof #1756: [BUG][ICL]When set capture volume to 0  DMIC 4ch still can capture voice

dengyangchao, 2019-08-26:
**Describe the bug**
When set capture volume to 0  DMIC 4ch still can capture voice, the issue exist on both 0,6 and 0,7 

**To Reproduce**
1. Reboot system via"sudo reboot"
2. wait it to suspend status
3. Set alsamixer DMIC capture volume to 0
4. Arecord via "arecord -Dhw:0,6 -r48000 -c4 -fs16_le 1.wav -vv -i" or "arecord -Dhw:0,7 -r48000 -c4 -fs16_le 1.wav -vv -i"
5. aplay the capture via "paplay 1.wav -vvvv"

**Reproduction Rate**
100%

**Expected behavior**
when set volume to 0, DMIC can't capture voice

**Impact**
DMIC still can capture voice

**Environment**
Firmware: 66da4e89 https://github.com/thesofproject/sof/commits/master
Kernel: 2a22b942 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-hda-generic-4ch.tplg same as firmware
platform：ICL-RVP

**Screenshots or console output**
[image]

**Dmesg**
```
[  159.485089] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50030000: GLB_COMP_MSG: SET_DATA
[  159.485091] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  159.485120] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  159.485784] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50030000: GLB_COMP_MSG: SET_DATA
[  159.485789] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  159.486167] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  159.486174] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50030000: GLB_COMP_MSG: SET_DATA
[  159.486632] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50030000: GLB_COMP_MSG: SET_DATA
[  159.486635] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  159.487135] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  159.487139] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  159.487473] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  159.487476] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  159.487850] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  159.487853] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  159.488333] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  159.488337] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  159.488672] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  159.488676] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  159.489205] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  159.489209] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  159.489540] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  159.489543] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  159.489879] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  159.489883] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[  159.490145] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[  159.490170] sof-audio-pci 0000:00:1f.3: pcm: open stream 7 dir 1
[  159.490172] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[  159.490173] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[  159.490173] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[  159.490355] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 7 dir 1
[  159.490357] sof-audio-pci 0000:00:1f.3: generating page table for 000000005acb3007 size 0xf9c0 pages 16
[  159.490389] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  159.490471] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  159.490482] sof-audio-pci 0000:00:1f.3: period_bytes:0x3e40
[  159.490483] sof-audio-pci 0000:00:1f.3: periods:4
[  159.490578] sof-audio-pci 0000:00:1f.3: stream_tag 2
[  159.490582] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  159.491143] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  159.491145] sof-audio-pci 0000:00:1f.3: pcm: stream dir 1, posn mailbox offset is 790528
[... 48 lines cut ...]
[  171.146877] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x5 successful
[  171.146903] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x140000 successful
[  171.146914] sof-audio-pci 0000:00:1f.3: Firmware download successful, booting...
[  171.152201] sof-audio-pci 0000:00:1f.3: ipc rx: 0x70000000: FW_READY
[  171.152209] sof-audio-pci 0000:00:1f.3: ipc: DSP is ready 0x70000000 offset 0x81000
[  171.152222] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x70000000: FW_READY
[  171.152312] sof-audio-pci 0000:00:1f.3: firmware boot complete
[  171.152337] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[  171.152353] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000002 at  00000048
[  171.152370] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000004 at  00000044
[  171.152388] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000044
[  171.152432] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  171.152477] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  171.152496] sof-audio-pci 0000:00:1f.3: period_bytes:0x0
[  171.152499] sof-audio-pci 0000:00:1f.3: periods:1
[  171.152527] sof-audio-pci 0000:00:1f.3: stream_tag: 1
[  171.152542] sof-audio-pci 0000:00:1f.3: ipc tx: 0x90030000: GLB_TRACE_MSG
[  171.153214] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x90030000: GLB_TRACE_MSG
[  171.153237] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x14001e successful
[  171.153251] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  171.153502] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
```

[dmesg.log](https://github.com/thesofproject/sof/files/3539914/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3539915/logger.log)

mengdonglin, 2019-09-02:
@dengyangchao Could you check the volume parameter value sent by IPC for the 4 channels? It should be one IPC for all 4 channels?

zrombel, 2019-09-02:
Hi,
I have tested this issue with slim driver with firmware 66da4e8 and master top 00828 and in both cases result are correct:

[image]
Audio format 48kHz 16b/16b 4ch

And this is the data that we are sending to FW via CompSetValue IPC
"data": [
    {
      "channel": 0,
      "value": 0
    },
    {
      "channel": 1,
      "value": 0
    },
    {
      "channel": 2,
      "value": 0
    },
    {
      "channel": 3,
      "value": 0
    }
  ]

dengyangchao, 2019-09-02:
@mengdonglin only two channel volume can be controlled by amixer/alsamixer:
```
numid=8,iface=MIXER,name='PGA10.0 10 Master Capture Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=70,step=0
  : values=52,51
  | dBscale-min=-50.00dB,step=1.00dB,mute=1
```
below audio is captured after changing volume to 0 via alsamixer
[image]

keyonjie, 2019-09-09:
This should be a tplg bug, it is defined 2 channels for the kcontrol, please see:
```
#
# Controls
#

# Volume Mixer control with max value of 32
SectionControlMixer."10 Master Capture Volume" {

        # control belongs to this index group
        index "10"

        #Channel register and shift for Front Left/Right
        channel."FL" {
                reg "1"
                shift "0"
        }
        channel."FR" {
                reg "1"
                shift "1"
        }

        # control uses bespoke driver get/put/info ID
        ops."ctl" {
                info "volsw"
                #256 binds the mixer control to volume get/put handlers
                get "256"
                put "256"
        }

        #
        max "70"
        invert "false"
        #TLV 80 steps from -50dB to +20dB for 1dB
        tlv "vtlv_m50s1"

}

```

keyonjie, 2019-09-09:
@singalsu comments?

singalsu, 2019-09-09:
I think there's problem that continuing this way the microphones > 2 should be mapped to ALSA channel maps those were designed for surround speakers system. I wonder if next two microphones could be surround-left (SL) and surround-right(SR). If it could work it could be low effort at topology level. But it could be very confusing since with microphones there's no clear "surround" usage. Also above 6 channels there's no main + surround speakers channels defined (except subwoofer and center channels).

Alternatively every > 2ch volume controller could be exposed as multiple 2ch volume controllers. This would be cleaner but more effort in driver side. Also changing volumes exactly the same time would be challenge.

@kv2019i @juimonen What do you think?

kv2019i, 2019-09-19:
@singalsu wrote:
> @kv2019i @juimonen What do you think?

I think you are right. We need to expose the mixer controls as multiple stereo pair (smaller effort), or extend ALSA with the concept of >2ch mixer controls (much larger as none of the ALSA interfaces and the many tools are equipped to handle this).

plbossart, 2019-09-19:
It may be worth expanding ALSA definitions, e.g. by using USB audio class 3 representations, see table 4.8
Also for mics there are descriptions in NHLT of the position of each mic, so maybe we can tag text in the ALSA controls to provide additional information. I think in general the Microsoft definitions are used by everyone for mic arrays.

````
  | Channel Relationship | Acronym | Description
-- | -- | -- | --
Generic | RELATIONSHIP_UNDEFINED | UND | Relationship undefined or unknown
MONO | M | A mono channel
LEFT | L | A generic Left channel
RIGHT | R | A generic Right channel
ARRAY | AR | A channel that is part of an array configuration
Input-Related | PATTERN_X | PX | The X channel of an X/Y microphone
PATTERN_Y | PY | The Y channel of an X/Y microphone
PATTERN_A | PA | The A channel of an A/B microphone
PATTERN_B | PB | The B channel of an A/B microphone
PATTERN_M | PM | The M channel of an M/S microphone
PATTERN_S | PS | The S channel of an M/S microphone
Output-Related | Front Left | FL | Refer to Figure 4-3
Front Right | FR | Refer to Figure 4-3
Front Center | FC | Refer to Figure 4-3
Front Left of Center | FLC | Refer to Figure 4-3
Front Right of Center | FRC | Refer to Figure 4-3
Front Wide Left | FWL | Refer to Figure 4-3
Front Wide Right | FWR | Refer to Figure 4-3
Side Left | SL | Refer to Figure 4-3
Side Right | SR | Refer to Figure 4-3
Surround Array Left | SAL | Refer to Figure 4-3
Surround Array Right | SAR | Refer to Figure 4-3
Back Left | BL | Refer to Figure 4-3
Back Right | BR | Refer to Figure 4-3
Back Center | BC | Refer to Figure 4-3
Back Left of Center | BLC | Refer to Figure 4-3
Back Right of Center | BRC | Refer to Figure 4-3
Back Wide Left | BWL | Refer to Figure 4-3
Back Wide Right | BWR | Refer to Figure 4-3
Top Center | TC | Refer to Figure 4-3
Top Front Left | TFL | Refer to Figure 4-3
Top Front Right | TFR | Refer to Figure 4-3
Top Front Center | TFC | Refer to Figure 4-3
Top Front Left of Center | TFLC | Refer to Figure 4-3
Top Front Right of Center | TFRC | Refer to Figure 4-3
Top Front Wide Left | TFWL | Refer to Figure 4-3
Top Front Wide Right | TFWR | Refer to Figure 4-3
Top Side Left | TSL | Refer to Figure 4-3
Top Side Right | TSR | Refer to Figure 4-3
Top Surround Array Left | TSAL | Refer to Figure 4-3
Top Surround Array Right | TSAR | Refer to Figure 4-3
Top Back Left | TBL | Refer to Figure 4-3
Top Back Right | TBR | Refer to Figure 4-3
Top Back Center | TBC | Refer to Figure 4-3
Top Back Left Of Center | TBLC | Refer to Figure 4-3
Top Back Right Of Center | TBRC | Refer to Figure 4-3
Top Back Wide Left | TBWL | Refer to Figure 4-3
Top Back Wide Right | TBWR | Refer to Figure 4-3
Bottom Center | BC | Refer to Figure 4-3
Bottom Front Left | BFL | Refer to Figure 4-3
Bottom Front Right | BFR | Refer to Figure 4-3
Bottom Front Center | BFC | Refer to Figure 4-3
Bottom Front Left Of Center | BFLC | Refer to Figure 4-3
Bottom Front Right Of Center | BFRC | Refer to Figure 4-3
Bottom Front Wide Left | BFWL | Refer to Figure 4-3
Bottom Front Wide Right | BFWR | Refer to Figure 4-3
Bottom Side Left | BSL | Refer to Figure 4-3
Bottom Side Right | BSR | Refer to Figure 4-3
Bottom Surround Array Left | BSAL | Refer to Figure 4-3
Bottom Surround Array Right | BSAR | Refer to Figure 4-3
Bottom Back Left | BBL | Refer to Figure 4-3
Bottom Back Right | BBR | Refer to Figure 4-3
Bottom Back Center | BBC | Refer to Figure 4-3
Bottom Back Left Of Center | BBLC | Refer to Figure 4-3
Bottom Back Right Of Center | BBRC | Refer to Figure 4-3
Bottom Back Wide Left | BBWL | Refer to Figure 4-3
Bottom Back Wide Right | BBWR | Refer to Figure 4-3
Low Frequency Effects | LFE | Refer to Figure 4-3
Low Frequency Effects Left | LFEL | Refer to Figure 4-3
Low Frequency Effects Right | LFER | Refer to Figure 4-3
Headphone Left | HPL | Refer to Figure 4-3
Headphone Right | HPR | Refer to Figure 4-3
````

wenqingfu, 2019-10-24:
@singalsu  where should the fix go? do you think you can own this?

singalsu, 2020-01-08:
@wenqingfu Yes I can own this. The fix would be in ALSA, I need to discuss here with @kv2019i how it could go.

The definitions from USB audio class 3 look useful. Thanks @plbossart !

sinahuang, 2020-01-10:
Issue also can be reproduced on TGL-U with soundwire-latest(commit:921ce33) when arecord with DMIC 4ch(0,3 and 0,4)
**Test recipe**

Kernel:https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 921ce33
7fde475
FW: https://github.com/thesofproject/sof/commits/master commit: c6baa97
Tplg: Same with TGL FW branch : sof-tgl-rt711-rt1308.tplg
Platform:TGL-U RVP with codec ALC711 in SDW mode and  RT1308 in I2S mode

singalsu, 2020-02-13:
P2 - I need to increase priority for this. The FW is OK, the fix will be entirely on driver side.

singalsu, 2020-02-18:
This need to be moved to driver side. The driver sends only controls to channels 0 and 1.

```
     0      2          IPC          19840492.552083       208.750000    src/ipc/handler.c:1042	ipc: hdr 0x50010000 tx (108) > rx (92)
    0      2          IPC          19840496.875000         4.322917    src/ipc/handler.c:1055	ipc: comp 42 -> cmd 2
    0      2         COMP 7.42     19840499.427083         2.552083 udio/volume/volume.c:612 	volume_cmd()
    0      2         COMP 7.42     19840502.187500         2.760417 udio/volume/volume.c:526 	volume_ctrl_set_cmd(), SOF_CTRL_CMD_SWITCH, cdata->comp_id = 42
    0      2         COMP 7.42     19840504.791667         2.604167 udio/volume/volume.c:531 	volume_ctrl_set_cmd(), channel = 0, value = 1
    0      2         COMP 7.42     19840508.072917         3.281250 udio/volume/volume.c:531 	volume_ctrl_set_cmd(), channel = 1, value = 0
    0      2          IPC          19840594.010417        85.937500    src/ipc/handler.c:1042	ipc: hdr 0x50010000 tx (108) > rx (92)
    0      2          IPC          19840598.802083         4.791667    src/ipc/handler.c:1055	ipc: comp 42 -> cmd 0
    0      2         COMP 7.42     19840601.354167         2.552083 udio/volume/volume.c:612 	volume_cmd()
    0      2         COMP 7.42     19840604.010417         2.656250 udio/volume/volume.c:493 	volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, cdata->comp_id = 42
    0      2         COMP 7.42     19840606.718750         2.708333 udio/volume/volume.c:498 	volume_ctrl_set_cmd(), channel = 0, value = 65536
    0      2         COMP 7.42     19840609.895833         3.177083 udio/volume/volume.c:498 	volume_ctrl_set_cmd(), channel = 1, value = 65536

```