thesofproject/sof #1761: [BUG] assert call should not include code

finikorg, 2019-08-26:
assert() may be disabled and should not include any code, at the moment there are lots of memcpy constructions inside assert().

paulstelian97, 2019-08-26:
In the SOF codebase asserts cannot be disabled by any config. If that will change in the future, we do need to be aware of the pattern used with memcpy_s and on disabling still allow side effects to run.

For example, whenever we will disable assert we could do

    #if CONFIG_ASSERT
    #define assert(cond) (void)((cond) || (panic(SOF_IPC_PANIC_ASSERT), 0))
    #else
    #define assert(cond) (cond)
    #endif

Closing as invalid.

kv2019i, 2019-09-19:
I'd like to reopen this and add some folks @lgirdwood @jajanusz @tlauda 

To me SOF use of assert is not following normal industry standards and can lead to mistakes like we had in Zephyr PoC. E.g. 

ANSI C and opengroup -> assert() should not have side-effects
https://pubs.opengroup.org/onlinepubs/009695399/basedefs/assert.h.html

MS guidelines:
https://docs.microsoft.com/en-us/cpp/c-runtime-library/reference/assert-macro-assert-wassert?view=vs-2019

Why a custom assert practise?

paulstelian97, 2019-09-19:
@kv2019i From what I've seen many asserts with side effects in the arguments are really calls to memcpy_s. The memcpy_s function can be made to have the assert inside it (the C standard allows a "constraint handler" to be called in case one of the conditions making it return -EINVAL is encountered; this "constraint handler" may very well be a panic() call, possibly even with a different type from "assertion failure"). This way you can remove the assert() around memcpy_s calls and leave the others as-is (I don't think there is any other which relies on the SOF behavior, though I will still check)

kv2019i, 2019-09-19:
@paulstelian97 Ack, I think that would be much better.

Other alternative is to rename the assert() to something else (product_assert() or something better), so it is clear to people reading the code that this is not your standard ANSI C assert but something else.

finikorg, 2019-09-20:
Can we just have something like
```
ret = func()
assert(ret)
```

paulstelian97, 2019-09-20:
@kv2019i I'm doing some changes right now on this one. One question though: if memcpy_s detects a condition should it still trigger a panic even if assertions are disabled?

paulstelian97, 2019-09-20:
@finikorg @kv2019i Can you give me input on this draft PR? If it's good we can go forward.

lgirdwood, 2019-09-30:
@paulstelian97 @finikorg @kv2019i ACK, assert() should be non custom (operate/usage as expected) and we should rename with prefix if we need to customise.

paulstelian97, 2019-09-30:
@lgirdwood Would renaming the current `assert` function to some sort of `require` function be a nice change? I don't think `prod_assert` and similar would actually click.

lgirdwood, 2019-09-30:
@paulstelian97 best option is to change as per @finikorg comment. It's pretty industry standard.

PR #1849 (2019-10-30): assert: Do not call assert() with side effects in the argument
If the assert semantics are changed to be the more usual ones, on
disabling asserts the calls that I changed would have also removed the
desirable side effects when the assertion itself passed (no memcpy_s
would have occurred, in particular). I have changed the usage of the
assert() macro so that the code becomes immune to such a change.

Fixes #1761
closed 2019-10-30 by tlauda