thesofproject/sof #1771: [BUG]  - Timeout Error on play stream in parallel internal test.

jdrzaszcz, 2019-08-28:
**Describe the bug**
In parallel internal test TimeoutError on Play stream for second SSP.
Additionally on first SSP capture file got ~0.22 s when 3 s are played.

[image]

**To Reproduce**
Platform
WHL RVP

Run test_01_ptp_parallel_internal.py from https://gitlab.devtools.intel.com/cavs_scripts-py/cavs_scripts-py/tree/task-jdrzaszx-update_piplenes_in_mixer_test repository.

**Reproduction Rate**
100%

**Expected behavior**
Captured files on three SSP loopback with 3 second recording time.

example log file:
[parallel_internal.log](https://github.com/thesofproject/sof/files/3550416/parallel_internal.log)

lgirdwood, 2019-08-30:
@tlauda Seems the related to https://github.com/thesofproject/sof/issues/1762

tlauda, 2019-09-30:
@jdrzaszcz @lgirdwood Should be fixed with #1850.

closed 2019-10-17 by tlauda