thesofproject/sof #1772: [BUG] Recent change in calculations of frame count breaks KWD topology

mrajwa, 2019-08-28:
**Describe the bug**
Recent change in calculations of frame count from stream params made by commit id f9ec396b8ffe7ce4f72e11dc1f0e92222e6e78f5 |  pipeline: calculate frame count from stream params breaks buffer sizes in KWD topology therefore causing immediate XRUNs

**To Reproduce**
Use master branch + either KWD topology (i.e sof-cml-rt5682-kwd.tplg) + run arecord -Dhw:0,8 -M -N -r 16000 -c2 -f s24_le  -vvv -d 5 wov.wav

**Reproduction Rate**
100%

**Expected behavior**
No XRUN, right buffer sizes

juimonen, 2019-08-28:
trying to fix this in #1773. has to be tested I'm now with limited access to test devices

aiChaoSONG, 2019-08-29:
What machine do you use, cannot reproduce due to [WoV cannot trigger with topic/sof-dev(kernel) and master(sof) branch](https://github.com/thesofproject/sof/issues/1775)

wwittbrx, 2019-08-30:
I can confirm that #1773 fixes this issue.

jajanusz, 2019-08-30:
Fixed in #1773

closed 2019-08-30 by jajanusz