thesofproject/sof #1775: [BUG] WoV cannot trigger with topic/sof-dev(kernel)  and master(sof) branch

aiChaoSONG, 2019-08-29:
**Describe the bug**
Wov cannot trigger with topic/sof-dev(kernel)  and master(sof) branch

**To Reproduce**
1. run `arecord -Dhw:0,8 -M -N -r 16000 -c2 -f s24_le -vvv -d 5 wov.wav`
2. make some sound to trigger wov.

**Reproduction Rate**
100%

**Expected behavior**
Wov should be triggered and start  recording.

**Environment**
1) Kernel: topic/sof-dev
2) SOF: master
3) Topology: sof-cml-rt5682-max98357a.tplg(cml)

wwittbrx, 2019-08-29:
Could you attach sof-logger output? I wonder if this is a duplicate of https://github.com/thesofproject/sof/issues/1772.

Also I think topic/sof-dev does not have all PR's merged, https://github.com/thesofproject/linux/tree/topic/keyword_detect should work better, not sure though.

aiChaoSONG, 2019-09-02:
this issue exists even before the release of cml-007, I think it is not a duplicate of #1772. I think some code of sof or kernel  may not be put upstream. @wwittbrx @lgirdwood

keqiaozhang, 2019-09-02:
@mrajwa  
Please make sure all WoV-related fixes or patches are landed on master, we need to align the status with release branch. Thanks.

mrajwa, 2019-09-02:
@keqiaozhang, @aiChaoSONG, please wait for these two patches #1768 and #1689

aiChaoSONG, 2019-09-03:
On CML Chrome, after applying #1768 and #1689, still cannot trigger due to some errors. @keqiaozhang  @mrajwa 
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          MEM         112292132.343750 112292136.000000      src/lib/alloc.c:273 	error: 171 blocks needed for allocation but only 170 blocks are remaining
    0      1          DAI 8.48    112296571.041667      4438.697754      src/audio/dai.c:580 	dai_check_for_xrun() error: overrun due to no data available
    0      1         COMP         112296577.604167         6.562500 of/audio/component.h:758 	comp_overrun() error: dev->comp.id = 48, sink->free = 16, copy_bytes = 0
    0      1         PIPE         112296638.281250        60.677082 src/audio/pipeline.c:790 	pipeline_copy() error: ret = -61, start->comp.id = 48, dir = 0
    0      1         PIPE 8.49    112296645.781250         7.500000 src/audio/pipeline.c:984 	pipeline_task(): xrun recover failed! pipeline will be stopped!
```

YvonneYang2, 2019-09-03:
@mrajwa ,Wov shows IO error before trigger it on WHL platform with topic/sof-dev(kernel) and master(sof) branch. Issue still can be reproduced after applying #1768 and #1689 

**error logger without patch 1768 and 1689**
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      1          MEM         746878406.458333 746878400.000000      src/lib/alloc.c:273    error: 171 blocks needed for allocation but only 170 blocks are remaining
```

**error log after applying patch 1768 and 1689**
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          MEM         941441840.052083 941441856.000000      src/lib/alloc.c:273 	error: 171 blocks needed for allocation but only 170 blocks are remaining
    0      1          DAI 8.40    941446883.281250      5043.229004      src/audio/dai.c:580 	dai_check_for_xrun() error: overrun due to no data available
    0      1         COMP         941446889.687500         6.406250 of/audio/component.h:758 	comp_overrun() error: dev->comp.id = 40, sink->free = 16, copy_bytes = 0
    0      1         PIPE         941446950.104167        60.416668 src/audio/pipeline.c:790 	pipeline_copy() error: ret = -61, start->comp.id = 40, dir = 0
    0      1         PIPE 8.41    941446957.760417         7.656250 src/audio/pipeline.c:984 	pipeline_task(): xrun recover failed! pipeline will be stopped!
    0      1         COMP         942077383.125000    630425.375000 rc/audio/component.c:192 	comp_set_state() error: wrong state = 5, COMP_TRIGGER_RESET
    0      1         COMP         942077395.729167        12.604167 rc/audio/component.c:192 	comp_set_state() error: wrong state = 5, COMP_TRIGGER_RESET
```

**all logs**
[dmesg after applying patch.log](https://github.com/thesofproject/sof/files/3568009/dmesg.after.applying.patch.log)
[logger after applying patch.log](https://github.com/thesofproject/sof/files/3568010/logger.after.applying.patch.log)
[logger.log](https://github.com/thesofproject/sof/files/3568017/logger.log)
[dmesg.log](https://github.com/thesofproject/sof/files/3568018/dmesg.log)

mrajwa, 2019-09-03:
Have you applied #1768 and #1689 on top of the master branch? The XRUN issue was caused by https://github.com/thesofproject/sof/issues/1772 and should be fixed now on master branch. I did some testes yesterday and did not see this issue anymore. However I saw IPC timeout error in dmesg...I think this issue is handled by somebody already.

mrajwa, 2019-09-03:
@YvonneYang2, what I/O error are you talking about? All I see is XRUN?

aiChaoSONG, 2019-09-03:
PR #1773 fix #1772, and #1773 is merged, but we still see xrun with master branch. btw, we are using daily version of kernel and sof. @mrajwa

YvonneYang2, 2019-09-03:
> @YvonneYang2, what I/O error are you talking about? All I see is XRUN?

I/O error is showed in console when do arecord
[image]

aiChaoSONG, 2019-09-03:
retest with latest kernel and sof, no xrun, but there is a mem allocation erro , let't start a new issue. @YvonneYang2 @mrajwa

keqiaozhang, 2019-09-03:
@YvonneYang2 
I/O error is caused by the missing "-M -N" options, please retest it. Thanks!

YvonneYang2, 2019-09-03:
Retest it with right command " arecord -Dhw:0,8 -r 16000 -c2 -f s16_le -vvv -i 1.wav -M -N
", I/O error is gone. But still can reproduce the original issue with #1768 and #1689 .

**error logger**
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          MEM          23079106.562500  23079106.000000      src/lib/alloc.c:273 	error: 171 blocks needed for allocation but only 170 blocks are remaining
    0      1          DAI 8.40     23082850.364583      3743.802002      src/audio/dai.c:580 	dai_check_for_xrun() error: overrun due to no data available
    0      1         COMP          23082856.770833         6.406250 of/audio/component.h:758 	comp_overrun() error: dev->comp.id = 40, sink->free = 16, copy_bytes = 0
    0      1         PIPE          23082917.239583        60.468750 src/audio/pipeline.c:790 	pipeline_copy() error: ret = -61, start->comp.id = 40, dir = 0
    0      1         PIPE 8.41     23082924.791667         7.552083 src/audio/pipeline.c:984 	pipeline_task(): xrun recover failed! pipeline will be stopped!
    0      1         COMP          30841417.708333   7758493.000000 rc/audio/component.c:192 	comp_set_state() error: wrong state = 5, COMP_TRIGGER_RESET
    0      1         COMP          30841430.208333        12.500000 rc/audio/component.c:192 	comp_set_state() error: wrong state = 5, COMP_TRIGGER_RESET
```

**all logs**
[dmesg.log](https://github.com/thesofproject/sof/files/3568932/dmesg.log)
[error_logger.log](https://github.com/thesofproject/sof/files/3568933/error_logger.log)
[logger.log](https://github.com/thesofproject/sof/files/3568934/logger.log)

wwittbrx, 2019-09-03:
Please check the following recipe:

FW 90873c5 + #1689 + #1740 + #1768
Kernel f95e4eb + PR1037

I have checked this recipe today on my CML platform and it works for me with no issues.
As far as I know this is the only working recipe based on master branches. Please make sure you include all FW PRs and one kernel PR.

The test command I use for testing is
`sudo arecord -M -N -D hw:0,8 -f s24_le -c 2 -r 16000 -d 10 record_kd.wav`

YvonneYang2, 2019-09-04:
Test it with recipe as below, issue cannot reproduce on WHL platform.

FW: master branch 90873c5 + #1689  + #1740  + #1768 
Kernel: topic/sof-dev branch f95e4eb + PR1037

wwittbrx, 2019-09-04:
Ok, we are on the same page then. Lets wait for PRs to be merged to their respective master branches, retest and then close this issue.

wenqingfu, 2019-09-25:
All PRs above has been merged. Can this issue be closed now?  @keyonjie @mrajwa

aiChaoSONG, 2019-09-25:
@wenqingfu We will run tests on several platforms today, if everything is fine, then this issue can be closed.

YvonneYang2, 2019-09-25:
Issue cannot reproduce on whl-i2s and CML Chrome platforms. Close the bug.

Test recipe:
Firmware: d42c019 https://github.com/thesofproject/sof/commits/master
Kernel: c0f21e6 https://github.com/thesofproject/linux/commits/topic/sof-dev

closed 2019-09-25 by YvonneYang2