thesofproject/sof #1776: [BUG] KWD topology sof-glk-da7219-kwd.tplg load failed on GLK Chrome

aiChaoSONG, 2019-08-29:
**Describe the bug**
KWD topology load failed on GLK Chrome with return code -22

**To Reproduce**
Boot up and check on CML chrome notebook.

**Reproduction Rate**
100%

**Expected behavior**
topology should be loaded sucessfully

**Environment**
1) Kernel: `topic/sof-dev f95e4ebf`
2) SOF: `master 7791303c`
3) Topology: `sof-glk-da7219-kwd.tplg`
4) Platform: `GLK Chrome`
5) OS: `Ubuntu 1804`

```
[    2.344834] sof-audio-pci 0000:00:0e.0: buffer BUF8.0: size 7680 caps 0x71
[    2.344841] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[    2.344947] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[    2.344950] sof-audio-pci 0000:00:0e.0: tplg: ready widget id 54 pipe 8 type 31 name : BUF8.1 stream none
[    2.344951] sof-audio-pci 0000:00:0e.0: buffer BUF8.1: size 7680 caps 0x71
[    2.344954] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[    2.345091] sof-audio-pci 0000:00:0e.0: error: ipc error for 0x30200000 size 20
[    2.345092] sof-audio-pci 0000:00:0e.0: error: buffer BUF8.1 load failed
[    2.345094] sof-audio-pci 0000:00:0e.0: error: DSP failed to add widget id 0 type 31 name : BUF8.1 stream none reply -12
[    2.345097] sof-audio-pci 0000:00:0e.0: ASoC: failed to load widget BUF8.1
[    2.345118] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1010303 successful
[    2.345122] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1000303 successful
[    2.345126] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 1
[    2.345131] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[    2.345135] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[    2.345138] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 1
[    2.345142] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[    2.345146] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[    2.345149] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 1
[    2.345165] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[    2.345168] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[    2.345171] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 1
[    2.345176] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[    2.345180] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[    2.345182] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 1
[    2.345187] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[    2.345190] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[    2.345193] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 1
[    2.345210] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[    2.345213] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[    2.345216] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 1
[    2.345219] sof-audio-pci 0000:00:0e.0: error: tplg component load failed -12
[    2.345226] sof-audio-pci 0000:00:0e.0: error: failed to load DSP topology -22
[    2.345228] sof-audio-pci 0000:00:0e.0: ASoC: failed to probe component -22
```

[dmesg.log](https://github.com/thesofproject/sof/files/3554229/dmesg.log)

aiChaoSONG, 2019-08-29:
A previous similar load failed bug is fix on cml chrome #1727, #1727 may share the same cause with this one.

juimonen, 2019-08-29:
@aiChaoSONG so is it out of memory error? can you see dsp logs?

aiChaoSONG, 2019-08-29:
@juimonen  trace log and error log are both empty

juimonen, 2019-08-29:
@aiChaoSONG indeed the glk wov capture pipeline is configured for 48000kHz -> hence bigger buffers as in cml (where it is 16kHz). So let's change it to 16kHz

juimonen, 2019-08-29:
@aiChaoSONG can you try with https://github.com/thesofproject/sof/pull/1777

aiChaoSONG, 2019-08-29:
@juimonen #1777 fix this bug, let's wait it  to merge

aiChaoSONG, 2019-09-02:
#1777 is merged, so let's close this issue.

closed 2019-09-02 by aiChaoSONG