thesofproject/sof #1783: [BUG][SDW] Aplay failed for IO transfer timed out then "0x80010000: GLB_DAI_MSG: CONFIG" ipc timed out

ClarexZhou, 2019-08-30:
**Describe the bug**
IO transfer timed out then "0x80010000: GLB_DAI_MSG: CONFIG" ipc timed out when trigger aplay 

**To Reproduce**
1. aplay -Dhw:0,0 -r48000 -c2 -fs32_le wav/long.wav -vv -i -d 180
2. Ctrl-c to stop
3. aplay -Dhw:0,0 -r48000 -c2 -fs24_le wav/long.wav -vv -i -d 180

**frequency**
< 10%

**Actual result**
Aplay failed for IO transfer timed out then "0x80010000: GLB_DAI_MSG: CONFIG" ipc timed out 

Only do system reboot to recover.

**Expected result**
Aplay should be successfully.

**Environment**
linux: https://github.com/thesofproject/linux/commits/integration/soundwire-alpha commit: a8f3408
sof: https://github.com/thesofproject/sof/commits/sdw-001-drop-stable commit: 0223d94
Topology: file: sof-cml-rt700.m4 same path as firmware
platform: CML SDW RVP

**dmesg**
```
  807.251113] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[  807.251206] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[  807.252482] int-sdw int-sdw.1: sdw_cdns_thread: start
[  807.252494] int-sdw int-sdw.1: sdw_cdns_thread: Slave sticky status 00000002 00000000
[  807.252497] int-sdw int-sdw.1: sdw_handle_slave_status: start
[  807.252499] int-sdw int-sdw.1: Slave attached, programming device number
[  807.252684] int-sdw int-sdw.1: SDW Slave Addr: 10025d070000
[  807.252688] int-sdw int-sdw.1: SDW Slave class_id 0, part_id 700, mfg_id 25d, unique_id 0, version 1
[  807.252691] int-sdw int-sdw.1: in sdw_assign_device_num
[  807.252693] int-sdw int-sdw.1: Slave already registered, reusing dev_num:1
[  807.252696] int-sdw int-sdw.1: in sdw_assign_device_num, writing dev_num 1
[  807.281600] int-sdw int-sdw.1: intel_resume done
[  807.281626] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 0
[  807.281629] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[  807.281630] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[  807.281631] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[  807.281854] plb: soc_pcm_hw_params codec_dai name rt700-aif1
[  807.281859] rt700 sdw:1:25d:700:0:0: rt700_pcm_hw_params rt700-aif1
[  807.281935] int-sdw int-sdw.1: Msg Ack not received
[  807.281937] int-sdw int-sdw.1: Msg Ack not received
[  807.281938] int-sdw int-sdw.1: Msg Ack not received
[  807.281939] int-sdw int-sdw.1: Msg Ack not received
[  807.281939] int-sdw int-sdw.1: Msg Ack not received
[  807.281940] int-sdw int-sdw.1: Msg Ack not received
[  807.281942] int-sdw int-sdw.1: Msg ignored for Slave 0
[  807.281945] int-sdw int-sdw.1: No more devices to enumerate
[  807.281947] int-sdw int-sdw.1: End of sdw_program_device_num ret 0
[  807.281958] int-sdw int-sdw.1: sdw_cdns_thread: Slave sticky status 00000023 00000000
[  807.281962] int-sdw int-sdw.1: sdw_cdns_thread: end
[  807.281982] int-sdw int-sdw.1: sdw_cdns_thread: start
[  807.281989] int-sdw int-sdw.1: sdw_cdns_thread: Slave sticky status 00000021 00000000
[  807.281990] int-sdw int-sdw.1: sdw_handle_slave_status: start
[  807.281992] int-sdw int-sdw.1: processing Slave 1 Status 1
[  807.281993] int-sdw int-sdw.1: Slave 1 status attached, was already attached, ignoring
[  807.281995] int-sdw int-sdw.1: Updating Slave 1 status
[  807.281998] int-sdw int-sdw.1: sdw_handle_slave_status: Updating Slave 1 status done
[  807.282000] int-sdw int-sdw.1: sdw_handle_slave_status: end
[  807.282008] int-sdw int-sdw.1: sdw_cdns_thread: Slave sticky status 00000021 00000000
[  807.282012] int-sdw int-sdw.1: sdw_cdns_thread: end
[  809.335328] int-sdw int-sdw.1: IO transfer timed out
[  811.383406] int-sdw int-sdw.1: IO transfer timed out
[  813.431468] int-sdw int-sdw.1: IO transfer timed out
[  815.479448] int-sdw int-sdw.1: IO transfer timed out
[  815.479461] plb: soc_pcm_hw_params cpu_dai name SDW1 Pin2
[  815.479489] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  815.983443] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[  815.983459] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  815.983791] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  815.983806] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x80000000 intlctl 0xc0000001 rirb 00
[  815.983810] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x80000000 adspis 0x00000001
[  815.983821] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x90020000 mask 0x00000003
[  815.983823] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  815.983828] sof-audio-pci 0000:00:1f.3: error: failed to set DAI hw_params for link 1 dai->id 2 ALH 23
[  815.983834] int-sdw int-sdw.1: ASoC: can't set SDW1 Pin2 hw params: -110
[  815.983846]  SDW1-Playback: ASoC: hw_params BE failed -110
[  815.983848]  SDW1-speakers: ASoC: hw_params BE failed -110
[  815.984093] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
[  819.734373] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  819.734405] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  819.734474] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  819.734502] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  819.734942] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20140000 successful
[  819.769882] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  819.770045] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  819.770075] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[  819.770082] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1000f0f successful
[  819.770085] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[  819.770134] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000010 at  00000044
```

[ioTimedoutThenIPCtimedout_logger.txt](https://github.com/thesofproject/sof/files/3558022/ioTimedoutThenIPCtimedout_logger.txt)
[ioTimedoutThenIPCtimedout_dmesg.txt](https://github.com/thesofproject/sof/files/3558023/ioTimedoutThenIPCtimedout_dmesg.txt)

YvonneYang2, 2019-09-11:
Issue occurred one time during testing on ICL RVP with codec ALC700 in SDW mode. 

Test recipe:
linux : https://github.com/thesofproject/linux/commits/integration/soundwire-alpha  a8f3408
FW : https://github.com/thesofproject/sof/commits/sdw-001-drop-stable d9c0463
Topology: Same with FW branch: sof-icl-rt700.m4

slawblauciak, 2019-09-11:
Looks like there's an IO transfer timeout happening on the kernel side before the IPC

`[  809.335328] int-sdw int-sdw.1: IO transfer timed out
[  811.383406] int-sdw int-sdw.1: IO transfer timed out
[  813.431468] int-sdw int-sdw.1: IO transfer timed out
[  815.479448] int-sdw int-sdw.1: IO transfer timed out`